henne49/dbus-opendtu #4: multi inverter support

Fideus, 2022-11-22:
As discussed in victron Forum the json output of a OpenDTU with two inverter, and code example to select phase for each inverter independently. Current for multiple Inverter corrected. Also power Connection is selectable (victron AC IN / AC Out)

[multiphase.zip](https://github.com/henne49/dbus-opendtu/files/10071208/multiphase.zip)

henne49, 2022-11-22:
This is great, I will include most of it asap.

Just for me to get the logic right, you have now 2 Inverter.
* You want them to show on 2 different phases in Venus/Victron
* But they are both connected to the same Phase in real world?
* Why would you not want the total of Power and energy on the correct phase?
* Would it be better instead of having 1 opendtu inverter in Venus/Victron, having 2 but with correct real world phase?

Fideus, 2022-11-23:
Hello henne,
Yes, i connected two inverter on different phase in real. So the power should be shown for the correct phase.
I also thought about having two instance of the inverter, but this seems to be less structed, if you have more inverter.

henne49, 2022-11-23:
OK, I will rework the script to allow for multiple phases and also which Inverter should be on which phase, based on the serialnumber, in this way we can add multiple inverter on a single phase, but also account for multiple inverter on different phase.

henne49, 2022-11-23:
I have now added you code and also added ahoy support, please check it it works now for you

closed 2022-11-24 by henne49