henne49/dbus-opendtu #6: 2 Inverter, 1 phase, 1 dtu

Bonzai10, 2022-11-24:
hello henne,
thank you very much, i have now running the ahoy dtu in victron.

attached the json

Possible you already made, but i can´t check becaus its night.

Nice would be, to show instead of the "Phase"  the name of inverter 1,2,3 and also each power of the inverter because the phase is for me not interresting.

<html><body>
<div id="content"><div class="tabs "><nav class="tabs-navigation"><ul class="tabs-menu" role="tablist"><li class="tabs-menu-item json is-active" role="presentation"><span class="devtools-tab-line"></span></li></ul></nav></div></div><span class="devtools-tab-line"></span><span class="devtools-tab-line"></span><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"></div></div></div></div><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"><div class="devtools-separator"></div><div class="devtools-searchbox"><input class="searchBox devtools-filterinput" placeholder="JSON durchsuchen" value=""></div></div><div class="panelContent" id="json-scrolling-panel" tabindex="0">

  |  
-- | --
menu |  
name |  
0 | "Live"
1 | "Serial Console"
2 | "Settings"
3 | "-"
4 | "REST API"
5 | "-"
6 | "Update"
7 | "System"
link |  
0 | "/live"
1 | "/serial"
2 | "/setup"
3 | null
4 | "/api"
5 | null
6 | "/update"
7 | "/system"
trgt |  
0 | null
1 | null
2 | null
3 | null
4 | "_blank"
system |  
ssid | "WaveNetwork"
device_name | "AHOY-DTU"
version | "0.5.41"
build | "dec333f"
ts_uptime | 70
ts_now | 1669309931
ts_sunrise | 1669271678
ts_sunset | 1669304973
ts_sun_upd | 1669309922
wifi_rssi | -25
mac | "C8:C9:A3:56:AA:F1"
hostname | "AHOY-DTU"
pwd_set | false
sdk | "2.2.2-dev(38a443e)"
cpu_freq | 80
heap_free | 12080
sketch_total | 2662400
sketch_used | 468
core_version | "3.0.2"
flash_size | 4096
heap_frag | 2
max_free_blk | 11952
reboot_reason | "Software/System restart"
esp_type | "ESP8266"
inverter |  
0 |  
name | "Balkon_li_re"
channels | 4
power_limit_read | 65535
last_alarm | "nothing"
ts_last_success | 0
ch |  
0 |  
0 | 0
1 | 0
2 | 0
3 | 0
4 | 0
5 | 0
6 | 0
7 | 0
8 | 0
9 | 0
10 | 0
1 |  
0 | 0
1 | 0
2 | 0
3 | 0
4 | 0
5 | 0
2 |  
0 | 0
1 | 0
2 | 0
3 | 0
4 | 0
5 | 0
3 |  
0 | 0
1 | 0
2 | 0
3 | 0
4 | 0
5 | 0
4 |  
0 | 0
1 | 0
2 | 0
3 | 0
4 | 0
5 | 0
ch_names |  
0 | "AC"
1 | ""
2 | ""
3 | ""
4 | ""
1 |  
name | "Balkon_mitte"
channels | 4
power_limit_read | 65535
last_alarm | "nothing"
ts_last_success | 0
ch |  
0 |  
0 | 0
1 | 0
2 | 0
3 | 0
4 | 0
5 | 0
6 | 0
7 | 0
8 | 0
9 | 0
10 | 0
1 |  
0 | 0
1 | 0
2 | 0
3 | 0
4 | 0
5 | 0
2 |  
0 | 0
1 | 0
2 | 0
3 | 0
4 | 0
5 | 0
3 |  
0 | 0
1 | 0
2 | 0
3 | 0
4 | 0
5 | 0
4 |  
0 | 0
1 | 0
2 | 0
3 | 0
4 | 0
5 | 0
ch_names |  
0 | "AC"
1 | ""
2 | ""
3 | ""
4 | ""
refresh_interval | 5
ch0_fld_units |  
0 | "V"
1 | "A"
2 | "W"
3 | "Hz"
4 | ""
5 | "°C"
6 | "kWh"
7 | "Wh"
8 | "W"
9 | "%"
10 | "var"
ch0_fld_names |  
0 | "U_AC"
1 | "I_AC"
2 | "P_AC"
3 | "F_AC"
4 | "PF_AC"
5 | "Temp"
6 | "YieldTotal"
7 | "YieldDay"
8 | "P_DC"
9 | "Efficiency"
10 | "Q_AC"
fld_units |  
0 | "V"
1 | "A"
2 | "W"
3 | "Wh"
4 | "kWh"
5 | "%"
fld_names |  
0 | "U_DC"
1 | "I_DC"
2 | "P_DC"
3 | "YieldDay"
4 | "YieldTotal"
5 | "Irradiation"

</div></div></div></div>
</body>
</html>

henne49, 2022-11-24:
Hi, the Phase name is given by Victron I cannot change. But looking at the file, I would hope the change I made gives you the correct data. you can but the them on different phases to show up in Victron VRM, but logically that is wrong for calculation. but for visibility it makes sense. Hope that it helps

Bonzai10, 2022-11-24:
hi, 
i check this over the weekend. Tanks.

i have now the esp32 board, but i´m to stupid to find the bin files for flashing with Espressif Flash Download Tool. can you help me.

henne49, 2022-11-25:
No, you are not, I had also to install via vscode and platform.io. There is no working download currently. https://github.com/tbnobody/OpenDTU#with-visual-studio-code

Bonzai10, 2022-11-25:
Oh, nice that i´am not to stupid.

it's not working properly for me yet. I'm afraid my esp8266 is too unstable. I have breakouts all the time.
but what I could see is that they take P_DC instead of P_AC. P_AC is smaller since that is what goes into the grid minus the inverter losses.

If I understood correctly, I have to set the following in the ini: Inverter=2 but set all phase to 3 (for me).

henne49, 2022-11-25:
opendtu is esp32 only. Ahoy both esps. What do you mean with "they"? I'll take P_AC which is the 3rd element in JSON 
https://github.com/henne49/dbus-opendtu/blob/be3da2369bc3f8803b6545ec5a0e08eee47f54b2/dbus-opendtu.py#L173 
for the config.ini, yes you are right.

Bonzai10, 2022-11-25:
Ok. thank you, i have to check again.
i belive the esp8266 can not handle the request every second. i´m losing everytime the connection

Bonzai10, 2022-11-25:
something must be wrong on my side

[image]

henne49, 2022-11-25:
Vlt auf deutsch.

Was genau ist falsch die Phase L3 oder die AC Gesamt? Power?

Stimmt denn die AC Gesamt Energie mit 3510kwh?

Bonzai10, 2022-11-25:
anbei die dateien. Irgendwas stimmt hier gar nicht.

Phase L3 Stimmt, AC Gesamt stimmt nicht. er zeigt auch über AC Gesamt einen falschen Wert an

{"menu":{"name":["Live","Serial Console","Settings","-","REST API","-","Update","System"],"link":["/live","/serial","/setup",null,"/api",null,"/update","/system"],"trgt":[null,null,null,null,"_blank"]},"system":{"ssid":"WaveNetwork","device_name":"AHOY-DTU","version":"0.5.41","build":"dec333f","ts_uptime":76,"ts_now":1669380383,"ts_sunrise":1669358139,"ts_sunset":1669391348,"ts_sun_upd":1669380368,"wifi_rssi":-24,"mac":"C8:C9:A3:56:AA:F1","hostname":"AHOY-DTU","pwd_set":false,"sdk":"2.2.2-dev(38a443e)","cpu_freq":80,"heap_free":11280,"sketch_total":2662400,"sketch_used":468,"core_version":"3.0.2","flash_size":4096,"heap_frag":2,"max_free_blk":11056,"reboot_reason":"Software/System restart","esp_type":"ESP8266"},"inverter":[{"name":"Balkon_li_re","channels":4,"power_limit_read":65535,"last_alarm":"nothing","ts_last_success":1669380381,"ch":[[231,0.06,13,49.98,0.57,9.2,833.729,302,13.6,95.588,18.8],[32.5,0.02,0.6,3,254.538,0],[32.5,0.2,6.6,152,74.922,1.808],[33.7,0.17,5.7,143,223.844,1.562],[33.7,0.02,0.7,4,280.425,0]],"ch_names":["AC","","","",""]},{"name":"Balkon_mitte","channels":4,"power_limit_read":65535,"last_alarm":"nothing","ts_last_success":1669380376,"ch":[[229.2,0.09,20.9,49.98,0.714,9.8,1843.655,449,22,95,20.5],[32.6,0.02,0.8,4,467.981,0],[32.6,0.23,7.3,145,467.223,2],[32.3,0.21,6.8,145,487.436,1.863],[32.3,0.22,7.1,155,421.015,1.945]],"ch_names":["AC","","","",""]}],"refresh_interval":5,"ch0_fld_units":["V","A","W","Hz","","°C","kWh","Wh","W","%","var"],"ch0_fld_names":["U_AC","I_AC","P_AC","F_AC","PF_AC","Temp","YieldTotal","YieldDay","P_DC","Efficiency","Q_AC"],"fld_units":["V","A","W","Wh","kWh","%"],"fld_names":["U_DC","I_DC","P_DC","YieldDay","YieldTotal","Irradiation"]}
[image]
[image]

henne49, 2022-11-25:
Danke, das JSON ist perfekt, ich sehe schon den Fehler, da wird einmal zuviel der Balkon_li_re addiert, dann sollte es passen. Vielen Dank für den Input.

Bonzai10, 2022-11-25:
nach einem neustart ließt er gar nichts mehr 
[image]
[image]

henne49, 2022-11-25:
das sieht aber nach einem Problem mit dem ESP aus, weil wir sehen 0 Werte, die werden immer dann geschrieben, wenn wir den ESP nicht erreichen. Schau mal bitte ins current.log file im Venus OS/nanopie im Ordner, wo du den Service installiert hast.

Bonzai10, 2022-11-25:
ich hoffe du kannst was erkennen.

[image]

Bonzai10, 2022-11-25:
ich glaube auch das es an dem esp liegt. ich werde am wochenende ein esp32 mit ahoy machen. für opendtu fehlen mir die Kenntnisse

henne49, 2022-11-25:
Das ist ein Connection Problem, der ESP32 macht das besser.

Für OpenDTU versuche ich mal an die Binaries zu kommen, bzw habe bei OpenDTU nachgefragt

henne49, 2022-11-25:
Hier sind die Binaries: https://github.com/tbnobody/OpenDTU/suites/9427291362/artifacts/445493438

henne49, 2022-11-25:
I have also fixed it now for ahoy and it should be working
[image]

Bonzai10, 2022-11-25:
> 

Besten Dank für die bin files. habe es endlich geschaft. aber die machen es einem echt nicht leicht.
ahoy ist einfacher :-).
ich werde leider erst morgen sehen ob es funktioniert

henne49, 2022-11-25:
Hi, for sake of other reader I use English. I have tested both ahoy and opendtu with JSON testfiles with multiple inverter even 3. and it all worked. I had computation error for ahoy, which is now fixed.

Bonzai10, 2022-11-26:
Hi, with esp32 and opendtu it's all ok. I runs very fine. Thank you for your support.

I believe you should merge that this driver is good only for esp32, the esp8266 is to slow for victron and makes trouble

closed 2022-11-26 by Bonzai10

henne49, 2022-11-26:
Fair point, but then let’s reduce the polling from 1 second to less for esp8266. I‘ll put this in the config.ini or will even check by reading the config file.

Excellent idea!