henne49/dbus-opendtu #7: Solar and consumption data wrong

Fideus, 2022-11-26:
I noted following problem:
last night I turned off the esp32 running OpenDTU for 2 hours.
This morning I got a huge solar and consumption peak at the time when the inverter starts: ( they started 07:50)
[image]

The solar and consumption peak is 22kW/h, what is exactly the total yield total:
[image]

I don't think, that this issue is from your code. I could imagine following:
After restart the OpenDTU at night, there would be a total Yield total of "0", because OpenDTU don't store this value.
In the morning it will get the "total yield total" of e.g. 22kW/h and set it for one cycle as a peak production.

Would you suggest to implement a workaround in your code (e.g. limiting this value) or to open this issue in the openDTU?

I think this problem would be really annoying, when you have a high "total yield total" in your hoymilles, and this would ruin your stats.

henne49, 2022-11-26:
Hi, I had this as well, when I added a second Script for testing with the multi inverter problem. Can you check if Deviceinstance has changed? 

[image]

Fideus, 2022-11-26:
I had two, but the other was 7 days old. I changed the ID 7 days ago, because there was problems during coding the last change.
So I think that this isn't the problem because it is 7 days ago. I delete it now, and test it again.

henne49, 2022-11-26:
I have now included to not write a zero for Energy, when no data is received, hope that fiexes it.

henne49, 2022-11-26:
But now you cannot see the power in VRM, will have a final look tomorrow. Need to drop off now.

henne49, 2022-11-26:
it seems to work, will test tonight with a reboot

henne49, 2022-11-27:
it worked for me, no spikes.

henne49, 2022-11-29:
It is fixed for me, let me know otherwise

Fideus, 2022-11-29:
Did you already updated code?

henne49, 2022-11-29:
Yes on Saturday

Fideus, 2022-12-01:
Hi, I tested, but there is still the issue.
what was the change in the code what should solve this?

henne49, 2022-12-01:
you are right, it seems I have not checked in the code change, will do first thing tomorrow

henne49, 2022-12-02:
These are the fixes: 

https://github.com/henne49/dbus-opendtu/blob/7e194c6b013e503d25e67f4afb9b4758260a74a5/dbus-opendtu.py#L195
https://github.com/henne49/dbus-opendtu/blob/7e194c6b013e503d25e67f4afb9b4758260a74a5/dbus-opendtu.py#L213

henne49, 2022-12-05:
> Hi henne, sorry for late reply.
>
> I would test this this night. But I assume, that this would give a problem with multiple inverter, because I don' think they all start exactly at the same time.
> I would let you know tommorow.

Fair point, the wait. I have an idea to multiply and only update when alle are online

Fideus, 2022-12-08:
Hello Henne,
the last two times it worked. But I still think that this could be a problem, if they don't start at the same time.
I restarted the esp32, and if I look in the victron, the last total yield is still visible.
So I assume, it is also possible to read this data and compare with the new data? e.g. something like this:

          if total > self._dbusservice[pre + '/Energy/Forward'] :
            self._dbusservice[pre + '/Energy/Forward'] = total

The result would be, that we only update the total power, if we are higher.

henne49, 2022-12-08:
I‘ll check this, but l‘m currently on work travel.

Will check as soon as I can.

henne49, 2022-12-10:
I had a look at it. 

when you reboot the opendtu it shows empty values. 
Therefore it should only update when new values are larger than the old ones. but I cannot check the old values from Venus, as it is volatile and needs initialization. So if there is only slight delay with restart of other hoymiles, I do not think Victron is making a problem. I had mine for 20 minutes initialized with a low value and it did not do a spike. 

I do not have any good and easy idea. The only good idea is to save somewhere the old value. but not sure if that is the best idea, ie in a file.

Fideus, 2022-12-11:
I tested it two days again, and it seems to work.
But I think, that when this problem occur e.g. in a few month, there would be the whole statistics ruined. Or is there a way to delete this spikes?
When we could not access the victron data, maybe we could store the old value in a global variable? Then we only could get this problem when the venus device and the esp32 restart over night.
Additional we could only update total yield when we have more than ~20W total power. This would give the inverter some time to Start in the Morning.

    if total_power > 20 :
       if total_yield > total_yield_old :
        total_yield_old = total_yield
        self._dbusservice['/Ac/Energy/Forward'] = total_yield

[image]

henne49, 2022-12-11:
there is no way to fix history data on victron I checked that. I have implemented your idea already yesterday, but was not happy as you need to cover multiple cases. 

We also need to cover a restart of the venus python script, because for initialization at script startup, the value is empty. Not when opendtu restarts, then I can rely on venus. A global variable or in a file would also help. I'll have a look, if you find a good global variable in venus, let me know.

henne49, 2022-12-29:
Hi,

I had no errors the last weeks, even when restarting. All additions and changes complicate the script and may introduce more bugs than it helps. What maybe helpful is to change the log interval large enough in vrm portal configuration 
[image]

what do you think?

closed 2023-01-16 by henne49