henne49/dbus-opendtu #9: feature request - one device per inverter

locke981, 2022-12-04:
Hi, 
I'm already testing your script, works very well, thanks for sharing it!
There is another fork on github which also makes it possible to view Hoymiles inverters in ve.connect/vrm portal.
https://github.com/stromibaer/OpenDTU-Victron

Biggest difference is that it is running in opendtu but not merged at the moment.
With this fork it is possible to see every HM as a seperate inverter. So you are also able to see the power based on the inverter, even if the are on the same phase. From what I see from your script this is not possible at the moment, right?

[image]

Is that something you can imagine to implement as well?
Thanks Stefan

henne49, 2022-12-06:
Hi,

I used the other script as well for some time, it used mqtt to create the inverters. 

As I only have one inverter I did not have the use case to show more than one. Also for Summary purpose one consolidating should be enought. 

BUT as always, I see your use case. so you would want to create a single device per Inverter. This is a major rework, but should be possible. 

Just a quick question how much OpenDTU versions are you using a single one? because if you have an opendtu per Inverter you could run my script multiple times. 

I'll have a look into this later.

locke981, 2022-12-06:
Hello,
I've only one DTU running. But at the moment I have 2 Cerbo's (Raspi's) running, one for production with the other solution, one for testing your script. The usecase for me is to get the performance of each individual inverter also in my reporting in the vrm portal and a Grafana. 
Thanks again for sharing your work!

henne49, 2022-12-29:
I had a look, this would require a major rework and I do not have the time. An easy workaround would be 2 DTUs and the script running twice. I leave this open, if this really required. As for long term statistics I use Home-assistant and influxdb, whis is maybe better suited as vrm. Just my 2 cents

locke981, 2022-12-29:
Ok, no problem, thanks for checking!

henne49, 2023-01-04:
Hi @locke981, I have now implemented in a test Branch to select inverters to show, so that it should be possible to do, what you wanted by running the script multiple times, : Branch: https://github.com/henne49/dbus-opendtu/tree/9-feature-request-one-device-per-inverter

Description and change here: https://github.com/henne49/dbus-opendtu/commit/a5d0843edba1777ca0853271f4b41eff2dcc17c7

locke981, 2023-01-04:
Cool, thanks, will test that with my 3 hoymiles, I'll come back to you

locke981, 2023-01-05:
Hi,
I have tested with the test branch. I now have 3 different folders for my 3 HM1500. I put an example below and also an current.log where I do see errors. It's kind of working but not 100%  :-)
West = L1, Ost = L1, Balkon = L3, nothing on L2

The inversters show up on my Raspi:

[image]

As you can see the show Power Output for every ijnverter is the same, I compared it on the real numbers on my DTU, its identical to the sum of all 3 inverters and changes accordingly with the numbers on the DTU.

Inside the devices (AC Gesamt is only differen between the 3 inverters because of the different point in time when I made the screenshots):
[image]
[image]
[image]

West is different and shows up more infos....

Config.ini of Balkon:

[DEFAULT]
# Fixed value 'OnPremise'
AccessType= OnPremise

# Time in minutes how often a status is added to the log-file current.log with >
SignOfLifeLog= 1

# Unique ID identifying the OpenDTU in Venus OS
Deviceinstance=8

# Name shown in Remote Console (e.g. name of pv inverter)
CustomName=HM1500-Balkon

# 0=AC input 1; 1=AC output; 2=AC output 2
AcPosition=1

# Number ob Inverter connected to the OpenDTU
NumberOfInverters=1

# Which DTU to be used ahoy, opendtu or template
DTU=opendtu

#For ESP8266 reduce polling intervall to reduce load
ESP8266PollingIntervall=10000

#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
#To keep current.log small use ERROR
Logging=INFO

[ONPREMISE]
#IP of Device to query
Host=10.0.0.75

#leave empty if no authentication is required
Username =
Password =

[TEMPLATE]
## Tasmota Example
CUST_SN = 12345678
CUST_API_PATH= tasmota/tasmota-power-only
CUST_POLLING = 2000
CUST_Total= StatusSNS/ENERGY/Total
CUST_Power= StatusSNS/ENERGY/Power
CUST_Voltage=  StatusSNS/ENERGY/Voltage
CUST_Current=  StatusSNS/ENERGY/Current

## Other Examples
## Shelly Gen 1 authenticated and unathenticated
#CUST_SN = 12345678
#CUST_API_PATH= status
#CUST_POLLING = 2000
#CUST_Total= meters/0/total
###Shelly1PM Multiplier Watt/min 0.000017
#CUST_Total_Mult = 0.000017
#CUST_Power= meters/0/power
#CUST_Power_Mult = 1
#CUST_Voltage= none
#CUST_Current= none

## Shelly Gen 2 - Only works unauthenticated!
#CUST_SN = 12345678
#CUST_API_PATH= rpc/Switch.GetStatus?id=0
#CUST_POLLING = 2000
#CUST_Total= aenergy/total
#CUST_Power= apower
#CUST_Voltage=  voltage
#CUST_Current=  current

# 1st inverter
[INVERTER0]
Phase=L3
Name=Balkon

# 2nd inverter
#[INVERTER1]
#Phase=L2
#Name=

# 3rd inverter
#[INVERTER2]
#Phase=L3
#Name=

# 4th inverter
#[INVERTER3]
#Phase=L1
#Name=

# 5th inverter
#[INVERTER4]
#Phase=L2
#Name=

# 6th inverter
#[INVERTER5]
#Phase=L3
#Name=

# 7th inverter
#[INVERTER6]
#Phase=L1
#Name=

# 8th inverter
#[INVERTER7]
#Phase=L2
#Name=

# 9th inverter
#[INVERTER8]
#Phase=L3
#Name=

# 10th inverter
#[INVERTER9]
#Phase=L1
#Name=

current.log:

2023-01-05 08:35:12,495 root INFO Last '/Ac/Power': 56.5
2023-01-05 08:35:12,495 root INFO --- End: sign of life ---
2023-01-05 08:36:12,518 root INFO --- Start: sign of life ---
2023-01-05 08:36:12,519 root INFO Last _update() call: 1672907772.5180278
2023-01-05 08:36:12,519 root INFO Last '/Ac/Power': 60.70000076
2023-01-05 08:36:12,520 root INFO --- End: sign of life ---
2023-01-05 08:37:12,627 root INFO --- Start: sign of life ---
2023-01-05 08:37:12,628 root INFO Last _update() call: 1672907832.627704
2023-01-05 08:37:12,629 root INFO Last '/Ac/Power': 61.79999924
2023-01-05 08:37:12,630 root INFO --- End: sign of life ---
2023-01-05 08:37:42,579 root INFO Converting response to JSON failed
2023-01-05 08:37:42,580 root CRITICAL Error at _update
Traceback (most recent call last):
  File "/data/dbus-opendtu-ost/dbus-opendtu.py", line 185, in _update
    meter_data = self._getOpenDTUData()
  File "/data/dbus-opendtu-ost/dbus-opendtu.py", line 171, in _getOpenDTUData
    raise ValueError("Converting response to JSON failed")
ValueError: Converting response to JSON failed
2023-01-05 08:38:12,628 root INFO --- Start: sign of life ---
2023-01-05 08:38:12,629 root INFO Last _update() call: 1672907892.5571833
2023-01-05 08:38:12,629 root INFO Last '/Ac/Power': 73.40000153
2023-01-05 08:38:12,630 root INFO --- End: sign of life ---
2023-01-05 08:39:12,669 root INFO --- Start: sign of life ---
2023-01-05 08:39:12,670 root INFO Last _update() call: 1672907952.6693404
2023-01-05 08:39:12,674 root INFO Last '/Ac/Power': 74.70000458
2023-01-05 08:39:12,675 root INFO --- End: sign of life ---
2023-01-05 08:40:12,741 root INFO --- Start: sign of life ---
2023-01-05 08:40:12,744 root INFO Last _update() call: 1672908012.7413065
2023-01-05 08:40:12,745 root INFO Last '/Ac/Power': 67.59999847
2023-01-05 08:40:12,745 root INFO --- End: sign of life ---
2023-01-05 08:41:12,742 root INFO --- Start: sign of life ---
2023-01-05 08:41:12,742 root INFO Last _update() call: 1672908072.7096546
2023-01-05 08:41:12,743 root INFO Last '/Ac/Power': 68.09999847
2023-01-05 08:41:12,743 root INFO --- End: sign of life ---
2023-01-05 08:41:20,121 root CRITICAL Error at _update
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/urllib3/connection.py", line 159, in _>
    conn = connection.create_connection(
  File "/usr/lib/python3.8/site-packages/urllib3/util/connection.py", line 84, >
    raise err
  File "/usr/lib/python3.8/site-packages/urllib3/util/connection.py", line 74, >
    sock.connect(sa)
socket.timeout: timed out

locke981, 2023-01-05:
If you need something else just let me know...Thanks for sharing your work!

henne49, 2023-01-05:
Hi, I guess I see my problem, the inverter have an order on openDTU
1: West = L1, 
2: Ost = L1, 
3: Balkon = L3

So in the config file for West it is inverter0 or first
Name = West
inverter1
Name = None
inverter2 
Name= None

for Ost it would be
inverter0
Name = None
inverter1
Name = Ost
inverter 2
Name = None

For Balkon it would be
inverter0
Name = None
inverter1
Name = None
inverter2
Name = Balkon

I have to fix it better, but give it a try, so that the config just picks the inverter that is needed. 

Please try, if that works, I have to get my head around to fix this better

henne49, 2023-01-05:
and it would be helpful to share full json openDTU file  and attach the config.ini in full text.

thank you for testing

locke981, 2023-01-05:
I changed the config.ini's for Balkon and Ost and rebooted:

[image]
[image]
[image]
[image]

Where do I find the full json openDTU file?

Attached the config.ini and current.log from all 3 folders:
[OpenDTU Test.zip](https://github.com/henne49/dbus-opendtu/files/10351163/OpenDTU.Test.zip)

henne49, 2023-01-05:
http://X.X.X.X/api/livedata/status

would be the URL, if you share that, then I can fix it. Thank you again for helping here

locke981, 2023-01-05:
Hope that helps!
[DTU status.txt](https://github.com/henne49/dbus-opendtu/files/10351391/DTU.status.txt)

henne49, 2023-01-06:
It works now, ahoy would have worked out of the box, but opendtu was reading the summary still.

[image]

Code is updated

locke981, 2023-01-07:
Hi, I tested it, it seems to show only the results of "West" for all 3 inverters.
[image]

henne49, 2023-01-07:
[OpenDTU.Test.zip](https://github.com/henne49/dbus-opendtu/files/10366708/OpenDTU.Test.zip)

Your configs were not correct, sorry that I did not check that, now try with one the ones I have now uploaded. 

Number of Inverters must be 3 as you have 3 to check.

then you specifiy the name, if the script has a match it will use it or if it is empty, if there is no match then the value is ignored, therefore I have put non, it could also be ignore.

Please try again and let me know. I hope to make it easier in the future, but it should make it work for you, already like it is.

locke981, 2023-01-07:
No Problem, thanks for the update, I corrected my ini's now, it already looks better because I now see the correct Voltage for every inverter on the correct Phase. It's already dark ouside so no real testing is possible, but I will give you an update tomorrow.
Thanks you!

locke981, 2023-01-08:
Looks good now, I think everything is working as expected!
Thanks again for sharing your work!

[image]

henne49, 2023-01-08:
Ok, as we now know it works, I will check, how I can make it work in one script, but give me some time, to check this out!

dersch81, 2023-01-09:
I really like your work on this! i have 5 inverters and more are coming.

henne49, 2023-01-09:
@dersch81 it works with up to 10 already, just being summarised as one. Or you can have the script run multiple times, where you can also group inverted. so 3xtimes the script having 2 inverters each. 

can you guys explain why you want to have so much different devices? what is the logic behind it? intend of a summary?

dersch81, 2023-01-09:
> can you guys explain why you want to have so much different devices? what is the logic behind it? instead of a summary?

Statistics, for sure the total is the most important. But i want to see the energy and watt of the different directions and places in detail. Of course OpenDTU shows me and via MQTT i have everything in HA. But the Victron is my main energy monitor and therefore i want to see the details there, too.

locke981, 2023-01-10:
> can you guys explain why you want to have so much different devices? what is the logic behind it? instead of a summary?

Same for me, I just want to have individual statistics in one place, which is the victron vrm portal, for every inverter.

henne49, 2023-01-11:
Good news: there is testing branch, that should work in a single script, please test. It is working for me: https://github.com/henne49/dbus-opendtu/tree/Multi-Inverter

Based on the feedback, I will then merge into main.

locke981, 2023-01-13:
Hi, I have tested it, do far it looks good.
One question: Do we need the generic definition of DeviceID and AcPosition in config.ini, when it's defined for the inverter itself anyway?

# Unique ID identifying the OpenDTU in Venus OS
Deviceinstance=6

# 0=AC input 1; 1=AC output; 2=AC output 2
AcPosition=1

### Only needed for OpenDTU and ahoy
# 1st inverter
[INVERTER0]
Phase=L1
DeviceInstance=6
AcPosition=1

henne49, 2023-01-13:
Correct that can possibly also be moved, and cleaned up, we will need to do this, as the overall rehaul, will have a breaking impact.

we only use the upper deviceinstance and acposition for the generic Device.

closed 2023-01-16 by henne49