henne49/dbus-opendtu #10: Error (KeyError: 'inverters')

Sefina-DS, 2022-12-23:
Hi @ all,

this Project is nice but i become onley errors 
in the Logs come :

2022-12-23 18:39:06,820 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 272, in main
    pvac_output = DbusOpenDTUService(
  File "/data/dbus-opendtu/dbus-opendtu.py", line 54, in __init__
    self._dbusservice.add_path('/Serial', self._getOpenDTUSerial())
  File "/data/dbus-opendtu/dbus-opendtu.py", line 91, in _getOpenDTUSerial
    if not meter_data['inverters'][0]['serial']:
KeyError: 'inverters'

can anyone help me ?

Sefina-DS, 2022-12-23:
I have a AhoyDTU with a ESP32
I have change the : 
# Which DTU to be used ahoy or opendtu
DTU=ahoy

but after this i have this problem :
2022-12-23 19:09:31,530 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 272, in main
    pvac_output = DbusOpenDTUService(
  File "/data/dbus-opendtu/dbus-opendtu.py", line 54, in __init__
    self._dbusservice.add_path('/Serial', self._getOpenDTUSerial())
  File "/data/dbus-opendtu/dbus-opendtu.py", line 96, in _getOpenDTUSerial
    gobject.timeout_add(polling_interval, self._update)
UnboundLocalError: local variable 'polling_interval' referenced before assignment

henne49, 2022-12-29:
you have to use DTU=ahoy, as you did, as I do not have an ahoy, I as unable to check, let me have a look, where the problem comes from.

henne49, 2022-12-29:
I have found the problem, please test and let me know.

closed 2023-01-02 by henne49