henne49/dbus-opendtu #12: Problems with python errors on a Victron Cerbo

stxShadow, 2022-12-29:
Hi ! 

I've installed you script the first time today on a brand new Cerbo GX. I use OpenDTU and edited the config files as intended. 
After restarting the cerbo i allways get: 

root@einstein:~# ps | grep open
  823 root      1600 S    readproctitle service errors: ...ock   File "/data/dbus-opendtu/dbus-opendtu.py", line 90     polling_interval = 250     ^ IndentationError: expected an indented block   File "/data/d

I saw, that you did a commit 3 hours ago and tried the older version. Works like it should.

Thanks a lot !

Jens

henne49, 2022-12-29:
One should never do code changes on the go, I have changed, can you please test?

stxShadow, 2022-12-29:
looks good now ! Thanks for the quick response.
One other question: i use lots of Tasmota Devices in my PV System. Every Device also has a Status JSON (very simple).

{"StatusSNS":{"Time":"2022-12-29T14:46:01","ENERGY":{"TotalStartTime":"2022-12-18T11:05:28","Total":0.338,"Yesterday":0.031,"Today":0.009,"Power": 5,"ApparentPower":13,"ReactivePower":13,"Factor":0.35,"Voltage":231,"Current":0.058}}}

Do you think it is possible to adapt your source code to use this values ? If yes: i have to learn python :dagger: 
Thanks again ! 
Jens

tkuehne71, 2022-12-29:
Thanks for the prompt fix. I ran into the same issue. It seems fixed now.

henne49, 2022-12-29:
> looks good now ! Thanks for the quick response. One other question: i use lots of Tasmota Devices in my PV System. Every Device also has a Status JSON (very simple).
> 
> {"StatusSNS":{"Time":"2022-12-29T14:46:01","ENERGY":{"TotalStartTime":"2022-12-18T11:05:28","Total":0.338,"Yesterday":0.031,"Today":0.009,"Power": 5,"ApparentPower":13,"ReactivePower":13,"Factor":0.35,"Voltage":231,"Current":0.058}}}
> 
> Do you think it is possible to adapt your source code to use this values ? If yes: i have to learn python 🗡️ Thanks again ! Jens

It is possible only that you need to create a gen metr instead of a PV Inverter in dbus (https://github.com/victronenergy/venus/wiki/dbus#grid-and-genset-meter). Question is why? I use home-assistant / influxdb & grafana for all detailed recording.

stxShadow, 2022-12-29:
> It is possible only that you need to create a gen metr instead of a PV Inverter in dbus (https://github.com/victronenergy/venus/wiki/dbus#grid-and-genset-meter). Question is why? I use home-assistant / influxdb & grafana for all detailed recording.

No .... the Sonoff Devices (with Tasmota firmware) monitor a variety of micron inverters which are not able to communicate by itselfs. I would like to see them also in the Victron VRM (like Hoymiles) but there is no dbus driver for Tasmota jet.
Thanks 
Jens

henne49, 2022-12-29:
that would work, we could enhance my script to also retrieve tasmota values. Can you share an example json and the URL?

closed 2022-12-29 by henne49