henne49/dbus-opendtu #13: Feature request: add Tasmota Devices

stxShadow, 2022-12-30:
Hi ! 
you asked for some informations yesterday to add tasmota support.

The URL is: http://x.x.x.x/cm?cmnd=status%208 for power informations 
Complete system informations URL is http://x.x.x.x/cm?cmnd=status%2000

I've attached you  to appropriate json outputs of both cases.

Thanks a lot for your work an effort ! 

Jens

[tasmota-json.zip](https://github.com/henne49/dbus-opendtu/files/10323782/tasmota-json.zip)

henne49, 2022-12-30:
I will create a branch and try to add the tasmota, but you would then need to install 2 times the script, once for opendtu and for tasmota. 

later on, we could review to add multiple PV Inverter which is anyway a request, but for that I would need a clever solution and a major rework of the overall script. 

would that be ok as a start for you?

stxShadow, 2022-12-30:
yes ... that will work for me. I own a VenusOS Pi and a Victron Cerbo just for testing purposes (for my YT channel).

Thanks a lot

Jens

henne49, 2022-12-30:
If you install it in a different directory, it should be able to run from the same system no need for a second pi. 

@stxShadow did you reference the opendtu script in this video: https://youtu.be/PpjCz33pGkk ? 

Thank you for your channel, I learned a lot to setup my victron and pylontechs. The name is easy. Spitzname plus Trikotnummer

stxShadow, 2022-12-30:
Yes :D That is my latest video ...... i really had no clue what was ment by "henny49". Thank you for clearing that.
If the tasmota integration works, it will be worth a video too. 

Jens

PS: i know that we can speak german too .... but this is an international plattform :+1:

henne49, 2023-01-02:
Hi @stxShadow, I have added generic single phase Support for JSON and REST Based devices, even if they have password protection. The Example is based on Tasmota as you had provided, but is generic enough to even fit others. 

I will have to include more checks, if Current or Voltage is not available. 

Please test and let me know if it works: https://github.com/henne49/dbus-opendtu/tree/Generic-REST-Device

henne49, 2023-01-04:
@stxShadow I have added some more interesting stuff, maybe you are able to test as well

 Added Support for Generic Rest Devices

Tested Examples:
* Tasmota unathenticated
* Shelly 1 PM authenticated/unauthenticated
* Shelly Plus 1 PM unathenticated

All configuration is done via config.ini examples are uncommented

stxShadow, 2023-01-05:
Tasmota unauthenticated: works
Shelly 1 PM: will test it the next days
Shelly Plus 1 PM: sorry .... not owning one

Jens

henne49, 2023-01-06:
The changes are now merged in main and documentation is updated

stxShadow, 2023-01-09:
Shelly PM1 : works 

Fronius Icon: seem you are right. Aus soon as a fronius inverter is present, all inverters get the fronius icon.

closed 2023-01-16 by henne49