henne49/dbus-opendtu #15: Inverter 1 auf AC-in / Inverter 2 auf AC-Out

New-three, 2023-01-01:
Ich habe zwei WR angebunden. Einer ist auf AC in und der andere auf AC out angeschlossen.
Ist es möglich dies in der Config einzustellen? Oder gilt das immer für alle?

henne49, 2023-01-01:
the Script just generates one device in dbus, therefore it can be only in or out. if you want to split you would currently need to split into 2 opendtu and running the script twice, but honestly I do not see any benefit. to have multiple.

New-three, 2023-01-01:
Thank you for your prompt reply.
Then I will buy a second open dtu.

the following would not be possible?
script1 AC in - Config only inverter 0
script 2 AC out - Config only inverter 1
enter the same IP in both scripts, i.e. use the same DTU

henne49, 2023-01-01:
your idea is not bad, I could maybe just parse for a SN. The challenge I have is my time is limited to rewrite the whole code, to create multiple inverter instead of just summing up all inverters of opendtu and present as one. what you also could do is tot have them virtually on different phases for viewing purpose. 

For now buying a second opendtu is the only way, without me fixing it. so if you do not want to wait, then go andientest another 15€.

henne49, 2023-01-04:
Hi, I have now implemented in a test Branch to select inverters to show, so that it should be possible to do, what you wanted by running the script twice: Branch: https://github.com/henne49/dbus-opendtu/tree/9-feature-request-one-device-per-inverter

Description and change here: https://github.com/henne49/dbus-opendtu/commit/a5d0843edba1777ca0853271f4b41eff2dcc17c7

New-three, 2023-01-07:
Wow thanks...
But I don't understand exactly what I have to do to test it.
I installs two from this link?
https://github.com/henne49/dbus-opendtu/tree/9-feature-request-one-device-per-inverter
And in each I enter the same IP from Ahoy and for the first installation inverter 0 Yes and for the second other inverters 1 Yes?
can you give me an example?

Am I right that Tasmota/Shelly Gen1/Shelly Gen2 are also supported?

henne49, 2023-01-07:
Correct I have added generic REST template options and tested with the tasmota and shelly.

here is an example below for 3 inverters in opendtu, but will work the same with ahoy https://github.com/henne49/dbus-opendtu/issues/9

New-three, 2023-01-08:
Great work. However, I will only be able to test it in a few days.

How will this work with Tasmota?
Inverter0,1,2....ignore?
DTU=template
IP=Tasmota IP

Do I have to change something in the template?
TEMPLATE]
## Tasmota Example
CUST_SN = 12345678
CUST_API_PATH= tasmota/tasmota-power-only
CUST_POLLING = 2000
CUST_Total= StatusSNS/ENERGY/Total
CUST_Power= StatusSNS/ENERGY/Power
CUST_Voltage= StatusSNS/ENERGY/Voltage
CUST_Current= StatusSNS/ENERGY/Current

henne49, 2023-01-09:
I have update the config.ini, which should now work with Tasmota Default URL: https://github.com/henne49/dbus-opendtu/blob/main/config.ini 

DTU=template 

just the IP and maybe username and password should be required. Anything in inverter will be ignored.

New-three, 2023-01-09:
It works great :)

Today i try Shelly Gen1 and Gen2.

Now I have everything I need. Many thanks for the great work.

henne49, 2023-01-11:
Good news: there is testing branch, that should work in a single script, please test. It is working for me: https://github.com/henne49/dbus-opendtu/tree/Multi-Inverter

Based on the feedback, I will then merge into main.

closed 2023-01-16 by henne49

New-three, 2023-01-17:
I have now finally tried to display an inverter on AC In and another on AC Out.
I had to realize that only one instance is needed for this.
Good work.
Why is this entry necessary at the beginning of the config?
**# Unique ID identifying the OpenDTU in Venus OS
Device instance=43**

Below each inverter gets its own ID.
**### Only needed for OpenDTU and ahoy
# 1st inverter
[INVERTER0]
Phase=L1
DeviceInstance=34
AcPosition=1**

Is it possible to add a Shelly or/and Tastmota in the same config? Or do I need my own config for more Tasmota/Shelly?

henne49, 2023-01-17:
This will be removed in further versions it is used for Template device.

Next step up to 10 template devices and after that a mix anf match of opendtu or ahoy and template devices.

But that‘ll take some time. I have a couple of busy work weeks ahead