henne49/dbus-opendtu #16: Error at Main Ahoy 0.5.66

New-three, 2023-01-02:
I updated my Ahoy DTU to 0.5.66. Since then I can no longer get a connection in VenusOS. Is this actually due to the Ahoy version or is the error somewhere else?

```
2023-01-02 13:39:38,755 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu-1/dbus-opendtu.py", line 274, in main
    pvac_output = DbusOpenDTUService(
  File "/data/dbus-opendtu-1/dbus-opendtu.py", line 54, in __init__
    self._dbusservice.add_path('/Serial', self._getOpenDTUSerial())
  File "/data/dbus-opendtu-1/dbus-opendtu.py", line 86, in _getOpenDTUSerial
    if meter_data['system']['esp_type']=='ESP8266':
KeyError: 'system'
2023-01-02 13:39:41,760 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu-1/dbus-opendtu.py", line 274, in main
    pvac_output = DbusOpenDTUService(
  File "/data/dbus-opendtu-1/dbus-opendtu.py", line 54, in __init__
    self._dbusservice.add_path('/Serial', self._getOpenDTUSerial())
  File "/data/dbus-opendtu-1/dbus-opendtu.py", line 86, in _getOpenDTUSerial
    if meter_data['system']['esp_type']=='ESP8266':
KeyError: 'system'
2023-01-02 13:39:45,76 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu-1/dbus-opendtu.py", line 274, in main
    pvac_output = DbusOpenDTUService(
  File "/data/dbus-opendtu-1/dbus-opendtu.py", line 54, in __init__
    self._dbusservice.add_path('/Serial', self._getOpenDTUSerial())
  File "/data/dbus-opendtu-1/dbus-opendtu.py", line 86, in _getOpenDTUSerial
    if meter_data['system']['esp_type']=='ESP8266':
KeyError: 'system'
```

henne49, 2023-01-02:
it seems they changed the API. can you share the json? would you know how to do this?

New-three, 2023-01-02:
Export json in web interface?
 Settings/Download your settings (JSON file)](http://192.168.10.182/get_setup) (only saved values)

henne49, 2023-01-02:
They changed the API, it is fixed now and supports old and new API.

New-three, 2023-01-02:
Many Thanks. Connection works again.

I saw that the config file now supports 10 inverters?

henne49, 2023-01-02:
Correct, as this is also the limit in OpenDTU. Will close the issue

henne49, 2023-01-02:
changed fixed it

closed 2023-01-02 by henne49