henne49/dbus-opendtu #19: JSON from OpenDTU not parsed / different JSON content

lexxman123, 2023-01-09:
I am running OpenDTU (SDK Version  v4.4.2, Config Version 0.1.22). Just one inverter is connected to OpenDTU.
JSON returned by OpenDTU cannot be parsed by dbus-opendtu ("inverters" array is missing, maybe because just one inverter is configured).

Can you please enhance the script to also work with such a setup (without "inverters" array)?
In addition, JSON does not contain
['total']['Power']['v'] 
['total']['YieldTotal']['v']

Example of JSON returned by my OpenDTU is attached to this issue. 

[opendtu.json.txt](https://github.com/henne49/dbus-opendtu/files/10372949/opendtu.json.txt)

Thank you.

henne49, 2023-01-09:
Hi, please try upgrading openDTU, as I run also only 1 inverter and it is working for me with a newer Config Version of 0.1.23.

lexxman123, 2023-01-09:
Thank you for the quick response. After updating OpenDTU to the current release fixed the issue, all is working now. 
Thank you again.

closed 2023-01-09 by lexxman123