henne49/dbus-opendtu #21: 2W power at night

dsteinkopf, 2023-01-10:
At the moment the script is sending 2.2W to venus/VRM because of this JSON result from Ahoy:
```
{
  "menu": {
    "name": [
...
    ],
    "link": [
...
    ],
    "trgt": [
...
    ]
  },
  "generic": {
    "version": "0.5.66",
    "build": "f8fe044",
    "wifi_rssi": -64,
    "ts_uptime": 30499,
    "esp_type": "ESP8266"
  },
  "inverter": [
    {
      "enabled": true,
      "name": "hoymiles1",
      "channels": 1,
      "power_limit_read": 100,
      "last_alarm": "Inverter start",
      "ts_last_success": 1673365160,
      "ch": [
        [
          231.3,
          0.01,
          2.2,
          49.99,
          1.005,
          8,
          90.185,
          327,
          2.3,
          95.652,
          0
        ],
        [
          30,
          0.08,
          2.3,
          327,
          90.185,
          2.3
        ]
      ],
      "ch_names": [
        "AC",
        "einzel"
      ]
    },
    {
      "enabled": true,
      "name": "hoymiles2",
      "channels": 4,
      "power_limit_read": 100,
      "last_alarm": "Inverter start",
      "ts_last_success": 1673365165,
      "ch": [
        [
          230.7,
          0,
          0,
          49.99,
          0,
          6.4,
          1.153,
          631,
          3.5,
          0,
          20.7
        ],
        [
          31.7,
          0.02,
          0.6,
          193,
          0.362,
          0.16
        ],
        [
          31.7,
          0.01,
          0.5,
          3,
          0.007,
          0
        ],
        [
          31.9,
          0.04,
          1.2,
          199,
          0.37,
          0.32
        ],
        [
          31.9,
          0.04,
          1.2,
          236,
          0.414,
          0.32
        ]
      ],
      "ch_names": [
        "AC",
        "M1",
        "no",
        "M3",
        "M4"
      ]
    }
  ],
  "refresh_interval": 5,
  "ch0_fld_units": [
    "V",
    "A",
    "W",
    "Hz",
    "",
    "°C",
    "kWh",
    "Wh",
    "W",
    "%",
    "var"
  ],
  "ch0_fld_names": [
    "U_AC",
    "I_AC",
    "P_AC",
    "F_AC",
    "PF_AC",
    "Temp",
    "YieldTotal",
    "YieldDay",
    "P_DC",
    "Efficiency",
    "Q_AC"
  ],
  "fld_units": [
    "V",
    "A",
    "W",
    "Wh",
    "kWh",
    "%"
  ],
  "fld_names": [
    "U_DC",
    "I_DC",
    "P_DC",
    "YieldDay",
    "YieldTotal",
    "Irradiation"
  ]
}
```

henne49, 2023-01-10:
that makes sense, but the hoymiles is also simply wrong, let‘s include this a feature request.

Very well spotted

henne49, 2023-01-10:
We should then send none, because we gave to update. Ob the other hand it has no impact on energy reporting. It is rather cosmetic

dsteinkopf, 2023-01-10:
I've prepared a branch in my fork of this repo: https://github.com/dsteinkopf/dbus-opendtu/tree/check_ts_last_success
... will create a PR after the values tomorrow arrive correctly.

dsteinkopf, 2023-01-10:
> that makes sense, but the hoymiles is also simply wrong, let‘s include this a feature request.

I am not sure if this should be seen as a bug in Ahoy. The data is correct, just a bit old. But the age (timestamp) is in the data.

dsteinkopf, 2023-01-10:
Yes, it IS a bug in ahoy - at least at some other place: Latest value sent via MQTT also is `2.2`.

dsteinkopf, 2023-01-11:
I've just created an Ahoy issue: lumapu/ahoy#575

PR #22 (2023-01-19): Check ts_last_success
Check ts_last_success to prevent obsolete values from being sent.

Fixes #21
closed 2023-01-19 by henne49