henne49/dbus-opendtu #25: Return "power = 0" in case of not producing inverter

speedy1980, 2023-01-29:
Problem: In Victron the last producing power is shown in case of not producing inverter. Should be reported as 0 watt in that case. We should change line 350 in dbus-opendtu.py like this:
if meter_data['inverters'][self.pvinverternumber]['producing']:
power = 0
else:
power = meter_data['inverters'][self.pvinverternumber]['0']['Power']['v']

I'm not a python programmer - so please check syntax.

speedy1980, 2023-01-30:
Another solution:
power = meter_data['inverters'][self.pvinverternumber]['0']['Power']['v'] * meter_data['inverters'][self.pvinverternumber]['reachable']

dsteinkopf, 2023-01-31:
I assume you are using a dtuvariant == OpenDTU (not Ahoy or template).
As I am not familial with openDTU, I have to ask: 
- What are the values that are sent in `producing` for true and false? (Do we need some conversion to bool here?)
- What is the semantic of producing==false? -> Only the power value is "invalid"? Or other values, too? In the latter case I'd prefer to put the suggested check in `_isDataUpToDate`.

speedy1980, 2023-01-31:
Oh sorry, yes opendtu is correct. "producing" is true or false. "reachable" is 0 or 1.
Problem is: Inverter stop's sending data cause of no power from solar panels. In opendtu and victron it looks like the inverter is "producing" the whole night with the last power value.

dsteinkopf, 2023-01-31:
Yes, we had the same problem using Ahoy (of course...).
Maybe we should do it this way:

- if not reachable -> then consider all data invalid
- if not producing -> then use 0 W power

ok?

speedy1980, 2023-01-31:
Sounds good for me.

henne49, 2023-01-31:
> Yes, we had the same problem using Ahoy (of course...). Maybe we should do it this way:
> 
>     * if not reachable -> then consider all data invalid
> 
>     * if not producing -> then use 0 W power
> 
> 
> ok?
> 
> @henne49 what do you think?

I think this makes sense, I was not aware of these fields anymore. So we should add this additional checking

dsteinkopf, 2023-02-02:
I will create a PR

henne49, 2023-02-03:
Thank you Dirk

dsteinkopf, 2023-02-09:
See #26  (sorry for the delay)

dsteinkopf, 2023-02-13:
@speedy1980 : #26 is merged. Does it work for you?
(I'll close this ticket now - let's reopen or create a new one if needed.)

closed 2023-02-13 by dsteinkopf