henne49/dbus-opendtu #33: [Multi-Template] Problem: openDTU no values ​​for VenusOS

e-i-k-e, 2023-02-19:
Hello,
The Tasmota and Shelly devices transmit values ​​to the Venus OS. However, I don't get any values ​​from the opendtu WR (yellow marking). There are values ​​in the opendtu.

Do I have an error in the config?

[image]

[JSONE open dtu.log](https://github.com/henne49/dbus-opendtu/files/10777557/JSONE.open.dtu.log)
[current.log](https://github.com/henne49/dbus-opendtu/files/10777561/current.log)

henne49, 2023-02-19:
this is all fine, looking at your dtu json, the inverters are not producing and that is why we do not update any values check tomorrow, when sun is there and they are producing, then all should be fine.

to not screw up reporting, we are not updating venus and vrm when we are not sure we get valid data and the hoymiles is only online when it is producing. when you reboot the esp all data is gone in the night which breaks things, if we do not handle it how we handle it at night.

Glad that the template things work. also check the main branch, we have now merged and that you have the latest code.

e-i-k-e, 2023-02-19:
One of the inverters (inverter 1) definitely produced something today and these values ​​were also visible in the openDTU, but not in Venus. Not even after a restart.

I'm using the current multi branch.
So I can switch to the main show now?

But I'm also glad that it works and that you brought this great project to life! :)

henne49, 2023-02-19:
yes switch to main, we merged at 17:00 today. 

please check tomorrow again and let me know if opendtu shows values and this script does not.

the files you provided today where after dusk, therefor this seems normal. if it is different tomorrow I need full logs when it happen and inverter produce.

e-i-k-e, 2023-02-19:
That's correct, I'll give you feedback on that tomorrow.
I stay on the current multi branch and switch to the main branch with the next update.

henne49, 2023-02-20:
reviewieing my installation and both seem good ans show values. as my Hoymiles are producing.

dsteinkopf, 2023-02-20:
In your current.log I see the following messages:
```
urllib3.exceptions.ConnectTimeoutError: (<urllib3.connection.HTTPConnection object at 0x75af84d8>, 'Connection to 192.168.3.48 timed out. (connect timeout=2.5)')
...
requests.exceptions.ConnectionError: HTTPConnectionPool(host='192.168.3.48', port=80): Max retries exceeded with url: /status (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x75af82f8>: Failed to establish a new connection: [Errno 113] No route to host'))
```

Your Venus apparently is not able to contact your DTU. You should check IP addresses, connections, routing etc. Try `ping 192.168.3.48` and `curl 192.168.3.48` from your Venus command line.

e-i-k-e, 2023-02-20:
192.168.3.48 is the IP of a Shelly.  192.168.3.38 is the one for the DTU.

  I still don't see any values ​​from the two inverters that are connected via openDTU.  The first inverter is currently not connected to the grid and is therefore not sending any data.  but that shouldn't have any effect from the second inverter functions, should it? 
 I won't be able to provide a log file until this evening.

henne49, 2023-02-20:
We need a log during the day please, when inverter is producing, also set the logging to Debug and give us the full current.log file please. Thanks

But  I still have the feeling there is also a connectivity issue somewhere

dersch81, 2023-02-20:
I have the same problem. No one of my inverters is showing any values. It happend after updating OpenDTU v4.4.3 https://github.com/tbnobody/OpenDTU/commits/6a6165e

The current log:
`023-02-20 12:02:13,258 root CRITICAL Error at _update
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 290, in _update
    power += meter_data['inverters'][actual_inverter]['0']['Power']['v']
KeyError: '0'
2023-02-20 12:02:13,816 root CRITICAL Error at _update
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 290, in _update
    power += meter_data['inverters'][actual_inverter]['0']['Power']['v']
KeyError: '0'
2023-02-20 12:02:14,240 root CRITICAL Error at _update
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 290, in _update
    power += meter_data['inverters'][actual_inverter]['0']['Power']['v']
KeyError: '0'
2023-02-20 12:02:14,588 root CRITICAL Error at _update
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 290, in _update
    power += meter_data['inverters'][actual_inverter]['0']['Power']['v']
KeyError: '0'
2023-02-20 12:02:14,928 root CRITICAL Error at _update
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 290, in _update
    power += meter_data['inverters'][actual_inverter]['0']['Power']['v']
KeyError: '0'
2023-02-20 12:02:15,326 root CRITICAL Error at _update
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 290, in _update
    power += meter_data['inverters'][actual_inverter]['0']['Power']['v']
KeyError: '0'
2023-02-20 12:02:18,251 root CRITICAL Error at _update
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 290, in _update
    power += meter_data['inverters'][actual_inverter]['0']['Power']['v']
KeyError: '0'
2023-02-20 12:02:18,495 root CRITICAL Error at _update
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 290, in _update
    power += meter_data['inverters'][actual_inverter]['0']['Power']['v']
KeyError: '0'
2023-02-20 12:02:18,803 root CRITICAL Error at _update
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 290, in _update
    power += meter_data['inverters'][actual_inverter]['0']['Power']['v']
KeyError: '0'
`

henne49, 2023-02-20:
This is a different problem, please install the latest code of dbus-opendtu, which addresses a change in the API of opendtu

dersch81, 2023-02-20:
Yes ok, just did the update of dbus-OpenDTU in a wrong way and still had the old release running. Could you kindly make a short howto  of the best practice to upgrade?

e-i-k-e, 2023-02-20:
Attached is the log and config file.

[current.log](https://github.com/henne49/dbus-opendtu/files/10784518/current.log)
[config.log](https://github.com/henne49/dbus-opendtu/files/10784519/config.log)

henne49, 2023-02-20:
Hey 

> Yes ok, just did the update of dbus-OpenDTU in a wrong way and still had the old release running. Could you kindly make a short howto of the best practice to upgrade?

ist ist essentially like a new install to be 100% sure

henne49, 2023-02-20:
> Attached is the log and config file.
> 
> [current.log](https://github.com/henne49/dbus-opendtu/files/10784518/current.log) [config.log](https://github.com/henne49/dbus-opendtu/files/10784519/config.log) [JSON DTU.log](https://github.com/henne49/dbus-opendtu/files/10784536/JSON.DTU.log)

Found it, you are not running the latest OpenDTU Version please upgrade OpenDTU to the latest version, they have changed the API.

we will put something in the code, that you have to upgrade. Then it should work.

closed 2023-02-20 by henne49

e-i-k-e, 2023-02-20:
Short feedback.
Values ​​are transmitted after an update of my openDTU.