henne49/dbus-opendtu #35: zero Values ​​different openDTU/Template

e-i-k-e, 2023-02-20:
Just a little thing. :)

The zero values ​​are displayed differently in Venus.
openDTU: 0W
Template (Tasmota/Shelly) 0.0W

[image]

henne49, 2023-02-20:
at night it should be ---

We take the numbers as we get them from the devices and do no manipulation, this is easier and less error prone. It is only cosmetic. Also OpenDTU shows 1 number after the dot as well if larger than 0. 

So this should all be good.

[image]

e-i-k-e, 2023-02-20:
Thanks for the info.

closed 2023-02-20 by e-i-k-e