henne49/dbus-opendtu #36: Error after updateing Cerbo to 2.93

bear30, 2023-02-21:
Hello,

after updating the cerbo from 2.92 to 2.93 my HM-1500 disapears in the victron remote console. By the way this victron update change my ssh password to ???. I think this is a victron mistake. Ok, i restore the original ssh password. Nothing happens. Then i reinstall dbus-opendtu to the newest version, but nothing changes. In dbus spy i can see the dbus-device for 1 sec, the it disapears for 5 sec, ... and so again and again. Then i look into the log file and see this error messages from screenshot. Did you have any idea to this problem?

[image]

Thank you for your work!!!

bear30, 2023-02-21:
[image]

henne49, 2023-02-21:
Hi, this seems to be a very old config and the deviceinstance should be rather something like 34.

Also your code seems very old, please upgrade to the new release code from yesterday. Repeat the installation steps.

and yes with every update the ssh password is overwritten, why you should use ssh key authentication: https://www.victronenergy.com/live/ccgx:root_access#installing_ssh_keys

bear30, 2023-02-21:
Hello, thank you for your fast response. 

By the way my installation was 5 weeks old... 

Now i delete first the directory from cerbo and reinstall and configure dbus-optendtu. Now it works. 

Thank you!

henne49, 2023-02-21:
Glad to hear, well the project is only 10 weeks old, so in that it is old, we did some huge in the meanwhile.

closed 2023-02-21 by henne49