henne49/dbus-opendtu #39: no mor data in venus os after update bin file

peregrines2, 2023-02-24:
This is such a great project! Thank you for all the hard work!

Unfortunately, after updating via opendtu-*.bin both my hoymiles inverts dont show any more data in venus os ( V 3.00-18).
Data is shown in opendtu webinterface, and both hoymiles show as "connected" in venus os.
Can anybody help me out?

henne49, 2023-02-24:
Hey, please update the script to the latest version: https://github.com/henne49/dbus-opendtu/releases/tag/v2023-02-21

What version are you running for openDTU, they changed the API.

peregrines2, 2023-02-24:
Thank You for your help henne49!
I did the update, but it seems the script is crashing still:
[image]
 My RC.local has been working with no problems so far!

#!/bin/bash
sh /data/etc/dbus-serialbattery/reinstalllocal.sh
/data/dbus-opendtu/install.sh
sleep 20 &&  python /data/dbus-fronius-smartmeter/dbus-fronius-smartmeter.py

peregrines2, 2023-02-24:
[DEFAULT]
# Time in minutes how often a status is added to the log-file current.log with log-level INFO
SignOfLifeLog= 1

# Number ob Template Inverter to query
NumberOfTemplates=2

# Which DTU to be used ahoy, opendtu, template
DTU=opendtu

# send YieldDay instead of YieldTotal
useYieldDay=0

#For ESP8266 reduce polling intervall to reduce load
ESP8266PollingIntervall=10000

#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
#To keep current.log small use ERROR
Logging=ERROR

# if ts_last_success is older than this number of seconds, it is not used.
# Set this to < 0 to disable this check.
MagAgeTsLastSuccess=600

# if this is not 0, then no values are actually sent via dbus to vrm/venus.
DryRun=0

#IP of Device to query
Host=192.168.178.24

# Username/Password leave empty if no authentication is required
Username =admin
Password =************* (hidden)

### Only needed for OpenDTU and ahoy
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2

# 1st inverter
[INVERTER0]
Phase=L1
DeviceInstance=34
AcPosition=0

# 2nd inverter
[INVERTER1]
Phase=L1
DeviceInstance=35
AcPosition=0

# 3rd inverter
[INVERTER2]
Phase=L3
DeviceInstance=36
AcPosition=1

# 4th inverter
[INVERTER3]
Phase=L1
DeviceInstance=37
AcPosition=1

# 5th inverter
[INVERTER4]
Phase=L2
DeviceInstance=38
AcPosition=1

# 6th inverter
[INVERTER5]
Phase=L3
DeviceInstance=39
AcPosition=1

# 7th inverter
[INVERTER6]
Phase=L1
DeviceInstance=40
AcPosition=1

# 8th inverter
[INVERTER7]
Phase=L2
DeviceInstance=41
AcPosition=1

# 9th inverter
[INVERTER8]
Phase=L3
DeviceInstance=42
AcPosition=1

# 10th inverter
[INVERTER9]
Phase=L1
DeviceInstance=43
AcPosition=1

################## TEMPLATES #####################
####Rearrange and customite Templates as necessary
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2

[TEMPLATE0]
## Tasmota Example
##
## Username/Password leave empty if no authentication is required
Username = 
Password = 
Host=172.16.1.1
CUST_SN = 12345678
CUST_API_PATH= cm?cmnd=STATUS+8
CUST_POLLING = 2000
CUST_Total= StatusSNS/ENERGY/Total
CUST_Total_Mult = 1
CUST_Power= StatusSNS/ENERGY/Power
CUST_Power_Mult = 1
CUST_Voltage=  StatusSNS/ENERGY/Voltage
CUST_Current=  StatusSNS/ENERGY/Current
Phase=L1
DeviceInstance=47
AcPosition=1
Name= Tasmota

[TEMPLATE1]
## Shelly Gen 1 authenticated and unathenticated example
## Documentation: https://shelly-api-docs.shelly.cloud/gen1/#shelly1-1pm-status
##
## Username/Password leave empty if no authentication is required
Username = 
Password = 
Host=172.16.1.1
CUST_SN = 12345678
CUST_API_PATH= status
CUST_POLLING = 2000
CUST_Total= meters/0/total
###Shelly1PM Multiplier Watt/min 0.000017
CUST_Total_Mult = 0.000017
CUST_Power= meters/0/power
CUST_Power_Mult = 1
CUST_Voltage= none
CUST_Current= none
Phase=L2
DeviceInstance=45
AcPosition=1
Name= Shelly 1PM

[TEMPLATE2]
## Shelly Gen 2 - Only works unauthenticated!
## Documentation https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/EM#status
##
## Username/Password leave empty if no authentication is required
Host=172.16.1.1
CUST_SN = 12345678
CUST_API_PATH= rpc/Switch.GetStatus?id=0
CUST_POLLING = 2000
CUST_Total= aenergy/total
CUST_Total_Mult = 1
CUST_Power= apower
CUST_Power_Mult = 1
CUST_Voltage=  voltage
CUST_Current=  current
Phase=L3
DeviceInstance=46
AcPosition=1
Name= ShellyPlus1PM

peregrines2, 2023-02-24:
My config:

[DEFAULT]
# Time in minutes how often a status is added to the log-file current.log with log-level INFO
SignOfLifeLog= 1

# Number ob Template Inverter to query
NumberOfTemplates=2

# Which DTU to be used ahoy, opendtu, template
DTU=opendtu

# send YieldDay instead of YieldTotal
useYieldDay=0

#For ESP8266 reduce polling intervall to reduce load
ESP8266PollingIntervall=10000

#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
#To keep current.log small use ERROR
Logging=ERROR

# if ts_last_success is older than this number of seconds, it is not used.
# Set this to < 0 to disable this check.
MagAgeTsLastSuccess=600

# if this is not 0, then no values are actually sent via dbus to vrm/venus.
DryRun=0

#IP of Device to query
Host=192.168.178.24

# Username/Password leave empty if no authentication is required
Username =admin
Password =17491597!sS

### Only needed for OpenDTU and ahoy
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2

# 1st inverter
[INVERTER0]
Phase=L1
DeviceInstance=34
AcPosition=0

# 2nd inverter
[INVERTER1]
Phase=L1
DeviceInstance=35
AcPosition=0

# 3rd inverter
[INVERTER2]
Phase=L3
DeviceInstance=36
AcPosition=1

# 4th inverter
[INVERTER3]
Phase=L1
DeviceInstance=37
AcPosition=1

# 5th inverter
[INVERTER4]
Phase=L2
DeviceInstance=38
AcPosition=1

# 6th inverter
[INVERTER5]
Phase=L3
DeviceInstance=39
AcPosition=1

# 7th inverter
[INVERTER6]
Phase=L1
DeviceInstance=40
AcPosition=1

# 8th inverter
[INVERTER7]
Phase=L2
DeviceInstance=41
AcPosition=1

# 9th inverter
[INVERTER8]
Phase=L3
DeviceInstance=42
AcPosition=1

# 10th inverter
[INVERTER9]
Phase=L1
DeviceInstance=43
AcPosition=1

################## TEMPLATES #####################
####Rearrange and customite Templates as necessary
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2

[TEMPLATE0]
## Tasmota Example
##
## Username/Password leave empty if no authentication is required
Username = 
Password = 
Host=172.16.1.1
CUST_SN = 12345678
CUST_API_PATH= cm?cmnd=STATUS+8
CUST_POLLING = 2000
CUST_Total= StatusSNS/ENERGY/Total
CUST_Total_Mult = 1
CUST_Power= StatusSNS/ENERGY/Power
CUST_Power_Mult = 1
CUST_Voltage=  StatusSNS/ENERGY/Voltage
CUST_Current=  StatusSNS/ENERGY/Current
Phase=L1
DeviceInstance=47
AcPosition=1
Name= Tasmota

[TEMPLATE1]
## Shelly Gen 1 authenticated and unathenticated example
## Documentation: https://shelly-api-docs.shelly.cloud/gen1/#shelly1-1pm-status
##
## Username/Password leave empty if no authentication is required
Username = 
Password = 
Host=172.16.1.1
CUST_SN = 12345678
CUST_API_PATH= status
CUST_POLLING = 2000
CUST_Total= meters/0/total
###Shelly1PM Multiplier Watt/min 0.000017
CUST_Total_Mult = 0.000017
CUST_Power= meters/0/power
CUST_Power_Mult = 1
CUST_Voltage= none
CUST_Current= none
Phase=L2
DeviceInstance=45
AcPosition=1
Name= Shelly 1PM

[TEMPLATE2]
## Shelly Gen 2 - Only works unauthenticated!
## Documentation https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/EM#status
##
## Username/Password leave empty if no authentication is required
Host=172.16.1.1
CUST_SN = 12345678
CUST_API_PATH= rpc/Switch.GetStatus?id=0
CUST_POLLING = 2000
CUST_Total= aenergy/total
CUST_Total_Mult = 1
CUST_Power= apower
CUST_Power_Mult = 1
CUST_Voltage=  voltage
CUST_Current=  current
Phase=L3
DeviceInstance=46
AcPosition=1
Name= ShellyPlus1PM

henne49, 2023-02-24:
Please set Logging to DEBUG and rerun the script.

Also please not paste but attach the current.log file and the config.ini file

When did you update the venus OS Version to version 3?

peregrines2, 2023-02-24:
I am not completely sure, but when I use:

wget https://github.com/henne49/dbus-opendtu/archive/refs/heads/main.zip
unzip main.zip "dbus-opendtu-main/*" -d /data
mv /data/dbus-opendtu-main /data/dbus-opendtu
chmod a+x /data/dbus-opendtu/install.sh
nano /data/dbus-opendtu/config.ini

Will this also install the newest (fixed) version?
Here is my log and config (changed .ini to .txt)

[current.log](https://github.com/henne49/dbus-opendtu/files/10823817/current.log)

[config.txt](https://github.com/henne49/dbus-opendtu/files/10823860/config.txt)

henne49, 2023-02-24:
yes that installs the new version

henne49, 2023-02-24:
Hi, I can see a connection error

raise ConnectionError("No response from OpenDTU - %s" % (url))
ConnectionError: No response from OpenDTU - http://XXXXX:YYYYYY@192.168.178.24/api/livedata/status

can you try to not use a password and check connectivity?

what setting are you using?

[image]