henne49/dbus-opendtu #40: After Installation  - no service

exenbema, 2023-02-25:
Hi 
i installed today the newest script according to https://github.com/henne49/dbus-opendtu
then i tried to dbus-spy - but i see no opendtu
i tried all of this below, but get not aswer why the service is not running. i have superuser access and installed allready some guthub drivers on my cerbo. 
when i try to open the log file i get access denied.

thx markus!!

[image]

svstat /service/dbus-opendtu show if the service (our script) is running. If number of seconds show is low, the it is probably restarting and you should look into /data/dbus-opendtu/current.log.

/data/dbus-opendtu/uninstall.sh stops the service and prevents it from being restarted (e.g. after a reboot).

/data/dbus-opendtu/install.sh installs the service persistently (see above).

/data/dbus-opendtu/restart.sh restarts the service - e.g. after a config.ini change.

dbus-spy show all DBus values interactively.

henne49, 2023-02-25:
please increase logging to at least info, better debug

check the DeviceInstance you are using, if that is already used in venus/vrm?

do ./uninstall.sh
try to start the scrip using python3 dbus-opendtu.py from console

exenbema, 2023-02-25:
thx!! 

logging ok done
deviceinstance is not in use
after the python 3 command: python3: can't open file 'dbus-opendtu.py': [Errno 2] No such file or directory

there  are to videos: in this https://www.youtube.com/watch?v=UNuIOa72eP4&ab_channel=SchattenPV you see there are 3 more lines under code...but i dont see that anymore - he does something with update and python

can this be missing?

henne49, 2023-02-25:
bist du im richtigen Ordner sollte /data/dbus-opendtu sein

Hier ist die Anleitung wie man es installiert: https://github.com/henne49/dbus-opendtu#install--configuration

exenbema, 2023-02-25:
ich hab es vorher genauest nach Anleitung installiert - bin aber jetzt einen schritt weiter. ich sehe den Hoymiles schon im Victron system  - Yeah!

wie in dem vorigen video von schattenPV habe ich diese zusätzlichen 3 Zeiten installiert - jetzt gehts.

ich hab dann noch das ausgeführt, was du geschrieben hast - und ich erhielt dies, scheint mir ein IP Error zu sein, aber die IP kann ich pingen, ist im gleichen netz. und wie gesagt es wird mir bereits in der cerbo angezeigt

root@einstein:/data/dbus-opendtu# python3 dbus-opendtu.py
2023-02-25 19:09:08,60 root DEBUG _isDataUpToDate: inverter #0: age_seconds=2108770, max_age_ts=600
2023-02-25 19:09:08,63 root DEBUG _isDataUpToDate: inverter #0: age_seconds=10, max_age_ts=600
2023-02-25 19:09:08,65 root INFO Start
2023-02-25 19:09:08,121 urllib3.connectionpool DEBUG Starting new HTTP connection (1): 192.168.1.199:80
2023-02-25 19:09:09,124 root CRITICAL Error at main
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/urllib3/connection.py", line 159, in _new_conn
    conn = connection.create_connection(
  File "/usr/lib/python3.8/site-packages/urllib3/util/connection.py", line 84, in create_connection
    raise err
  File "/usr/lib/python3.8/site-packages/urllib3/util/connection.py", line 74, in create_connection
    sock.connect(sa)
OSError: [Errno 113] No route to host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/urllib3/connectionpool.py", line 665, in urlopen
    httplib_response = self._make_request(
  File "/usr/lib/python3.8/site-packages/urllib3/connectionpool.py", line 387, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.8/http/client.py", line 1256, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1302, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1251, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1011, in _send_output
    self.send(msg)
  File "/usr/lib/python3.8/http/client.py", line 951, in send
    self.connect()
  File "/usr/lib/python3.8/site-packages/urllib3/connection.py", line 187, in connect
    conn = self._new_conn()
  File "/usr/lib/python3.8/site-packages/urllib3/connection.py", line 171, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object at 0xb58d8d48>: Failed to establish a new connection: [Errno 113] No route to host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/requests/adapters.py", line 439, in send
    resp = conn.urlopen(
  File "/usr/lib/python3.8/site-packages/urllib3/connectionpool.py", line 719, in urlopen
    retries = retries.increment(
  File "/usr/lib/python3.8/site-packages/urllib3/util/retry.py", line 436, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='192.168.1.199', port=80): Max retries exceeded with url: /api/livedata/status (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb58d8d48>: Failed to establish a new connection: [Errno 113] No route to host'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "dbus-opendtu.py", line 498, in main
    service = DbusService(
  File "dbus-opendtu.py", line 71, in __init__
    self.numberofinverters  = self._getNumberOfInverters()
  File "dbus-opendtu.py", line 224, in _getNumberOfInverters
    meter_data = self._getData()
  File "dbus-opendtu.py", line 322, in _getData
    self._refreshData()
  File "dbus-opendtu.py", line 293, in _refreshData
    meter_r = requests.get(url = url, timeout=2.50)
  File "/usr/lib/python3.8/site-packages/requests/api.py", line 76, in get
    return request('get', url, params=params, **kwargs)
[... 8 lines cut ...]
requests.exceptions.ConnectionError: HTTPConnectionPool(host='192.168.1.199', port=80): Max retries exceeded with url: /api/livedata/status (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb58d8d48>: Failed to establish a new connection: [Errno 113] No route to host'))
root@einstein:/data/dbus-opendtu# python3 dbus-opendtu.py
2023-02-25 19:13:25,71 root DEBUG _isDataUpToDate: inverter #0: age_seconds=2109027, max_age_ts=600
2023-02-25 19:13:25,74 root DEBUG _isDataUpToDate: inverter #0: age_seconds=10, max_age_ts=600
2023-02-25 19:13:25,76 root INFO Start
2023-02-25 19:13:25,200 urllib3.connectionpool DEBUG Starting new HTTP connection (1): 192.168.1.199:80
2023-02-25 19:13:25,303 urllib3.connectionpool DEBUG http://192.168.1.199:80 "GET /api/livedata/status HTTP/1.1" 200 944
2023-02-25 19:13:25,317 root INFO Number of Inverters found: 1
2023-02-25 19:13:25,319 root DEBUG com.victronenergy.pvinverter /DeviceInstance = 34
2023-02-25 19:13:25,338 root CRITICAL Error at main
Traceback (most recent call last):
  File "dbus-opendtu.py", line 498, in main
    service = DbusService(
  File "dbus-opendtu.py", line 80, in __init__
    self._dbusservice = VeDbusService("{}.http_{:02d}".format(servicename, self.deviceinstance),dbusConn)
[... cut ...]
  File "/usr/lib/python3.8/site-packages/dbus/service.py", line 144, in __new__
    raise NameExistsException(name)
dbus.exceptions.NameExistsException: Bus name already exists: com.victronenergy.pvinverter.http_34
Exception ignored in: <function VeDbusService.__del__ at 0xb58b09b8>
Traceback (most recent call last):
  File "/opt/victronenergy/dbus-systemcalc-py/ext/velib_python/vedbus.py", line 93, in __del__
    if self._dbusname:
AttributeError: 'VeDbusService' object has no attribute '_dbusname'

henne49, 2023-02-25:
kannst du die ip der dtu vom cerbo pinger?

und hast du die latest ahoy und opendtu installiert? wenn ja welche?

und hast du auf der dtu einen password Schutz?

exenbema, 2023-02-25:
1) ja 
2) ja diese woche erst installiert. alles genau nach anleitung. version SDK Version 4.4.3 konfigversion 0.1.24 firmware 97bc964
3) ich hab admin und passwort nun eingegeben 

root@einstein:/data# cd dbus-opendtu
root@einstein:/data/dbus-opendtu# python3 dbus-opendtu.py
2023-02-25 19:34:02,263 root CRITICAL Error at main
Traceback (most recent call last):
  File "dbus-opendtu.py", line 498, in main
    service = DbusService(
  File "dbus-opendtu.py", line 80, in __init__
    self._dbusservice = VeDbusService("{}.http_{:02d}".format(servicename, self.deviceinstance),dbusConn)
  File "/opt/victronenergy/dbus-systemcalc-py/ext/velib_python/vedbus.py", line 77, in __init__
    self._dbusname = dbus.service.BusName(servicename, self._dbusconn, do_not_queue=True)
  File "/usr/lib/python3.8/site-packages/dbus/service.py", line 144, in __new__
    raise NameExistsException(name)
dbus.exceptions.NameExistsException: Bus name already exists: com.victronenergy.pvinverter.http_34
Exception ignored in: <function VeDbusService.__del__ at 0xb584a9b8>
Traceback (most recent call last):
  File "/opt/victronenergy/dbus-systemcalc-py/ext/velib_python/vedbus.py", line 93, in __del__
    if self._dbusname:
AttributeError: 'VeDbusService' object has no attribute '_dbusname'

henne49, 2023-02-25:
das sieht gut aus, der fehler kommt, weil der service schon läuft. mach mal ein ./uninstall.sh

dann ein python3 dbus-opendtu.py

wenn das gut aussieht, dann ein ./restart.sh und der Dienst sollte laufen. Daten wirst du wohl erst morgen sehen, wenn der Inverter auch produziert, auch die Gesamtwerte.

exenbema, 2023-02-25:
danke, ja das denke ich auch:) herzlichen dank fürs das durchbegleiten!

--  ./uninstall.sh ja mach ich aber vllt dumme frage. deinstalliere ich da eh nicht den serialbattery treiber. muss ich da vorher in das verzeichnnis wechseln? sorry bin da noch in diesen themen

henne49, 2023-02-25:
ja das uninstall machst du im verzeichnis, aber damit stoppst du den service nur, wir haben den so eingetragen dass er nach einem restart wieder da ist. und ein restart.sh lässt den dienst auch wieder als Service laufen. Also ist alles sicher.

exenbema, 2023-02-25:
danke ich werde morgen bei sonne mal schauen!

exenbema, 2023-02-26:
guten morgen! sehe ich das richtig, dass man mit dem script mehrere geräte abfragen kann. zb ich könnte eine opendtu und eine shelly abfragen mit dem gleichen script? sprich dann würde man sich das github shelly service zb sparen?

dh ich trage zb auch die opendtu im ersten template ein (dort wo tasmota steht) und im 2. template die shelly?
wenn dem so ist, welche IP trage ich dann im allgemeinen Bereich ein? 
#IP of Device to query
Host=xxxx

oder laufen in dem fall 2 scripts mit 2 config.ini datein (wenn ja, wie installiere ich das zusätzlich?)

danke!

henne49, 2023-02-26:
wie in der doku beschrieben: 

The code allows to query up to one DTU, either Ahoy or OpenDTU, plus multiple template based PV Inverter in a single script. This means, you can also only query template devices or only a dtu, or a mix of one DTU and template devices.

also ja, oben kommt die IP der ahoy oder opendtu hin. 

Alles in einem script. ja das shelly script kann man sich dann sparen

exenbema, 2023-02-26:
oh perfekt, das ist ja grossartig - danke!!

henne49, 2023-02-26:
> ja das uninstall machst du im verzeichnis, aber damit stoppst du den service nur, wir haben den so eingetragen dass er nach einem restart wieder da ist. und ein restart.sh lässt den dienst auch wieder als Service laufen. Also ist alles sicher.

hier muss noch ein ./install.sh hin natürlich vor dem restart.sh

closed 2023-02-28 by henne49