henne49/dbus-opendtu #41: Venus Data Partition 100% Full

exenbema, 2023-03-03:
Hi! thx for the great work - first of all. i tried opendtu and ahoy - both perfect working.

Now i have the data partition 100% so pls help me understand if i could free some data and what did i wrong

1) as i switched to ahoy i let this DEFAULT | ESP8266PollingIntervall | For ESP8266 reduce polling intervall to reduce load, default 10000ms -- is this bad and can this fill up the data partition of venus os

2) i inserted a 64GB SD Card in Venus - and after reboot there is 44 GB consumed. is this a swap file or something? is this maybe data load from ahoy or somewhere else

3) how can i free up some space and ca i direct the data load to a new sd card?

thx markus

henne49, 2023-03-03:
please adjust the config.ini, but actually this should be at on error or on info, if you have not changed it yourself. At a restart of the Venus OS we also delete the log files. 

DEFAULT | Logging | Valid options for log level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET, to keep logfile small use ERROR or CRITICAL

check the size and please delete current.log under /data/dbus-opendtu.

exenbema, 2023-03-03:
thx i did all that, cleaned the log file (was 200MB) - logging was all time only on error - i set back to opendtu instead of ahoybut the log file rise very fast. i reinstalled the service and now i increased the polling interval to 30000. i also set the value on the ahoy from 30seconds to 120. now everything seems ok - the only thing is my data volume on cerbo is still at 100%:) but maybe another error there

henne49, 2023-03-03:
that is what I expected, as the logging can be significant, but not that impacting. we have develeoped it with that in mind to not grow to big. 

would you know how long the opendtu was running, to reach this size?

exenbema, 2023-03-03:
Puh i think 2 weeks

dsteinkopf, 2023-03-03:
Can you give us a bit more about what's inside the log file? Maybe then, we are able to see what could be done to reduce it.

exenbema, 2023-03-03:
i am a newbee in linux - i find the commands first time - so (sorry!!) i deleted the log file and didnt see inside. i still had the 100% data partition - and now i think i solved this mysterium, in the data dir was a 4096 directory named shelly-1pm-inverter and  i removed that - after this action my data partition is 1% and not 100%. 

i think there was a problem with the shelly script - i had this as test - but im not a fan of shelly anymore, because they send so much infos to china - so i wanted to replace it with tasmota (i was unsuccessful in this) and maybe this failed transaction from shelly to tasmota was the spaming problem

exenbema, 2023-03-03:
my root dir is 88% ful but i think its not unusual. i thougt about to reinstall the complete venus and then with the guimods from kevins setup helper (package manager) - but right now i am a bit afraid of data load with this setup manager:) 

i also want to play with node red, but if am not sure about space,...hm

henne49, 2023-03-06:
issue solved, not a dbus-opendtu problem

closed 2023-03-06 by henne49