henne49/dbus-opendtu #42: Device inverter instead of pvinverter possible?

TheGrumpyBadger, 2023-03-05:
My HM-350 is connected to a LFP battery, which is charged by a Victron MPPT.
Is it possible to show the HM-350 as an inverter, as it is using DC power instead of PV ?

henne49, 2023-03-06:
as long as you have a DTU like ahoy or opendtu, that we can read data from, it can be displayed in venus OS

ude6, 2023-03-09:
Hi,
I agree. I am currently using your code to display the inverter (connected to the battery) as
a PV-Inverter. It runs very well. BUT it assumes the energy comes from PV where in reality it comes from the battery.

So when the inverter runs I see for example:

Battery (via dbus-serialbattery) -600w
PV-Inverter ( via your dbus-opendtu) +600w
DC-Power (calculated by victron) +600w 

The problem is that the cerbo is not netting the DC-Power (which is the input to the inverter) 
with the PV-Inverter.

So the question is: Can the Opendtu (or Ahoy) be used as an inverter that is powered by DC load?
[image]

Thanks

Andreas

ude6, 2023-03-09:
Yes, I would assume that something like this would be through the com.victronenergy.vebus.
It would need to look like a Multiplus to the Cerbo (I guess)....

henne49, 2023-03-09:
I planned something for shelly https://github.com/henne49/dbus-opendtu/issues/32, but it was not working. This requires some rework, give it a try and you can submit a PR to include into the code. But wait until we Merged https://github.com/henne49/dbus-opendtu/pull/43

ude6, 2023-03-09:
Sounds great!

ude6, 2023-03-09:
This looks close: https://github.com/Telekatz/venus.dbus-hoymiles

henne49, 2023-03-10:
we did a merge, your link is a different use case. Maybe that project is a better bit for your use case

ude6, 2023-03-10:
I will try to ask the autor there for a possible update to disble the steering of the HM input.

Head, 2023-03-14:
Was just reading the code and then found this _issue_.
This project simulates the opendtu as a pvinverter: https://github.com/victronenergy/venus/wiki/dbus#pv-inverters
_We_ would need an inverter: https://github.com/victronenergy/venus/wiki/dbus#inverter

It shouldn't be that hard to take this code and change `com.victronenergy.pvinverter` to `com.victronenergy.inverter` and also change the paths in the update funtion https://github.com/henne49/dbus-opendtu/blob/main/dbus-opendtu.py#L423 to match the ones required by the inverter interface. But maybe that should ne another project/fork of this one... Or maybe a config flag?

henne49, 2023-03-15:
I think it needs to be a config flag, which allows for different mode. If you are interested, please create a fork and test, we can help with the code and Pull Request. We would include this into code for sure, just have no time yet to do it myself.

dsteinkopf, 2023-03-25:
Maybe, the change #53 (based on feature request #52) will also help in this case. I will be able to configure servicename `com.victronenergy.inverter`.

dsteinkopf, 2023-03-25:
Closing thing because of #53. Please reopen or create a new issue if necessary.

henne49, 2023-03-26:
Reopening, as we need to also implement it for ahoy and opendtu and not just template

TheGrumpyBadger, 2023-03-28:
just tried it.
---
### Only needed for OpenDTU and ahoy
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2
# 1st inverter
[INVERTER0]
Phase=L1
DeviceInstance=34
AcPosition=0
Servicename=com.victronenergy.inverter

# 2nd inverter
[INVERTER1]
Phase=L1
DeviceInstance=35
AcPosition=0
Servicename=com.victronenergy.pvinverter
---
Inverter0 is not connected, Interverter1 works fine.

[image]

TheGrumpyBadger, 2023-03-28:
I think for inverter there are other variables:
https://github.com/victronenergy/venus/wiki/dbus#inverter

[image]

TheGrumpyBadger, 2023-04-02:
I think we will need:

`/Dc/0/Voltage             <- Battery voltage
/Ac/Out/L1/V              <- AC Output voltage
/Ac/Out/L1/I              <- AC Output current
/Ac/Out/L1/P              <- Not used on vedirect inverters 
/State                    <- 0=Off; 1=Low Power; 2=Fault; 9=Inverting`

Tom051276, 2023-04-27:
Dear all, was this feature request from @TheGrumpyBadger already built in ? I tried to use the Servicename=com.victronenergy.inverter for Inverter0 But without luck. Seems it is not yet implemented...

dsteinkopf, 2023-04-27:
Yes, setting the servicename was implemented in #53. But it seems there are devices (servicenames) where it is not sufficient to only set the servicename and some more attributes need to be set. This is why THIS ticket ist still open.

0x7878, 2023-04-27:
Hey @Tom051276 would you mind to post the debug log? 

Logging=DEBUG

Right now it looks correct for me. 

```python
if dtuvariant != constants.DTUVARIANT_TEMPLATE:
            logging.info("Registering dtu devices")
            servicename=get_config_value(config, "Servicename", "INVERTER", 0, "com.victronenergy.pvinverter")
            service = DbusService(
                servicename=servicename,
                paths=paths,
                actual_inverter=0,
            )
```

Maybe there's something wrong with the getter. I'm not able to debug that right now.

TheGrumpyBadger, 2023-04-27:
As far as I see the paths are only implemented for pv inverter:
`        paths = {
            "/Ac/Energy/Forward": {
                "initial": None,
                "textformat": _kwh,
            },  # energy produced by pv inverter
            "/Ac/Power": {"initial": None, "textformat": _w},
            "/Ac/L1/Voltage": {"initial": None, "textformat": _v},
            "/Ac/L2/Voltage": {"initial": None, "textformat": _v},
            "/Ac/L3/Voltage": {"initial": None, "textformat": _v},
            "/Ac/L1/Current": {"initial": None, "textformat": _a},
            "/Ac/L2/Current": {"initial": None, "textformat": _a},
            "/Ac/L3/Current": {"initial": None, "textformat": _a},
            "/Ac/L1/Power": {"initial": None, "textformat": _w},
            "/Ac/L2/Power": {"initial": None, "textformat": _w},
            "/Ac/L3/Power": {"initial": None, "textformat": _w},
            "/Ac/L1/Energy/Forward": {"initial": None, "textformat": _kwh},
            "/Ac/L2/Energy/Forward": {"initial": None, "textformat": _kwh},
            "/Ac/L3/Energy/Forward": {"initial": None, "textformat": _kwh},
        }`

The paths for a "non-pv"-inverter I posted above.

0x7878, 2023-04-27:
Got you. I created a new branch [feature-non-pv-inverter](/henne49/dbus-opendtu/tree/feature-non-pv-inverter). I'm not able to test it yet but maybe it's a start

TheGrumpyBadger, 2023-04-27:
Yes, I got values:
[image]

But no visual:
[image]

TheGrumpyBadger, 2023-04-27:
2023-04-27 14:04:38,184 root DEBUG calling http://192.168.178.200/api/livedata/status with timeout=2.5
2023-04-27 14:04:38,191 urllib3.connectionpool DEBUG Starting new HTTP connection (1): 192.168.178.200:80
2023-04-27 14:04:38,261 urllib3.connectionpool DEBUG http://192.168.178.200:80 "GET /api/livedata/status HTTP/1.1" 200 1401
2023-04-27 14:04:38,265 root DEBUG function fetch_url finished in 81 ms
2023-04-27 14:04:40,972 root DEBUG _get_value_handler called for /
2023-04-27 14:04:40,973 root DEBUG {'Mgmt/ProcessName': dbus.String('/data/dbus-opendtu/dbus_service.py', variant_level=1), 'Mgmt/ProcessVersion': dbus.String('Unkown version, and running on Python 3.8.13', variant_level=1), 'Mgmt/Connection': dbus.String('OpenDTU HTTP JSON service', variant_level=1), 'DeviceInstance': dbus.Int32(34, variant_level=1), 'ProductId': dbus.Int32(65535, variant_level=1), 'ProductName': dbus.String('OpenDTU', variant_level=1), 'CustomName': dbus.String('HM350-Akku', variant_level=1), 'Connected': dbus.Int32(1, variant_level=1), 'Latency': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'FirmwareVersion': dbus.Double(0.1, variant_level=1), 'HardwareVersion': dbus.Int32(0, variant_level=1), 'Position': dbus.Int32(1, variant_level=1), 'Serial': dbus.String('112183215725', variant_level=1), 'UpdateIndex': dbus.Int32(0, variant_level=1), 'StatusCode': dbus.Int32(0, variant_level=1), 'Ac/Energy/Forward': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/Power': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/L1/Voltage': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/L2/Voltage': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/L3/Voltage': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/L1/Current': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/L2/Current': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/L3/Current': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/L1/Power': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/L2/Power': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/L3/Power': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/L1/Energy/Forward': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/L2/Energy/Forward': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/L3/Energy/Forward': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/Out/L1/I': dbus.Double(1.059999943, variant_level=1), 'Ac/Out/L1/V': dbus.Double(234.1000061, variant_level=1), 'Dc/0/Voltage': dbus.Int32(49, variant_level=1)}

TheGrumpyBadger, 2023-04-27:
Perhaps the /state has to be 0 when I is 0 and 9 when I is > 0.
And /Mode to 2 when working.

0x7878, 2023-04-27:
@dsteinkopf I have no idea how to contact you on Discord. Just add me there `Maurice#2144`

@TheGrumpyBadger I'll change/check this later. As I said I couldn't test it ;-)

TheGrumpyBadger, 2023-04-27:
`            "/Mode": {"initial": 2, "textformat": None},

            "/State": {"initial": 9, "textformat": None}
`

did it.

Line 87/77 in dbus-opendtu.py

[image]

[image]

0x7878, 2023-04-27:
Great job! 👍

henne49, 2023-04-27:
@0x7878 will you update the branch and create the PR? Well done guys

Tom051276, 2023-04-28:
@0x7878 @TheGrumpyBadger Awesome !!! I will give it a new try this weekend. Sounds great, Thank you for your work.

ude6, 2023-04-29:
Hi,
great progress and I am excited to use this.
When I tried the branch on a spare Rspi I found that for me the changes requested by @TheGrumpyBadger looked like they were not in the code?
For ref:
"        "/Mode": {"initial": 2, "textformat": None},
        "/State": {"initial": 9, "textformat": None}
did it.

Line 87/88 in dbus-opendtu.py"

Also the inverter is inverter no2 in my list and it did not get any values. (But that may be due to my hasty / maybe faulty setup of the raspi..

Thanks!

ude6

TheGrumpyBadger, 2023-04-29:
Hey ude6,

this two lines has to be added to the code, has to look like this:

            "/Ac/Out/L1/I": {"initial": None, "textformat": _a},
            "/Ac/Out/L1/V": {"initial": None, "textformat": _v},
            "/Ac/Out/L1/P": {"initial": None, "textformat": _w},
            "/Dc/0/Voltage": {"initial": None, "textformat": _v},
            "/Mode": {"initial": 2, "textformat": None},
            "/State": {"initial": 9, "textformat": None}
        }

TheGrumpyBadger, 2023-04-29:
@henne49 : I created a pull request for this to lines.

0x7878, 2023-04-29:
This is not necessary as I am still working on it. Dbus_opendtu.py isn't really the best place to put the code either. In this section are some functions with some callbacks on text format.

The last commit broke some things that I already fixed but didn't push. I hope I'll be done with that tomorrow. I will also check why you might not be getting the values.

Tom051276, 2023-05-01:
@TheGrumpyBadger Is The total consumption calculated correct? I am wondering that now only the net in is summed up, but not the ac-out delivered from the hoymiles…
Before these change the hoymiles deliver was summed up in PV Power in, which was also not correct… Not sure how to solve this… I do not have the DC module like you have…

Tom051276, 2023-05-01:
[image]

TheGrumpyBadger, 2023-05-01:
This look good on my side:
[image]

in VRM it look not fine, but its ok for me:
[image]

Tom051276, 2023-05-01:
@TheGrumpyBadger Ah okay, then you have the same problem on vrm site like I have. Any ideas how to solve this miscalculation on vrm?
Would be great if vrm site an also be fixed...

TheGrumpyBadger, 2023-05-02:
As far as I see there are no Victron inverters that can synchronize with die grid.
So the usecase we have is not supported in VRM. It is not possible to add grid and inverter power in the charts.

TheGrumpyBadger, 2023-05-04:
I found some miscalculations in VRM with AcPosition=1.
Doing some more testing, but AcPosition=0 seems to be better for the inverter.

0x7878, 2023-05-04:
Good to know. If so, such information belongs to the readme. I also figured out that (at least in my installation) it takes up to 10 minutes before the VRM-Portal starts showing the live values of an e.g. (PV-)inverter.

henne49, 2023-05-04:
The 10 minutes can be configured in vrm or Venus, as the graphs you see are historical, that is why you have a delay.

I would at least use a delay of 10 minutes, as too short delays, you can have historical errors in case of unreachable or reboot of opendtu when opendtu shows 0

dsteinkopf, 2023-05-04:
@henne49 ah yes, that's probably the rease for the delay. I am "happily" using 1 Minute.

> I found some miscalculations in VRM with AcPosition=1. Doing some more testing, but AcPosition=0 seems to be better for the inverter.

It's in the config.ini: `# 0=AC input 1; 1=AC output; 2=AC output 2`. Yes, it should also be in the readme. But it's no "soft decision", it's a fact that must be configured correctly (where is the inverter connected), isn't it?

TheGrumpyBadger, 2023-05-04:
looks good so far:
[image]

[image]

0x7878, 2023-05-04:
In general: If a device becomes an inverter, the device in the middle (with the Victron logo) represents that device. This is why there is a label "Inverting, Off, etc.".

Furthermore, it works like that: The inverter takes Energy from the Battery and pushes it to AC-Loads. which then shows up in "consumption".

Feel free to correct me if something that I said isn't right.

TheGrumpyBadger, 2023-05-04:
You are right. On the screenshot are two HM-350 on a single openDTU-onbattery.
The first one ist an L1 and the inverter, it is feeded by an 24V LFP battery,
second one is on L3 and is a normal pvinverter on L3.
Both are now on AcPosition=0 and the numbers seems to be right.

TheGrumpyBadger, 2023-05-04:
> It's in the config.ini: `# 0=AC input 1; 1=AC output; 2=AC output 2`. Yes, it should also be in the readme. But it's no "soft decision", it's a fact that must be configured correctly (where is the inverter connected), isn't it?

for the inverter config it's indeed a soft decision. An inverter has no "AC input", so I thought "AC output" must be right, but perhaps not. That's why I'm in testing.

closed 2023-05-04 by henne49

0x7878, 2023-05-16:
Hey 👋
Please remove the hash sign "#" otherwise it will work as default (in this case PV inverter). After that check the values on your cerbo or via the terminal. See the readme for useful commands

DrknMonkey, 2023-05-17:
Thank you... you mean it so or:

[image]

Unfortunately in the Cerbo is the HM-300 Status then unknown:
[image]

0x7878, 2023-05-17:
That's right. I'll try your setup with a test device the next days. It could be that an Inverter does not have ac in which lead to your problem. Could you try ac out?

DrknMonkey, 2023-05-17:
Not so easy... but i have activate the AC-IN from the Multiplus, unfortunately still the same.

0x7878, 2023-05-17:
Ok I see. First of all this setup was implemented without a Multiplus in place. But still... I'm wondering why there are no values at all in your setup. I'll try to figure it out but because of "Vatertag" it has to wait ;-)

DrknMonkey, 2023-05-17:
Thank you... it would be nice if it works with a Multiplus too :). But first i wish you a great "Vatertag" ;).

WilliDerTiger, 2023-05-21:
Hello,
Thanks for the good integration into the Victron world ( Cerbo GX ).
I have a Shelly 3EM working on the grid, which I have successfully connected via [https://github.com/fabian-lauer/dbus-shelly-3em-smartmeter](url).
I have a Victron 150/35 MPPT pv-charger for my battery 48V.
From the battery, I feed a Hoymiles HM-800 into the L3 grid.
A second Hoymiles HM-600 (PV inverter) feeds PV directly to L3 in the grid ( 1st inverter ).
This has all been integrated correctly,
except that the HM-800 (inverter) feeds into L1 according to the display.

I have set it to L3 in config.ini ( 2nd inverter ).
[image]

How can I change it to L3?

[image]

[image]
[image]

[image]

Thanks

0x7878, 2023-05-21:
Victron doesn't support it. 
[image]
They have no other endpoint than L1 and I didn't test a different one because it would then be an undocumented endpoint.

WilliDerTiger, 2023-05-21:
OK, then I have to live with it.
Thanks

dsteinkopf, 2023-05-21:
We are currently (atm) doing several experiments, but yes, it seems Victron does not support it... At least not this way, maybe there is another one...

TheGrumpyBadger, 2023-05-21:
Please keep in mind Victron never intended to use an inverter on a system conncted with the grid.

WilliDerTiger, 2023-05-22:
Yes, will certainly be better,
to imitate Multiplus.

dsteinkopf, 2023-05-22:
What do you mean by mimic/imitate a Multiplus?

DrknMonkey, 2023-05-27:
> Ok I see. First of all this setup was implemented without a Multiplus in place. But still... I'm wondering why there are no values at all in your setup. I'll try to figure it out but because of "Vatertag" it has to wait ;-)

Hello,

gives it already a Solution or is it not possible :)

[4 more replies]

PR #77 (2023-05-04): Feature non pv inverter
This PR resolves #42 

I Added new dbus-paths to the codebase and add a new condition in the update function where the assignment takes place.

```python
 if self._servicename == "com.victronenergy.inverter":
 ```
 furthermore, there is a new template variable added to the code (`CUST_DCVoltage`). DC Voltage should also be written to the dbus system while retrieving the data from open_dtu respectively AhoyDTU
 
 Please notice this is not tested. (I'm still working on setup a dev device (venus os on a Raspberry Pi))