henne49/dbus-opendtu #48: help needed

hurly88, 2023-03-18:
Hi, great work! My first installation on a raspberry Pi with venus os, 3 weeks ago worked perfectly and is still working fine! This week i tried to repeat the procedure on my new Multiplus II GX, the installation worked fine but there is no communication between my opendtu and the gx device in my Multiplus. Did anybody test it on a Multiplus II GX?

henne49, 2023-03-18:
I have an Multi II GX myself. i would guess it is a connection problem.

please follow the Troubleshooting section and set logging to debug and share the logs please.

hurly88, 2023-03-20:
I tried to follow the troubleshooting section and i compared several times settings to my pi with venus os where it is working. 
In dbus-spy i can´t see the opendtu, in the other opened issue i found this command and tried it:

root@nanopi:~# svstat /service/dbus-opendtu
/service/dbus-opendtu: up (pid 31841) 3 seconds

Sorry that i don´t know much about debugging or logs, is there anything else i can try to make you understand better my situation what i am doing wrong on my seccond instance.

hurly88, 2023-03-20:
i also copy/pasted my config.ini from my running installation to the multiplus gx device without success, i also reinstalled everything 2 or 3 times in the hope that i find my mistake. Thanks in advance for your help

henne49, 2023-03-20:
please follow this https://github.com/henne49/dbus-opendtu#troubleshooting

you have to change debug settings and then please share current.log

henne49, 2023-03-26:
as of no response, we close the issue

closed 2023-03-26 by henne49

jernejnose, 2023-05-22:
Hi,
in case, someone else experiences same/similar problem. I also had problems running this on Multiplus-II GX. Communication worked fine, but service kept restarting and no log file was generated. In my case problem was in old Venus OS, because versions prior v2.8 had Python2 installed by default. After upgrading to v2.94 everything worked fine. Other option would be to install Python3 package.