henne49/dbus-opendtu #49: OpenDTU integration does not work

chrubler, 2023-03-19:
Hello,
I have a problem with the integration of my OpenDTU into the Victron system. It had worked before, but after a few settings it was gone again. I reinstalled everything according to the instructions, but still nothing. Finally, I reset my Cerbo GX completely to factory settings and set it up again, including the dbus-opendtu, but still nothing. I also completely reinstalled OpenDTU on the ESP32, but it just doesn't work anymore. OpenDTU itself works (can access it and get the data), but it just won't appear in the Victron system.

I have checked the IP in the config.ini several times and it matches. I then entered the command dbus-spy, but I don't see OpenDTU there either.

With the command svstat /service/dbus-opendtu I get values between 1 and 3, which indicates an error. The current.log then says the following:

line 1: 2023-03-19: command not found
line 2: syntax error near unexpected token `most'
line 2: 'Traceback (most recent call last)'.

I don't know what this means. I am a newcomer to Linux and have only ever followed the instructions. Can someone help me?
Die config und json Datei habe ich [hier](https://fotocloud.mitunszusammen.de/s/fdBAQRxg9L8e4FM) hochgeladen.

Regards

henne49, 2023-03-20:
Can you please change this in config.ini
Logging=DEBUG

and I would then need the current.log

I hope this helps

chrubler, 2023-03-20:
I changed the config.ini
That hasn't fixed the problem yet. I have uploaded the new config as config copy.ini again and also the current.log file.

henne49, 2023-03-20:
requests.exceptions.HTTPError: 401 Client Error: Unauthorized for url: http://192.168.1.26/api/livedata/status

it seems you have password protected your opendtu, but not put your login credentials and your config has no credentials

# Username/Password leave empty if no authentication is required
Username =
Password =

chrubler, 2023-03-20:
Hello, indeed. I was not aware that the admin password had to be entered. I can access the DTU web interface without a password and only need it for some settings.
But thank you very much, it works now.

henne49, 2023-03-20:
[image]

It should work when you have the setting like this

henne49, 2023-03-20:
we will update documentation to have this available for anyone much easier next time

chrubler, 2023-03-20:
Perfect. I had understood the setting the other way around.

closed 2023-03-20 by henne49