henne49/dbus-opendtu #50: Thanks & how hard would it be to make your driver more flexible?

stundenblume, 2023-03-20:
Hello Henne,
Thank you very much for this extremely useful and simple driver! It was possible to integrate Ahoiy without any problems.

So far I have used [dbus-mqtt](https://github.com/freakent/dbus-mqtt-devices) for my application. A very flexible driver that can register all kinds of devices on the DBus.
The big advantage is that you can store the definitions of the possible[ DBus devices](https://github.com/victronenergy/venus/wiki/dbus) in a .yml file.
The disadvantage is that the MQTT broker of venusOS is not accessible via WIFI and not all devices can be configured to send the appropriate start message.

My application would be the following:
Ahoy, RasPi with venusOS (via WIFI) and a Tasmota device on the Smartmeter.

With your driver I already have Ahoy in the system.

How much effort would it take to integrate this driver with devices other than "pvinverter"? Specifically "Gridmeter" and "Temperature" (I use the temperature to display a solar forecast in the VRM Dashboard).

Many thanks
Sebastian

henne49, 2023-03-20:
I had a look, it is not hard, but work :-) 

Why should we implement this in this driver, if it works using mqtt? What is the value? I think this is use case, but rather a small subset of users, who would use it. 

If you want to add it, feel free and we can work together.

stundenblume, 2023-03-20:
dbus-mqtt
Advantages:
 - any dbus service can be registered, very flexible
Disadvantages:
 - Registration sequence is relatively cumbersome
 - The device must write to the appropriate topics

dbus-opendtu
Advantages:
 - Flexibly fetches data on its own from any structure. The source device can be very dumb.
Disadvantages:
 - only inverters as dbus service are possible at the moment.

I see two main areas of application:
1. almost everyone who wants to run a Victron ESS and already has a smartmeter installed by their grid operator could save an additional gridmeter. (I know installations with Smartmeter, Fronius Gridmeter and Victron EM24, all three do the same thing).
2. people who will install a 600W mini PV and get a smartmeter installed, together with a RasPi with venusOS can use an excellent portal without having to buy an expensive gridmeter (EM24).

I think this great driver would greatly enhance the flexibility of more than just "inverter".

I have a basic understanding of programming, but am only a mechanical engineer. At the same time, I'm happy to do work. I will have a look if I understand what happens in your code.

Here are two concrete examples of what I have in mind:
dbus Gridmeter, source: [Tasmota Smartmeter](https://tasmota.github.io/docs/Smart-Meter-Interface/)
```
{
  "StatusSNS": {
    "Time": "2023-03-20T22:43:44",
    "SML": {
      "Total_in": 1584.735,
      "Total_out": 449.21,
      "Power_curr": 14,
      "Volt_p1": 234.2,
      "Volt_p2": 233.6,
      "Volt_p3": 233.1,
      "Amperage_p1": 1.1,
      "Amperage_p2": 0.3,
      "Amperage_p3": 0.6,
      "Power_curr_p1": -42,
      "Power_curr_p2": 17,
      "Power_curr_p3": 39,
      "frequency": 50
    }
  }
}
```

dbus Temperatur, source: [Solarprognose.de](https://www.solarprognose.de/web/en/solarprediction/page/api)
```
{
  "preferredNextApiRequestAt": {
    "secondOfHour": 3558,
    "epochTimeUtc": 1679284758
  },
  "status": 0,
  "iLastPredictionGenerationEpochTime": 1679283198,
  "weather_source_text": "<a target=\"_blank\" href=\"https://www.dwd.de/DE/service/copyright/copyright_node.html\"><img src=\"https://www.solarprognose.de/web/img/dwd_logo_258x69.png\"></a>",
  "datalinename": "Germany > Niedersachsen",
  "data": {
    "20230320": 8.837,
    "20230321": 5.587
  }
}
```

henne49, 2023-03-21:
Tasmota devices should work already. The only thing is to create different gridmeter or.

In Template devices the intelligence to use also a URL out of the internet should also work.

A lot is already there, but it would need a rework for sure. 

My constraint is the time to do it at the moment

henne49, 2023-03-22:
we are working on it for
grid
pvinverter
acload/genmeter

temp or solarmeter, would be later

stundenblume, 2023-03-23:
Hello Henne,
thank you for considering supporting these services.
I checked your code but I'm not very good in understanding code from someone else. :-/
I have realized my requirements with tools I can handle. In this case node-RED and the dbus-mqtt driver.

But I will try your code on a test system and will give feedback.

Sebastian