henne49/dbus-opendtu #52: Feature request: register device as a "ac load"

stxShadow, 2023-03-22:
Hi ! 
you've integrated tasmota from my last feature request. Thanks a again and i owe you a video.
I've just integrated a split air-con unit to my house and would like to use the tasmota not as a pvinverter but as a "ac load" meter.
Do you think that is easy to implement (and worth to implement) ? I was not able to find anything like this on github.
Thanks a lot 
Jens

stxShadow, 2023-03-22:
as far as i see you only have to register "com.victronenergy.acload" at line 617 right ?

henne49, 2023-03-22:
Thank you for owing us a video, I have great help with @dsteinkopf. 

There is no acload in dbus specification only genset or grid: 

https://github.com/victronenergy/venus/wiki/dbus#grid-and-genset-meter

I was unable to create one last time, it would also help with request #50

Will give it another try

henne49, 2023-03-22:
we are working on it for
grid
pvinverter
acload/genmeter

stxShadow, 2023-03-23:
> Thank you for owing us a video, I have great help with @dsteinkopf.
>
> There is no acload in dbus specification only genset or grid: [...]

Its working for me now. Just changed line 617 to acload and i get:

[image]

Setting it in config.ini would be very nice.

stundenblume, 2023-03-23:
@stxShadow
while this driver is very comfortable, you can also have a look for the [dbus-mqtt](https://github.com/freakent/dbus-mqtt-devices) driver. It can create any dbus service via MQTT. For some devices it may be more difficult to write to the correct topics. But with HA, ioBroker, node-RED, ... it's easy.
I was able to get a Gosund SP1 via RULES act like a pvinverter or evcharger. I also display solar forecast data as temperature in VRM. In the [wiki](https://github.com/freakent/dbus-mqtt-devices/wiki) I have collected some Tasmota examples.

henne49, 2023-03-23:
> Its working for me now. Just changed line 617 to acload and i get:
>
> [image]
>
> Setting it in config.ini would be very nice.

The idea is to change only in settings.ini, we just need some time to make it flexible and properly software engineered. 

@stundenblume can you share your working mqtt settings? if they are different then the ones you shared from the wiki, we will base it on these working examples.

henne49, 2023-03-23:
https://github.com/henne49/dbus-opendtu/tree/template-gridmeter%2C-acload

is the testing branch working now for acload and pvinverter from config.ini. 

grid, needs some more work

stundenblume, 2023-03-23:
The driver will create a dbus-grid-service (is that the correct designation?) and will publish the device-id created from the victron system.
I than send via MQTT all values into the corresponding topics. `W/[VRM ID]/grid/[device-id]/Ac/...`

Is that the information you need?

stundenblume, 2023-03-23:
`N/[portal ID]/grid/1/Role`= `{"value": "grid"}`

stundenblume, 2023-03-24:
I will test gridmeter tomorrow.

dsteinkopf, 2023-03-25:
Gridmeter will be a bit more difficult as we currently assume, that every device uses only one phase (L1, L2 _or_ L3). Gridmeter should supply all 3 phases.

dsteinkopf, 2023-03-25:
...but as I am also using a Grid meter (https://github.com/dsteinkopf/venus.dbus-MqttToGridMeter/), I think it could be a good idea to be integrated here.

stundenblume, 2023-03-25:
With `PhaseConfig`=`3` (0=3P.n; 1=3P.l; 2=2P; 3=1P; 4=3P) it should be fine.
But for bigger (normal) installations it would be necessary to write to all phases.

stundenblume, 2023-03-25:
$ /data/dbus-opendtu/uninstall.sh
    $ rm -r /data/dbus-opendtu/
    $ reboot
    $ wget https://github.com/henne49/dbus-opendtu/archive/refs/heads/template-gridmeter,-acload.zip
    $ unzip template-gridmeter,-acload.zip "dbus-opendtu-template-gridmeter-acload/*" -d /data
    $ nano /data/dbus-opendtu-template-gridmeter-acload/config.ini
    $ /data/dbus-opendtu-template-gridmeter-acload/install.sh

[image]

The "grid" service is not running.

http://192.168.123.105/cm?cmnd=STATUS+8
```
{
  "StatusSNS": {
    "Time": "2023-03-25T11:24:56",
    "COUNTER": {
      "C1": 1073963
    },
    "SML": {
      "Total_in": 1588.235,
      "Total_out": 459.685,
      "Power_curr": 595,
      "Volt_p1": 231,
      "Volt_p2": 233.5,
      "Volt_p3": 234,
      "Amperage_p1": 0.9,
      "Amperage_p2": 0.4,
      "Amperage_p3": 4.4,
      "Power_curr_p1": -110,
      "Power_curr_p2": 10,
      "Power_curr_p3": 694,
      "frequency": 50
    }
  }
}
```

config.ini:
```
GNU nano 4.9.3                                                                                                                                                                                                                                                                                                                                                                                                                                                    /data/dbus-opendtu-template-gridmeter-acload/config.ini                                                                                                                                                                                                                                                                                                                                                                                                                                                              
[DEFAULT]
# Time in minutes how often a status is added to the log-file current.log with log-level INFO
SignOfLifeLog= 1

# Number ob Template Inverter to query
NumberOfTemplates=1

# Which DTU to be used ahoy, opendtu, template
DTU=ahoy

# send YieldDay instead of YieldTotal
useYieldDay=0

#For ESP8266 reduce polling intervall to reduce load
ESP8266PollingIntervall=10000

#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
#To keep current.log small use ERROR
Logging=ERROR

# if ts_last_success is older than this number of seconds, it is not used.
# Set this to < 0 to disable this check.
MagAgeTsLastSuccess=600

# if this is not 0, then no values are actually sent via dbus to vrm/venus.
DryRun=0

#IP of Device to query
Host=192.168.123.132

HTTPTimeout=2.5

# Username/Password leave empty if no authentication is required
Username =
Password =

### Only needed for OpenDTU and ahoy
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2
# 1st inverter
[INVERTER0]
Phase=L1
DeviceInstance=34
AcPosition=1

# 2nd inverter
[INVERTER1]
Phase=L2
DeviceInstance=35
AcPosition=0

# 3rd inverter
[INVERTER2]
Phase=L3
DeviceInstance=36
AcPosition=1

# 4th inverter
[INVERTER3]
Phase=L1
DeviceInstance=37
AcPosition=1

# 5th inverter
[INVERTER4]
Phase=L2
DeviceInstance=38
AcPosition=1

# 6th inverter
[INVERTER5]
Phase=L3
DeviceInstance=39
AcPosition=1

# 7th inverter
[INVERTER6]
Phase=L1
DeviceInstance=40
AcPosition=1

# 8th inverter
[INVERTER7]
Phase=L2
DeviceInstance=41
AcPosition=1

# 9th inverter
[INVERTER8]
Phase=L3
DeviceInstance=42
AcPosition=1

# 10th inverter
[INVERTER9]
Phase=L1
DeviceInstance=43
AcPosition=1

################## TEMPLATES #####################
####Rearrange and customite Templates as necessary
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2

[TEMPLATE0]
## Tasmota Example
##
## Username/Password leave empty if no authentication is required
Username =
Password =
Host=192.168.123.105
CUST_SN = 12345678
CUST_API_PATH= cm?cmnd=STATUS+8
CUST_POLLING = 5000
CUST_Total= StatusSNS/SML/Total_in
CUST_Total_Mult = 1
CUST_Power= StatusSNS/SML/Power_curr_p1
CUST_Power_Mult = 1
CUST_Voltage=  StatusSNS/SML/Vol_p1
CUST_Current=  StatusSNS/SML/Amperage_p1
Phase=L1
DeviceInstance=47
AcPosition=1
Name= Tasmota
Servicename=com.victronenergy.grid

[TEMPLATE1]
## Shelly Gen 1 authenticated and unathenticated example
## Documentation: https://shelly-api-docs.shelly.cloud/gen1/#shelly1-1pm-status
##
## Username/Password leave empty if no authentication is required
Username =
Password =
Host=172.16.1.1
CUST_SN = 12345678
CUST_API_PATH= status
CUST_POLLING = 2000
CUST_Total= meters/0/total
###Shelly1PM Multiplier Watt/min 0.000017
CUST_Total_Mult = 0.000017
CUST_Power= meters/0/power
CUST_Power_Mult = 1
CUST_Voltage= none
CUST_Current= none
Phase=L2
DeviceInstance=45
AcPosition=1
Name= Shelly 1PM
Servicename=com.victronenergy.genset

[TEMPLATE2]
## Shelly Gen 2 - Only works unauthenticated!
## Documentation https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/EM#status
##
## Username/Password leave empty if no authentication is required
Host=172.16.1.1
CUST_SN = 12345678
CUST_API_PATH= rpc/Switch.GetStatus?id=0
CUST_POLLING = 2000
CUST_Total= aenergy/total
CUST_Total_Mult = 1
CUST_Power= apower
CUST_Power_Mult = 1
CUST_Voltage=  voltage
CUST_Current=  current
Phase=L3
DeviceInstance=46
AcPosition=1
Name= ShellyPlus1PM
Servicename=com.victronenergy.pvinverter
```

stundenblume, 2023-03-25:
also with this config.ini (NumberOfTemplates=2) it isn't created:
```
GNU nano 4.9.3                                                                                                                                                                                                                                                                                                                                                                                                                                                    /data/dbus-opendtu-template-gridmeter-acload/config.ini                                                                                                                                                                                                                                                                                                                                                                                                                                                              
[DEFAULT]
# Time in minutes how often a status is added to the log-file current.log with log-level INFO
SignOfLifeLog= 1

# Number ob Template Inverter to query
NumberOfTemplates=2

# Which DTU to be used ahoy, opendtu, template
DTU=ahoy

# send YieldDay instead of YieldTotal
useYieldDay=0

#For ESP8266 reduce polling intervall to reduce load
ESP8266PollingIntervall=10000

#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
#To keep current.log small use ERROR
Logging=ERROR

# if ts_last_success is older than this number of seconds, it is not used.
# Set this to < 0 to disable this check.
MagAgeTsLastSuccess=600

# if this is not 0, then no values are actually sent via dbus to vrm/venus.
DryRun=0

#IP of Device to query
Host=192.168.123.132

HTTPTimeout=2.5

# Username/Password leave empty if no authentication is required
Username =
Password =

### Only needed for OpenDTU and ahoy
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2
# 1st inverter
[INVERTER0]
Phase=L1
DeviceInstance=34
AcPosition=1

################## TEMPLATES #####################
####Rearrange and customite Templates as necessary
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2

[TEMPLATE0]
## Tasmota Example
##
## Username/Password leave empty if no authentication is required
Username =
Password =
Host=192.168.123.105
CUST_SN = 12345678
CUST_API_PATH= cm?cmnd=STATUS+8
CUST_POLLING = 5000
CUST_Total= StatusSNS/SML/Total_in
CUST_Total_Mult = 1
CUST_Power= StatusSNS/SML/Power_curr_p1
CUST_Power_Mult = 1
CUST_Voltage=  StatusSNS/SML/Vol_p1
CUST_Current=  StatusSNS/SML/Amperage_p1
Phase=L1
DeviceInstance=47
AcPosition=1
Name= Tasmota
Servicename=com.victronenergy.grid
```

stundenblume, 2023-03-25:
I got:
```
root@raspberrypi2:~# svstat /service/dbus-opendtu
/service/dbus-opendtu: unable to chdir: file does not exist
```
after `mv /data/dbus-opendtu-template-gridmeter-acload /data/dbus-opendtu` and `/data/dbus-opendtu/install.sh` I get:
```
root@raspberrypi2:~# svstat /service/dbus-opendtu
/service/dbus-opendtu: up (pid 22514) 1 seconds
```
It's always very low seconds.

`/data/dbus-opendtu/current.log`does not exist.

dsteinkopf, 2023-03-25:
We need the see the logs. Try `tail -F /data/dbus-opendtu/current.log` (capital -F).

If that does not show the logs, try this:

```
/data/dbus-opendtu/install.sh
python3 /data/dbus-opendtu/dbus-opendtu.py
```
...and send us the output.

I assume that for gridmeter, is not sufficient to send some different "initializing values" (lines 126 ff.). My working grid meter script (https://github.com/dsteinkopf/venus.dbus-MqttToGridMeter/blob/master/MQTTtoGridMeter.py) does send some different values (lines 127ff there) which I never understood and only got it working bases on trial and error.

Are you able to do some experiments and find working values? Then we could implement some config.ini-entries which make the script send these entries... just and idea...

dsteinkopf, 2023-03-25:
@henne49 I suggest finishing THIS issue (acload - configurable servicename) now and try to get grid to work in a different issue.

henne49, 2023-03-25:
100% can you review and let me know, when to merge?

dsteinkopf, 2023-03-25:
@stxShadow please reopen or create a new issue if your request is not (completely) fulfilled.
@stundenblume Please open a new issue about the grid meter problem containing your new observations and results.
Thank you.

closed 2023-03-25 by dsteinkopf