henne49/dbus-opendtu #56: grid service not created

stundenblume, 2023-03-25:
This is a follow up from [here](https://github.com/henne49/dbus-opendtu/issues/52#issuecomment-1483780654).

- pvinverter is created as expected.
- no grid service created.
- `current.log`is empty.
- `/data/dbus-opendtu/uninstall.sh` and `python3 /data/dbus-opendtu/dbus-opendtu.py` gives no output.

dsteinkopf, 2023-03-25:
`python3 /data/dbus-opendtu/dbus-opendtu.py` starts python running the named script. So you should see some console output - at least an error or so. If not, then something is completely wrong.

stundenblume, 2023-03-25:
```
root@raspberrypi2:~# /data/dbus-opendtu/uninstall.sh 
root@raspberrypi2:~# rm -r /data/dbus-opendtu/
root@raspberrypi2:~# unzip main.zip "dbus-opendtu-main/*" -d /data
root@raspberrypi2:~# mv /data/dbus-opendtu-main /data/dbus-opendtu
root@raspberrypi2:~# nano /data/dbus-opendtu/config.ini 
root@raspberrypi2:~# python3 /data/dbus-opendtu/dbus-opendtu.py
```
http://192.168.123.105/cm?cmnd=STATUS+8
```
{
  "StatusSNS": {
    "Time": "2023-03-25T11:24:56",
    "COUNTER": {
      "C1": 1073963
    },
    "SML": {
      "Total_in": 1588.235,
      "Total_out": 459.685,
      "Power_curr": 595,
      "Volt_p1": 231,
      "Volt_p2": 233.5,
      "Volt_p3": 234,
      "Amperage_p1": 0.9,
      "Amperage_p2": 0.4,
      "Amperage_p3": 4.4,
      "Power_curr_p1": -110,
      "Power_curr_p2": 10,
      "Power_curr_p3": 694,
      "frequency": 50
    }
  }
}
```

config.ini:
```
[DEFAULT]
# Time in minutes how often a status is added to the log-file current.log with log-level INFO
SignOfLifeLog= 1

# Number ob Template Inverter to query
NumberOfTemplates=1

# Which DTU to be used ahoy, opendtu, template
DTU=ahoy

# send YieldDay instead of YieldTotal
useYieldDay=0

#For ESP8266 reduce polling intervall to reduce load
ESP8266PollingIntervall=10000

#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
#To keep current.log small use ERROR
Logging=ERROR

# if ts_last_success is older than this number of seconds, it is not used.
# Set this to < 0 to disable this check.
MagAgeTsLastSuccess=600

# if this is not 0, then no values are actually sent via dbus to vrm/venus.
DryRun=0

#IP of Device to query
Host=192.168.123.132

HTTPTimeout=2.5

# Username/Password leave empty if no authentication is required
Username =
Password =

### Only needed for OpenDTU and ahoy
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2
# 1st inverter
[INVERTER0]
Phase=L1
DeviceInstance=34
AcPosition=1

# 2nd inverter
[INVERTER1]
Phase=L2
DeviceInstance=35
AcPosition=0

# 3rd inverter
[INVERTER2]
Phase=L3
DeviceInstance=36
AcPosition=1

# 4th inverter
[INVERTER3]
Phase=L1
DeviceInstance=37
AcPosition=1

# 5th inverter
[INVERTER4]
Phase=L2
DeviceInstance=38
AcPosition=1

# 6th inverter
[INVERTER5]
Phase=L3
DeviceInstance=39
AcPosition=1

# 7th inverter
[INVERTER6]
Phase=L1
DeviceInstance=40
AcPosition=1

# 8th inverter
[INVERTER7]
Phase=L2
DeviceInstance=41
AcPosition=1

# 9th inverter
[INVERTER8]
Phase=L3
DeviceInstance=42
AcPosition=1

# 10th inverter
[INVERTER9]
Phase=L1
DeviceInstance=43
AcPosition=1

################## TEMPLATES #####################
####Rearrange and customite Templates as necessary
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2

[TEMPLATE0]
## Tasmota Example
##
## Username/Password leave empty if no authentication is required
Username =
Password =
Host=192.168.123.105
CUST_SN = 12345678
CUST_API_PATH= cm?cmnd=STATUS+8
CUST_POLLING = 5000
CUST_Total= StatusSNS/SML/Total_in
CUST_Total_Mult = 1
CUST_Power= StatusSNS/SML/Power_curr_p1
CUST_Power_Mult = 1
CUST_Voltage=  StatusSNS/SML/Vol_p1
CUST_Current=  StatusSNS/SML/Amperage_p1
Phase=L1
DeviceInstance=47
AcPosition=0
Name= Tasmota
Servicename=com.victronenergy.grid

[TEMPLATE1]
## Shelly Gen 1 authenticated and unathenticated example
## Documentation: https://shelly-api-docs.shelly.cloud/gen1/#shelly1-1pm-status
##
## Username/Password leave empty if no authentication is required
Username =
Password =
Host=172.16.1.1
CUST_SN = 12345678
CUST_API_PATH= status
CUST_POLLING = 2000
CUST_Total= meters/0/total
###Shelly1PM Multiplier Watt/min 0.000017
CUST_Total_Mult = 0.000017
CUST_Power= meters/0/power
CUST_Power_Mult = 1
CUST_Voltage= none
CUST_Current= none
Phase=L2
DeviceInstance=45
AcPosition=1
Name= Shelly 1PM
Servicename=com.victronenergy.genset

[TEMPLATE2]
## Shelly Gen 2 - Only works unauthenticated!
## Documentation https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/EM#status
##
## Username/Password leave empty if no authentication is required
Host=172.16.1.1
CUST_SN = 12345678
CUST_API_PATH= rpc/Switch.GetStatus?id=0
CUST_POLLING = 2000
CUST_Total= aenergy/total
CUST_Total_Mult = 1
CUST_Power= apower
CUST_Power_Mult = 1
CUST_Voltage=  voltage
CUST_Current=  current
Phase=L3
DeviceInstance=46
AcPosition=1
Name= ShellyPlus1PM
Servicename=com.victronenergy.pvinverter
```

stundenblume, 2023-03-25:
I don't get any output.

If I interrupt the script with `ctl+c`I get:
```
root@raspberrypi2:~# python3 /data/dbus-opendtu/dbus-opendtu.py
^CTraceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 733, in <module>
    main()
  File "/data/dbus-opendtu/dbus-opendtu.py", line 639, in main
    mainloop.run()
  File "/usr/lib/python3.8/site-packages/gi/overrides/GLib.py", line 497, in run
    super(MainLoop, self).run()
  File "/usr/lib/python3.8/contextlib.py", line 120, in __exit__
    next(self.gen)
  File "/usr/lib/python3.8/site-packages/gi/_ossighelper.py", line 251, in register_sigint_fallback
    signal.default_int_handler(signal.SIGINT, None)
KeyboardInterrupt
```

henne49, 2023-03-26:
please read: https://github.com/henne49/dbus-opendtu/blob/main/README.md#troubleshooting

You have to change Logging=ERROR to Logging=DEBUG to see more

stundenblume, 2023-03-26:
Thanks for the hint.
I did everything on a fresh install.

Now it registered the grid service:
```
root@raspberrypi2:~# python3 /data/dbus-opendtu/dbus-opendtu.py
2023-03-26 06:40:04,929 root INFO Start
2023-03-26 06:40:05,116 root INFO Number of Inverters found: 1
2023-03-26 06:40:05,129 root INFO registered ourselves on D-Bus as com.victronenergy.pvinverter.http_34
2023-03-26 06:40:05,132 root INFO Name of Inverters found: Sven
2023-03-26 06:40:05,135 root INFO ESP8266 detected, reducing polling to 10000
2023-03-26 06:40:05,142 root INFO Number of Inverters found: 1
2023-03-26 06:40:05,143 root INFO Registering Templates
2023-03-26 06:40:05,175 root INFO registered ourselves on D-Bus as com.victronenergy.grid.http_47
2023-03-26 06:40:05,178 root INFO Name of Inverters found: Tasmota
2023-03-26 06:40:05,186 root INFO Connected to dbus, and switching over to gobject.MainLoop() (= event based)
```
[image]
[image]

stundenblume, 2023-03-26:
One thing that does not work probably is the representation and calculation in VRM portal.
If I do nothing the AC-IN is empty. Even if the representation in the remote console is correct.
[image]

After setting `N/b827eb4961c3/system/0/Ac/ActiveIn/Source` from `null` to `1` the representation works, but not the calculation.
Notice the leading `N/`. With `W/` it's not possible to change the value. After a while it will be reset to `null`.

[image]
 
It's the same with the dbus-mqtt driver. The diagrams show always the same for "Solar" and "Consumption". "From/To Grid" is always zero.
[image]

stundenblume, 2023-03-26:
Maybe it works as expected... I will report later this day.
[image]

stundenblume, 2023-03-27:
Good morning,
your driver works.
Thank you very much!
The only thing that left be open:
 - [3 phase support](https://github.com/henne49/dbus-opendtu/issues/59)
 -  correct representation/calculation of "to Grid". But that is not directly related to your driver.
 
[image]
[image]