henne49/dbus-opendtu #57: Dienst startet nicht

AndiJot, 2023-03-26:
Here is my config:

[DEFAULT]
# Time in minutes how often a status is added to the log-file current.log with log-level INFO
SignOfLifeLog= 1

# Number ob Template Inverter to query
NumberOfTemplates=0

# Which DTU to be used ahoy, opendtu, template
DTU=ahoy
# send YieldDay instead of YieldTotal
useYieldDay=0

#For ESP8266 reduce polling intervall to reduce load
ESP8266PollingIntervall=10000

#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
#To keep current.log small use ERROR
Logging=ERROR

# if ts_last_success is older than this number of seconds, it is not used.
# Set this to < 0 to disable this check.
MagAgeTsLastSuccess=600

# if this is not 0, then no values are actually sent via dbus to vrm/venus.
DryRun=0

#IP of Device to query
Host=192.168.178.56

HTTPTimeout=2.5

# Username/Password leave empty if no authentication is required
Username =
Password =

### Only needed for OpenDTU and ahoy
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2
# 1st inverter
[INVERTER0]
Phase=L1
DeviceInstance=34
AcPosition=1

Here is a link to my problem : https://youtube.com/shorts/sVrybaGSI4A

What can i do?

henne49, 2023-03-27:
**Follow the Troubleshooting Section**
Change the Logging Parameter under DEFAULT in /data/dbus-opendtu/config.ini to Logging = DEBUG, please revert, once debugging and troubleshooting is complete. Rerun the script and share the current.log and config.ini file.
[Troubleshooting Section](https://github.com/henne49/dbus-opendtu#troubleshooting )

AndiJot, 2023-03-27:
thank you for the answer.
Here is the current.log

2023-03-26 00:30:14,839 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 608, in main
    service = DbusService(
  File "/data/dbus-opendtu/dbus-opendtu.py", line 149, in __init__
    self._dbusservice.add_path("/Serial", self._get_serial(self.pvinverternumber))
  File "/data/dbus-opendtu/dbus-opendtu.py", line 242, in _get_serial
    raise ValueError("Response does not contain name")
ValueError: Response does not contain name
2023-03-26 00:30:18,90 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 608, in main
    service = DbusService(
  File "/data/dbus-opendtu/dbus-opendtu.py", line 149, in __init__
    self._dbusservice.add_path("/Serial", self._get_serial(self.pvinverternumber))
  File "/data/dbus-opendtu/dbus-opendtu.py", line 242, in _get_serial
    raise ValueError("Response does not contain name")
ValueError: Response does not contain name
2023-03-26 00:30:21,252 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 608, in main
    service = DbusService(
  File "/data/dbus-opendtu/dbus-opendtu.py", line 149, in __init__
    self._dbusservice.add_path("/Serial", self._get_serial(self.pvinverternumber))
  File "/data/dbus-opendtu/dbus-opendtu.py", line 242, in _get_serial
    raise ValueError("Response does not contain name")
ValueError: Response does not contain name
2023-03-26 00:30:24,427 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 608, in main
    service = DbusService(
  File "/data/dbus-opendtu/dbus-opendtu.py", line 149, in __init__
    self._dbusservice.add_path("/Serial", self._get_serial(self.pvinverternumber))
  File "/data/dbus-opendtu/dbus-opendtu.py", line 242, in _get_serial
    raise ValueError("Response does not contain name")
ValueError: Response does not contain name
2023-03-26 00:30:27,718 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 608, in main
    service = DbusService(
  File "/data/dbus-opendtu/dbus-opendtu.py", line 149, in __init__
    self._dbusservice.add_path("/Serial", self._get_serial(self.pvinverternumber))
  File "/data/dbus-opendtu/dbus-opendtu.py", line 242, in _get_serial
    raise ValueError("Response does not contain name")
ValueError: Response does not contain name
2023-03-26 00:30:30,988 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 608, in main
:

henne49, 2023-03-27:
I guess you do not have a name in ahoy specified: 
raise ValueError("Response does not contain name")

closed 2023-04-06 by henne49