henne49/dbus-opendtu #59: Feature request: Support for 3 phase "grid" service

stundenblume, 2023-03-27:
It would be nice if your driver would support 3 phase gridmeter.

The benefit would be, that no EM24 would be necessary. A Tasmota device connected to a smartmeter would be enough.

dsteinkopf, 2023-04-09:
I would like to start by implementing this as a replacement for my own https://github.com/dsteinkopf/venus.dbus-MqttToGridMeter/ which is not really not very sophisticated... but it is stable. Nevertheless, it would be a nice thing if I could also collect grid meter values in this one script here.

First exercise: How to get the grid meter values? In my case, I am collecting them via home assistant edl21 integration. So to have a fit for our "templates" I fetch values via API:

(Entitiy names are only valid for my installation of course.)

`curl --silent --header "Authorization: Bearer $HA_TOKEN" --header Content-Type:\ application/json http://192.168.40.227:8123/api/states/sensor.hausstrom_sum_active_instantaneous_power` | jq
-> 
```
{
  "entity_id": "sensor.hausstrom_sum_active_instantaneous_power",
  "state": "5",
  "attributes": {
    "state_class": "measurement",
    "val_time": [
      [
        2,
        1
      ],
      [
        5,
        34215552
      ]
    ],
    "unit_of_measurement": "W",
    "device_class": "power",
    "friendly_name": "Hausstrom Sum active instantaneous power"
  },
  "last_changed": "2023-04-09T07:10:17.428600+00:00",
  "last_updated": "2023-04-09T07:10:17.428600+00:00",
  "context": {
    "id": "01GXJDKWAMPWSCQ69W0ZFY8H9W",
    "parent_id": null,
    "user_id": null
  }
}
```

L1: /api/states/sensor.hausstrom_l1_active_instantaneous_power
L2: /api/states/sensor.hausstrom_l2_active_instantaneous_power
L3: /api/states/sensor.hausstrom_l3_active_instantaneous_power
Sum: /api/states/sensor.hausstrom_sum_active_instantaneous_power

dsteinkopf, 2023-04-09:
**First challenge**: How to configure everything in config.ini? 

First idea:
```
[TEMPLATE]                                                                                                                                            
Host=192.168.40.244                                                                                                                                   
My_Values_To_Poll=L1_Current,L2_Current,L3_Current,L1_Power,L2_Power,L3_Power                                                                         
CUST_SN = 12345678                                                                                                                                    
# ??? CUST_API_PATH= cm?cmnd=STATUS+8                                                                                                                 
CUST_POLLING = 1000                                                                                                                                   
                                                                                                                                                      
My_Value_L1_Power_API_PATH = api/states/sensor.hausstrom_l1_active_instantaneous_power                                                                
My_Value_L2_Power_API_PATH = api/states/sensor.hausstrom_l2_active_instantaneous_power                                                                
My_Value_L3_Power_API_PATH = api/states/sensor.hausstrom_l3_active_instantaneous_power                                                                
                                                                                                                                                      
My_Value_L1_Power = status # JSONPath                                                                                                                 
My_Value_L2_Power = status # JSONPath                                                                                                                 
My_Value_L3_Power = status # JSONPath                                                                                                                 
                                                                                                                                                      
My_Value_L1_Power_DBUS_PATH = /Ac/L1/Power                                                                                                            
My_Value_L2_Power_DBUS_PATH = /Ac/L2/Power                                                                                                            
My_Value_L3_Power_DBUS_PATH = /Ac/L3/Power                                                                                                            
                                                                                                                                                      
CUST_Total_API_PATH = api/states/sensor.hausstrom_positive_active_energy_total                                                                        
CUST_Total=status # JSONPath                                                                                                                          
CUST_Total_Mult = 1                                                                                                                                   
                                                                                                                                                      
CUST_Power_API_PATH = api/states/sensor.hausstrom_sum_active_instantaneous_power                                                                      
CUST_Power = status                                                                                                                                   
CUST_Power_Mult = 1                                                                                                                                   
                                                                                                                                                      
CUST_Voltage=230 # fixed value                                                                                                                        
                                                                                                                                                      
# not needed CUST_Current=  StatusSNS/ENERGY/Current                                                                                                  
```

Still missing/open: 
How to configure calculated values? 
The "Currents" are not provided in my case and must be calculated: 230 / power
/Ac/Energy/Forward and /Ac/Energy/Reverse must be derived by distinguishing between negative and positive energy values. But I don't get negative energy values from my meter...

Jaaxy, 2023-10-29:
Hiho, i have scripted a Tasmota Script for different gridMeter. Here is the Ouput off the EM24.

{
    "StatusSNS": {
        "Time": "2023-10-29T19:59:41",
        "EM24": {
            "Voltage_L1": 229.4,
            "Voltage_L2": 229.4,
            "Voltage_L3": 228.3,
            "Current_L1": 1.346,
            "Current_L2": 1.774,
            "Current_L3": 0.298,
            "Power_L1": 246.3,
            "Power_L2": 327.4,
            "Power_L3": 55.6,
            "Power_va_L1": 311.8,
            "Power_va_L2": 395.0,
            "Power_va_L3": 70.0,
            "Power_var_L1": -183.1,
            "Power_var_L2": -236.4,
            "Power_var_L3": -42.4,
            "Energy_Demand_W_max": 4078.2,
            "Energy_Demand_W": 599.8,
            "Energy_Demand_VA_max": 4116.5,
            "Energy_Demand_VA": 757.7,
            "Power_L1_p": 861.1,
            "Power_L2_p": 1829.1,
            "Power_L3_p": 882.4,
            "Energy_Tot_Import": 3573.3,
            "Energy_Tot_Export": 0.0
        }
    }
}

henne49, 2023-10-29:
Can you give some more infos background on your json?

Jaaxy, 2023-10-30:
I need an example to display the Json in the cerbo.

The EM24 is a gridmeter logging the power off a part of the house. 

At this Moment I use 3 Templates Sonoff:
[image]

[image]

I want to take it in one Template with displaying all measures and summary