henne49/dbus-opendtu #64: no connection DTU <- > Venus

Gluckennest, 2023-03-31:
Hello

I am struggling now the two days to integrate the DTU in venus without success. Hope you can help me

Venus System **Cerbo GX**/ **FW: v2.93**

According to the instructions I have obviously installed everything correctly. Host entered etc. I do not understand why I do not get a connection.

[image]

config:
->also testet without username/ password
[DEFAULT]
# Time in minutes how often a status is added to the log-file current.log with log-level INFO
SignOfLifeLog= 1

# Number ob Template Inverter to query
NumberOfTemplates=1

# Which DTU to be used ahoy, opendtu, template
DTU=opendtu

# send YieldDay instead of YieldTotal
.useYie.ldDay=2

#For ESP8266 reduce polling intervall to reduce load
ESP8266PollingIntervall=10000

#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
#To keep current.log small use ERROR
Logging=DEBUG

# if ts_last_success is older than this number of seconds, it is not used.
# Set this to < 0 to disable this check.
MagAgeTsLastSuccess=600

# if this is not 0, then no values are actually sent via dbus to vrm/venus.
DryRun=0

#IP of Device to query
Host=192.168.1.180

HTTPTimeout=2.5

# Username/Password leave empty if no authentication is required
Username =admin
Password =openDTU42

### Only needed for OpenDTU and ahoy
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2
# 1st inverter
[HM1500L]
Phase=L1
DeviceInstance=34
AcPosition=0
Servicename=com.victronenergy.charger

# 2nd inverter
[HM1500R]
Phase=L2
DeviceInstance=35
AcPosition=0

[image]

Logging 

2023-03-31 10:16:55,920 root DEBUG is_data_up2date: inverter #0: age_seconds=5014437, max_age_ts=600
2023-03-31 10:16:55,922 root DEBUG is_data_up2date: inverter #0: age_seconds=10, max_age_ts=600
2023-03-31 10:16:55,924 root INFO Start
2023-03-31 10:16:55,935 root INFO Registering dtu devices
2023-03-31 10:16:55,937 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 612, in main
    servicename=get_config_value(config, "Servicename", "INVERTER", 0, "com.victronenergy.pvinverter")
  File "/data/dbus-opendtu/dbus-opendtu.py", line 61, in get_config_value
    if name in config[f"{inverter_or_template}{pvinverternumber}"]:
  File "/usr/lib/python3.8/configparser.py", line 960, in __getitem__
    raise KeyError(key)
KeyError: 'INVERTER0'
2023-03-31 10:16:59,613 root DEBUG is_data_up2date: inverter #0: age_seconds=5014441, max_age_ts=600
2023-03-31 10:16:59,616 root DEBUG is_data_up2date: inverter #0: age_seconds=10, max_age_ts=600
2023-03-31 10:16:59,621 root INFO Start
2023-03-31 10:16:59,633 root INFO Registering dtu devices
2023-03-31 10:16:59,635 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 612, in main
    servicename=get_config_value(config, "Servicename", "INVERTER", 0, "com.victronenergy.pvinverter")
  File "/data/dbus-opendtu/dbus-opendtu.py", line 61, in get_config_value
    if name in config[f"{inverter_or_template}{pvinverternumber}"]:
  File "/usr/lib/python3.8/configparser.py", line 960, in __getitem__
    raise KeyError(key)
KeyError: 'INVERTER0'
2023-03-31 10:17:03,501 root DEBUG is_data_up2date: inverter #0: age_seconds=5014445, max_age_ts=600
2023-03-31 10:17:03,504 root DEBUG is_data_up2date: inverter #0: age_seconds=10, max_age_ts=600
2023-03-31 10:17:03,505 root INFO Start
2023-03-31 10:17:03,518 root INFO Registering dtu devices
2023-03-31 10:17:03,519 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 612, in main
    servicename=get_config_value(config, "Servicename", "INVERTER", 0, "com.victronenergy.pvinverter")
  File "/data/dbus-opendtu/dbus-opendtu.py", line 61, in get_config_value
    if name in config[f"{inverter_or_template}{pvinverternumber}"]:
  File "/usr/lib/python3.8/configparser.py", line 960, in __getitem__
    raise KeyError(key)
KeyError: 'INVERTER0'

henne49, 2023-03-31:
a few questions to your config.ini

why do you have NumberOfTemplates=1 but no template?
why do you have .useYie.ldDay=2 that should be useYieldDay=0 or useYieldDay=1

and please uncomment line Servicename=com.victronenergy.charger or delete it please, this was wrong in the latest config.ini and try again.

henne49, 2023-03-31:
I would also not recommend to use opendtu with password protection for getting live data, therefore
Username =
Password =

and use authentication like this: 

https://github.com/henne49/dbus-opendtu#security

everyone can read the plaint text password so not using username password is mor esecure for polling

Gluckennest, 2023-03-31:
Always the same, no goal in this case. It does not connect or start the service
I change all in the config.ini
I have no idea whats going wrong

# Time in minutes how often a status is added to the log-file current.log with log-level INFO
SignOfLifeLog= 1

# Number ob Template Inverter to query
NumberOfTemplates=0

# Which DTU to be used ahoy, opendtu, template
DTU=opendtu

# send YieldDay instead of YieldTotal
useYieldDay=0

#For ESP8266 reduce polling intervall to reduce load
ESP8266PollingIntervall=10000

#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
#To keep current.log small use ERROR
Logging=DEBUG

# if ts_last_success is older than this number of seconds, it is not used.
# Set this to < 0 to disable this check.
MagAgeTsLastSuccess=600

# if this is not 0, then no values are actually sent via dbus to vrm/venus.
DryRun=0

#IP of Device to query
Host=192.168.1.180

HTTPTimeout=2.5

# Username/Password leave empty if no authentication is required
Username =
Password =

### Only needed for OpenDTU and ahoy
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2
# 1st inverter
[INVERTER0]
Phase=L1
DeviceInstance=34
AcPosition=0

# 2nd inverter
[INVERTER1]
Phase=L2
DeviceInstance=35
AcPosition=0

**current LOG**

KeyError: 'useYieldDay'
2023-03-31 11:32:48,352 root DEBUG is_data_up2date: inverter #0: age_seconds=5018990, max_age_ts=600
2023-03-31 11:32:48,354 root DEBUG is_data_up2date: inverter #0: age_seconds=10, max_age_ts=600
2023-03-31 11:32:48,356 root INFO Start
2023-03-31 11:32:48,367 root INFO Registering dtu devices
2023-03-31 11:32:48,394 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 613, in main
    service = DbusService(
  File "/data/dbus-opendtu/dbus-opendtu.py", line 111, in __init__
    self._read_config_dtu(actual_inverter)
  File "/data/dbus-opendtu/dbus-opendtu.py", line 180, in _read_config_dtu
    self.useyieldday = int(config["DEFAULT"]["useYieldDay"])
  File "/usr/lib/python3.8/configparser.py", line 1254, in __getitem__
    raise KeyError(key)
KeyError: 'useYieldDay'
2023-03-31 11:32:52,644 root DEBUG is_data_up2date: inverter #0: age_seconds=5018994, max_age_ts=600
2023-03-31 11:32:52,647 root DEBUG is_data_up2date: inverter #0: age_seconds=10, max_age_ts=600
2023-03-31 11:32:52,649 root INFO Start
2023-03-31 11:32:52,659 root INFO Registering dtu devices
2023-03-31 11:32:52,686 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 613, in main
    service = DbusService(
  File "/data/dbus-opendtu/dbus-opendtu.py", line 111, in __init__
    self._read_config_dtu(actual_inverter)
  File "/data/dbus-opendtu/dbus-opendtu.py", line 180, in _read_config_dtu
    self.useyieldday = int(config["DEFAULT"]["useYieldDay"])
  File "/usr/lib/python3.8/configparser.py", line 1254, in __getitem__
    raise KeyError(key)
KeyError: 'useYieldDay'

henne49, 2023-03-31:
it seems your config.ini is somehow corrupted. The error says useYieldDay is missing under [DEFAULT]

does your config.ini start with [DEFAULT] and instead of copying your config.ini, can you upload it, or use this one attached?

please rename config.txt to config.ini
[config.txt](https://github.com/henne49/dbus-opendtu/files/11121386/config.txt)

Gluckennest, 2023-03-31:
Yes correct. The config was corrupted. Now it works! Thanks for your support

hans771, 2023-03-31:
I also have no connection between VenusGX an AhoyDTU. 

_Ahoy WebIF is working fine:_

[image]

 _IP/api/live shows:_
[image]

_config.ini like:_
[image]

_Starting the service on my Venus GX shows "file exists" and Log in Debug do not work ->"permission denied":_
[image]

_dbus-spy do not show any of my 5 inverters:_
[image]

Same problems with ahoy version 0.5.11 an 0.6.0 - perhaps anyone can help me? Thanks a lot!

henne49, 2023-03-31:
Please do not report a different error in this issue. 

I guess your problem is this, we are working on a fix: https://github.com/henne49/dbus-opendtu/pull/54

Please use an older ahoy firmware, or with esp32 you can use opendtu instead of ahoy

closed 2023-03-31 by henne49