henne49/dbus-opendtu #68: Nulleinspeisung

henne49, 2023-04-06:
https://github.com/Selbstbau-PV/Selbstbau-PV-Hoymiles-nulleinspeisung-mit-OpenDTU-und-Shelly3EM/blob/main/nulleinspeisung.py

GallusMax, 2023-04-13:
Im Ernst? Alle 5s wird die DTU per API abgefragt? Das ist sportlich! Die DTU bekommt vielleicht alle 30s neue Werte von den Invertern, besonders wenn  mehrere abgefragt werden. 

Und das Setzen des  (der?) neuen Limits auch alle 5s?

Was passiert, wenn mehrere Inverter im System sind ?  
Überlegung: Wenn die begrenzte Gesamtleistung auf mehrere Inverter aufgeteilt wird, kann sie ggf. nicht mehr erreicht werden.

Noch ein Hinweis zu den  (Dummy) Ip Adressen: die liegen in Columbus, Ohio, und gelten als U.S. Military. :-)

henne49, 2023-04-13:
das ist ein Idee Reminder, mehr nicht, mal sehen wie und ob überhaupt das kommen soll

onceuponatime78, 2023-05-02:
Could it be an option to implement /Ac/PowerLimit according to https://github.com/victronenergy/venus/wiki/dbus#inverter to enable venus os to control the current output power of the inverter to achieve a zero export?

onceuponatime78, 2023-05-02:
Close to zero will be good enough for most use cases. Giving the venus OS control would on the one hand enable the control over how much power is exported and on the other hand also allow it to lower the power for the case the hoymiles inverter is connected to ac out and power is lost (without having to completely shut it down via frequency shift)

onemorename, 2023-05-30:
Well i have a pretty good use case here: When the SOC of the Battery reaches for example 98% reduce the inverter power to 10% (or whatever is enough for absorption / loading with low power). This ensures you're not exporting a high number of energy to the Grid in case this is not wanted and keeps charging balanced at the top. Running this via Home Assistant automation now, but if this could be controlled from the Cerbo directly this would be perfect

0x7878, 2023-05-31:
Hey @onemorename 
I think what you describe is the ESS feature "keep Battery charged". In this mode, the Multiplus would normally reduce the charging current till the battery reach 100%.
Furthermore, if set up, it won't feed into the grid and try to reduce any controllable inverter.

[image]

In your case, you would just reduce the power limit to not feed into the grid. 

However,  when we implement a "Multiplus" device that might be possible then.

VOPAtec, 2023-06-11:
Hallo zusammen,
diese Erweiterung fände ich auch hilfreich.
Wenn alle Wasserspeicher auf max Temp sind und immernoch zu viel Energie übrig ist, würde ich gerne automatisch abregeln können.
Mir würde ein 1 Minuten Takt vollkommen ausreichen. Auch 5 Minuten wären noch tragbar.
Bei wechselhaften Lichtverhältnissen ist die Herausforderung, dass ein dauerhaftes entladen des Akkus verhindert wird.

Ein Tolles Projekt. Vielen Dank.

henne49, 2023-06-13:
Vielen Dank erstmal, das freut uns!

Schau doch mal, ob nicht genau dazu das Projekt hier schon die Lösung ist: https://github.com/Selbstbau-PV/Selbstbau-PV-Hoymiles-nulleinspeisung-mit-OpenDTU-und-Shelly3EM/blob/main/nulleinspeisung.py

Wir können zwar einen Gridmeter anlegen, aber es kann ja auch sein, dass der Gridmeter kein Shelly ist oder eben auch nicht per JSON oder REST API abgefragt werden kann, daher müssten wir den dbus lesen. 

Dazu haben wir aber noch nichts gemacht. @0x7878  und @dsteinkopf was haltet ihr von der Idee eine Regelung in unserem Projekt einzubauen?

onceuponatime78, 2023-06-13:
ich hab es in Kombination mit https://github.com/reserve85/HoymilesZeroExport verwendet (diese Variante bietet mehr Flexibilität, beide Projekte wollen im Prinzip das gleiche erreichen). Beide haben den Nachteil, dass die Victron Laderegelung und die Nulleinspeisung über das zusätzliche Script zueinander konkurierend sind. Im Prinzip geht es so halbwegs, wenn man das nulleinspeisescript auf einen negativen Wert konfiguriert (z.B. -100W die "eingespeist werden dürfen), damit der Multiplus dann wiederum diesen Überschuss zum Laden nutzen kann - das Spiel kommt dann so lange halbwegs auf null raus, bis die Akkus voll sind. 

Ich würds für die Beste Lösung halten, wenn man dahin kommt mit der opendtu integration quasi zu den Funktionen zu kommen die Victron in Kombination mit Fronius und anderen sunspec kompatiblen Wechselrichtern erreichen kann - also das Venus OS selbst den Wechselrichter runterregeln kann um zu null zu kommen, bzw. im Falle von nicht vorhandenem Netz die Leistung reduzieren kann.

Soweit ich das in https://github.com/victronenergy/dbus-fronius/blob/master/software/src/inverter.cpp nachvollziehen kann, sollte es möglich sein dass auch mit der opendtu zu erreichen.

0x7878, 2023-06-13:
> 

Genau das hatte ich mir angesehen und versucht zu testen, aber mein System schreibt dort nichts rein. Ich denke, das wird nur aktiviert, wenn es sich wirklich um ein Fronius Wechselrichter handelt. Auch muss man im Fronius dann Modbus aktivieren.

```c++
// Enable the power limiter and initialise it to maxPower.
	if (deviceInfo.powerLimitScale)
``` 
sieht auch für mich danach aus, dass das device victron sagt "Hey ich bin limitierbar"

onceuponatime78, 2023-06-13:
ggf könnte der Ansatz von hier auch zum Ziel führen:
https://github.com/Telekatz/venus.dbus-hoymiles/blob/51bcb620a499445caf6cc5a2bd725da21de78c46/HMpvinverter.py#L1085

0x7878, 2023-06-14:
Danke dir für den link. Hilfreich ist es in jedem Fall. Er "emuliert" einen Multiplus und nutzt folgendes:

```text
Grid Power Setpoints - /Hub4/Lx/AcPowerSetpoint
This includes a path for each available phase
/Hub4/L1/AcPowerSetpoint (setpoint for L1)
/Hub4/L2/AcPowerSetpoint (setpoint for L2, where available)
/Hub4/L3/AcPowerSetpoint (setpoint for L3, where available)
When set to a positive value, power will be taken from the grid on phase x.
When set to a negative value, power will be fed into the grid on phase x.
``` 

Er schaut sich an, was Victron ins Netz schieben würde, um eine Nulleinspeisung zu realisieren und passt alle Parameter danach an. Deshalb funktioniert sein Code auch nicht, wenn bereits ein Multiplus im System ist. 

Ich wusste auch nicht, dass wir eigene Parameter ins dBus schieben können. Das macht die Sache einfacher. Werde da mal mit experimentieren :-)

dsteinkopf, 2023-06-14:
> Dazu haben wir aber noch nichts gemacht. @0x7878 und @dsteinkopf was haltet ihr von der Idee eine Regelung in unserem Projekt einzubauen?

Also ich finde ja tendenziell, dass das nicht in unser Projekt "passt". Es geht dabei ja darum, Daten von Geräten zu Victron zu schieben. Da sehe ich eigentlich keine Regelung. Das wäre m.E. etwas Anderes und auch sehr spezifisch für einzelne regelbare Geräte.

VOPAtec, 2023-06-14:
Hallo zusammen,
freut mich dass ihr daran auch Interesse habt.
Kurzfristig ist mein Plan, per Node-Red Zusatzverbraucher über ESPxx(xx) Module zu schalten. Wenn diese möglichkeit jedoch ausgeschöpft/voll/zu-gering ist, sollen sich einzelne HM1500 abregeln oder die Gesamtleistung verringert werden.

Der Start in eurem Projekt wäre ja erstmal die Abregelung von Ahoy/opendtu. Ich denke das passt noch ganz gut zum Projekt. Ihr würdet damit dem Fronius/Victron Verbund Konkurenz machen und es wäre noch ein Pluspunkt für die Hoymiles Wechselrichter.
Da ich über einfache Node-Red Verknüpfungen, programmiertechnisch nicht hinauskomme, freue ich mich über alles was kommt. 
Vielleicht ist es möglich eine Art "Add on" zu machen?

onceuponatime78, 2023-06-14:
> Genau das hatte ich mir angesehen und versucht zu testen, aber mein System schreibt dort nichts rein. Ich denke, das wird nur aktiviert, wenn es sich wirklich um ein Fronius Wechselrichter handelt. [...]

um mal selbst zu sehn ob sich irgendwas auf der victron seite tut hab ich mal sehr hacky  set_dbus_values erweitert um
            self._dbusservice["/Ac/MaxPower"] = 600
            self._dbusservice["/Ac/PowerLimit"] = 600
damit wird zumindest mal die gesetzte Leistungsbegrenzung (neuer Unterpunkt "Leistungsbegr. Zero-Feed-In") in der Konsole angezeigt - ist es das was du meinst mit "es schreibt nichts rein" - also du siehst keine veränderung bei vorhandenem überschuss?

[image]

0x7878, 2023-06-14:
Ich bleibe da am Ball. Ich dachte halt, das der Multiplus / cerbo von selbst werte zur Verfügung stellt die dann den fronius beispielsweise limitiert (Wunschdenken). Super Sache @onceuponatime78  ich denke, wenn wir die Punkte schonmal über Venus konfigurieren können ist es schonmal ein guter Start. Meine Idee bisher wäre ähnlich wie es Telekatz macht regelmäßig neu zu limitieren nur bekomme brauche ich eine andere Quelle für die Berechnung der Limitierung. (Zähler wäre wohl am sinnvollsten). Hub4/acPowerSetpoint liefert bei mir das was der Multiplus setzt um auszugleichen. Das macht er solange wie er kann. Irgendwann ist die Batterie aber voll und da der Multiplus dann nicht mehr laden kann ist der Wert 0.

0x7878, 2023-06-15:
Ja, ich hatte sogar 100w zugelassen. Vieleicht klappt es ja bei dir und ich habe bei mir nur irgendwas falsch eingestellt.

0x7878, 2023-06-15:
Mir ist aufgefallen, dass die normale Regelung irgendwie Probleme macht, wenn ein Inverter diesen Path bereitstellt. Zumindest hat sich meine Anlage nicht mehr auf 0 einpendeln können.

Auch im Internet habe ich heute nichts weiter über dieses Thema finden können außer ein Forum Eintrag bei Victron dazu und ein anderes Projekt, welches versucht, dieses Feature zu integrieren. Erfolgreicher war er dabei auch nicht und beschreibt das selbe Problem. --> [Victron Community](https://community.victronenergy.com/questions/194871/ess-regulation-becomes-slow-and-less-accurate-when.html)

Ich habe eine externe Steuerung fast fertig, die meine PV-Inverter regelt. Diese Logik könnte ich auch in das Projekt bringen. Abhängig wäre die Logik lediglich vom Zähler. Nach wie vor halte ich das aber für doppelt gemoppelt, da es ja so was schon gibt (PowerLimit) bzw. geben müsste.  Vielleicht macht Victron hier noch was.  

@henne49, @dsteinkopf  Was ist eure Meinung dazu?

onceuponatime78, 2023-06-15:
ergänzend dazu noch meine Erfahrung: mit "Maximale Einspeisung" = 0 wird mit gestztem PowerLimit Strom aus dem Netz bezogen, obwohl nicht notwendig aufgrund der Gesamtproduktion, mit 100W regelt Victron auf pi mal Daumen null. 

Zusätzlich hab ich die initialisierung bei mir noch von PowerLimit=MaxPower auf PowerLimit=0W geändert (wie in  https://github.com/victronenergy/dbus-fronius/blob/dd92f42aa2b23830d66e5be5228d3fb19d3f94e3/software/src/inverter.cpp#L215) um zu sehen, ob das ESS das PowerLimit im Betrieb dann hochregelt - leider auch mit dem Ergebnis dass sich nichts im PowerLimit Parameter ändert...irgendwas fehlt uns wohl im Verständnis oder der Vollständigkeit der API Dokumentation um die Regelungsfunktion vom venus zu nutzen...

closed 2023-08-28 by henne49

henne49, 2023-11-21:
Zum Thema Null Einspeisung https://github.com/helgeerbe/OpenDTU-OnBattery/wiki

markus2701, 2025-02-21:
Hallo.
Gibt es hierzu schon neue Erkenntnisse?

Lieben Gruß, Markus