henne49/dbus-opendtu #72: "null" als Werte für Power, Current, Volt ?

buntelichter, 2023-04-14:
Hallo zusammen,
ich versuche mit der Templatefunktion von dbus-opendtu die Werte von ElecPow (PowR2) in den Victron zu bekommen.
Das funktioniert schon ganz gut, leider wird nach dem Abschalten vom Relais "null" für Power, Current, Volt? übertragen, was die letzten Werte "einfrieren" lässt. Gibt es eine Möglichkeit "null" zuzulassen (event. Wert ---). Vielen Dank

henne49, 2023-04-15:
Was ist das für ein Template? und schaltest du den ElecPow (PowR2) aus? Weil wenn er Null anzeigt, sollte das auch funktionieren. Oder geht es darum, dass wir dann die letzten noch zu lesenden Werte anzeigen?

@dsteinkopf das erscheinbt mir wie das Problem das wir mal bei der ahoy hatten, sprich wenn der Timestamp zu alt ist, schreibe 0 ?

dsteinkopf, 2023-04-15:
> @dsteinkopf das erscheinbt mir wie das Problem das wir mal bei der ahoy hatten, sprich wenn der Timestamp zu alt ist, schreibe 0 ?

Ja, das glaube ich auch: Unser Code ist so gebaut, dass er **keine Werte meldet, wenn er keine Werte lesen kann** (Timeout, Fehler beim Parsen etc.). Wenn da wirklich "null" als Wert kommt, kann er das nicht zu einer Zahl machen und bricht den Durchlauf ab. Bei Victron (Venus) **bleibt also der letzte Wert stehen**, weil er überhaupt nichts mehr gesendet bekommt. Das ist bei kurzen Aussetzern auch gut und richtig.

Bei den Invertern haben wir eine **Prüfung auf Aktualität** (Ahoy: Timestamp von dem die letzten Werte stammen, OpenDTU: Flag "reachable"). Wenn die feststellt, dass keine aktuellen Daten vorliegen, dann meldet sich unser Script beim Venus, schickt aber keine neuen Werte. Dies wird dann offenbar dort **als "---" dargestellt**.

Wir bräuchten für die Templates also auch so eine "Aktualitätsprüfung". Wir müssen nur vorsichtig sein:
* Wir sollten nicht fälschlich 0-Werte senden, weil die dann Graphen kaputt aussehen lassen. Oder andere Skripten/Logiken interpretieren das als "jetzt ist die Nacht vorbei" und fangen z.B. neu an zu zählen.
* Auch die "---"-Werte sollte nur kommen, wenn wir uns sicher sind, dass es nicht nur ein temporärer Fehler ist.
* Die Logik muss allgemeingültig für alle "Template"-Geräte sein. Schlimmstenfalls konfigurierbar.

Wie könnte also so eine Logik zu Aktualitätsprüfung aussehen?

**Erste Idee**: Wir speichern im Script, wann der letzte Wert erfolgreich übermittelt wurde und nach einstellbarer Zeit (MagAgeTsLastSuccess vielleicht) wird dann "---" genommen (also wir melden uns beim Venus, aber ohne Werte).

henne49, 2023-04-15:
ja der manage Ansatz klingt mir am Vernüftigsten und ist ja auch auch schon implementiert für die DTUs

buntelichter, 2023-04-15:
Hallo zusammen,
das ist das original Template, nur die Pfade an ElecPow angepasst. ElecPow (PowR2) bleibt an, nach Abschalten vom Relais wird nur noch der  Spannungswert gesendet und Power und Current als "null". Alle drei Werte bleiben ab dann wie eingefroren  in der Victron- Anzeige stehen bis das Relais wieder einschaltet und normale Werte gesendet werden.

henne49, 2023-04-15:
Ich verstehe nicht, welches Relais das sein soll? Und was dann einfriert? Hast du screenshots?

buntelichter, 2023-04-15:
Hallo ,
das Relais ist im POWR2, schaltet den Ausgang und wird von ElecPow gesteuert (Sonnenaufgang /Sonnenuntergang).
Ich nutze den POWR2 als Stromzähler, wie viele es mit Tasmota machen, ich aber mit ElecPow.

https://www.coolhardware.de/doku.php?id=elecpow:sonoffpowr2.

[image]

Die Werte für Spannung, Strom, Leistung bleiben nach Abschalten des POWR2- Relais in der Victron Anzeige AC Phase L1 stehen. Das hat elektrisch nichts miteinander zu tun, ElecPow sendet unterschiedlich (Relais an/aus).

henne49, 2023-04-15:
Jetzt verstehe ich das Problem :-). Also als naheliegender Variante, schalte das Relais mal nicht mehr ab über Nacht und schau mal ob die Daten dann stimmen. Und versuche mal später abzuschalten zb bei Sonne bei -10°. Dann sollte der Inverter irgendwann auf Null gehen, das Problem würde ich ungern bei uns fixen, wenn es spezifisch am Relais liegt.

buntelichter, 2023-04-15:
Hallo, 
vergesse mal das Relais, das ist bei mir eh überbrückt.
ElecPow sagt Relais an und sendet "Zahlenwert" , "Zahlenwert" , "Zahlenwert" = alle drei Anzeigen funktionieren und werden aktualisiert (bei mir alle 5 sec.)
ElecPow sagt Relais aus und sendet "Zahlenwert" , "null" , "null"  = keine der drei Anzeigen wird erneuert.

Wenn man jetzt einen Verbraucher darüber abfragt (com.victronenergy.acload), (Relais schaltet wirklich zb. 1000W) dann bekomme ich immer 1000W angezeigt, einmal aktuell wenn Relais an und wirklich verbraucht wird, und als stehende Anzeige wenn Relais aus ist und nicht verbraucht wird.

dsteinkopf, 2023-04-15:
Was sagst du zu meinen Überlegungen oben? Lassen sich deine Beobachtungen damit überhaupt erklären?

buntelichter, 2023-04-15:
Hallo,
genau das sehe ich auch so:
"Wenn da wirklich "null" als Wert kommt, kann er das nicht zu einer Zahl machen und bricht den Durchlauf ab. Bei Victron (Venus) bleibt also der letzte Wert stehen, weil er überhaupt nichts mehr gesendet bekommt. Das ist bei kurzen Aussetzern auch gut und richtig."

Der Wert "null" ist kein Aussetzer, der kommt mit jeder Anfrage von ElecPow "zuverlässig", solange Relais = aus. 
"null" = ein empfangener Wert für keine Werte vorhanden = (--).

buntelichter, 2023-04-29:
Hallo zusammen,
dbus-opendtu funktioniert mit ElecPow >1.3k  ( REST-API : http://elecpow.local/sensor) ohne Probleme, solange ElecPow das Relais eingeschaltet hat (wie oben beschrieben). 
Ist der Schaltzustand "aus" bleiben die letzten übertragenen Werte im Victron (Venus) stehen bis der Schaltzustand von ElecPow wieder auf "ein" wechselt, dann werden sofort die Werte wieder übertragen und angezeigt. (also kein "Einfrieren" wie von mir befürchtet)
Danke an **@dsteinkopf** für die Erklärung der Arbeitsweise von dbus-opendtu.
Ich habe meine Schaltzeiten daraufhin angepasst. Danke an **@henne49** für den Tip.
Für die kommenden Versionen von dbus-opendtu wäre es jedoch schön, wenn "null" als kein Fehler erkannt und verarbeitet würde :) .
Vielen Dank an alle Beteiligten von dbus-opendtu.
Wer es selber probieren möchte, die einzutragenden Pfade sind :
ENERGY/Total
ENERGY/Power
ENERGY/Voltage
ENERGY/Current

dsteinkopf, 2023-04-29:
> Wie könnte also so eine Logik zu Aktualitätsprüfung aussehen?
> 
> **Erste Idee**: Wir speichern im Script, wann der letzte Wert erfolgreich übermittelt wurde und nach einstellbarer Zeit (MagAgeTsLastSuccess vielleicht) wird dann "---" genommen (also wir melden uns beim Venus, aber ohne Werte).

**Zweite Idee**: Wir führen einen neuen Konfigurationsparameter ein (z.B. SendNullAsNoValue), mit dem wir einschalten, dass immer dann, wenn null als Wert kommt, eben keiner gesendet werden soll. Das kommt mir jetzt fast sinnvoller vor, weil dann immer sofort die Werte verschwinden und nicht erst nach einer gewissen einstellbaren Zeit.

@buntelichter kannst du uns Beispiel-JSONs schicken (mit und ohne null)?

buntelichter, 2023-04-29:
Hallo zusammen,
**@dsteinkopf**  ich hoffe du kannst damit etwas anfangen ?

 http://elecpow.local/sensor

Relais an:

{"ENERGY":{"Total":2332.164,"Yesterday":10.078,"Today":7.350,"PowerPeakTotal":2991.6,"PowerPeakYesterday":2974.6,"PowerPeakToday":2929.8,"Power":0.0,"ApparentPower":41.6,"ReactivePower":41.6,"Factor":0.00,"Voltage":225.01,"Current":0.185},"AIR":{"Temperature":null,"Humidity":null,"Pressure":null,"Gas":null,"TempDew":null}}

Relais aus:

{"ENERGY":{"Total":2332.164,"Yesterday":10.078,"Today":7.350,"PowerPeakTotal":2991.6,"PowerPeakYesterday":2974.6,"PowerPeakToday":2929.8,"Power":null,"ApparentPower":null,"ReactivePower":null,"Factor":null,"Voltage":225.66,"Current":null},"AIR":{"Temperature":null,"Humidity":null,"Pressure":null,"Gas":null,"TempDew":null}}

0x7878, 2023-05-06:
Hi @buntelichter,

ich habe mal naiv wie ich bin einen Nullcheck eingebaut. Das Ganze kannst du, wenn du magst, testen.  [feateure-null-value-handling](https://github.com/henne49/dbus-opendtu/tree/feateure-null-value-handling). Das  ist von mir nicht getestet und blind einfach runter geschrieben.

dsteinkopf, 2023-05-06:
@0x7878 Wenn ich es richtig verstanden habe, dann ist das nicht der Wunsch. Bei null soll nicht 0 kommen, sondern „—“. 

> **Zweite Idee**: Wir führen einen neuen Konfigurationsparameter ein (z.B. SendNullAsNoValue), mit dem wir einschalten, dass immer dann, wenn null als Wert kommt, eben keiner gesendet werden soll. [...]

Das ist nach wie vor meine favorisierte Lösung.

buntelichter, 2023-05-07:
Hallo zusammen,
ich denke auch, die richtige Lösung ist @dsteinkopf s "Zweite Idee".
"null" ist eben nicht 0 sondern heißt "undefiniert, unbekannt, nicht vorhanden, nicht gemessen etc.".
Wenn 0 gemessen wird, dann wird 0 auch gesendet und angezeigt. 
Vielen Dank @0x7878 für den ersten Ansatz.

0x7878, 2023-05-08:
Bitte versuche diese Branch mal aus. Standardmäßig werden null Values als None interpretiert. Was Victron damit macht, weiß ich nicht. Welcher wert anstelle von "null" auftauchen soll, lässt sich in der config einstellen. 

Dazu gibt es folgende Settings: 

- CUST_Power_Default
- CUST_Total_Default
- CUST_Voltage_Default
- CUST_Current_Default
- CUST_DCVoltage

Berichte bitte ob es damit klappt. Pull request ist bereits erstellt aber ohne echten Test wird es nicht gemerged.

buntelichter, 2023-05-08:
Hallo zusammen, 
ich bekomme leider beim Installieren eine Fehlermeldung ?
[image]
Muss ich andere Pfadnamen nehmen ?

0x7878, 2023-05-08:
Ich glaube beim entpacken heißt der Ordner anders (der Feature branch statt main) mit `ls` kannst du dir ansehen was im Ordner vorhanden ist. Deine Vermutung war richtig, der Pfad muss geändert werden .

buntelichter, 2023-05-09:
Hallo zusammen,
@0x7878 vielen Dank für die gute Anleitung,
leider funktioniert es nicht 0sec und auch keine Log ?
[image]

buntelichter, 2023-05-09:
Hallo @0x7878,
ich habe das ganze gerade frisch installiert.
Muss die Main schon installiert sein?
[image]

Wo finde ich die Settings :
    CUST_Power_Default
    CUST_Total_Default
    CUST_Voltage_Default
    CUST_Current_Default
    CUST_DCVoltage
In der Config.ini  sind die nicht ?

0x7878, 2023-05-09:
Wie ich gesagt habe, bitte das zip herunterladen. Das ist die alte Version mit dem Bug. Du musst keine configs setzen. Der Standart ist None,

buntelichter, 2023-05-09:
Hallo @0x7878 ,
alles neu aufgespielt mit den oben genannten Pfaden.
Laufzeit 1 sec.
[image]
[image]

0x7878, 2023-05-09:
Ja aber es wurde zip1 gespeichert und du entpackt die zip also alte Version.

buntelichter, 2023-05-09:
Hallo @0x7878 
vielen Dank, ohne deine Hilfe keine Chance.......aber......es läuft , nicht nur das Programm, auch die Funktion :

Relais an:
[image]

Relais aus:
[image]
👍

0x7878, 2023-05-09:
@buntelichter das freut mich sehr! 🎉

Dann werde ich den Code noch ein bisschen aufhübschen. Danach landet dann das Feature mit im Main branch.  ☺️

buntelichter, 2023-05-09:
@0x7878 
Super 👍

buntelichter, 2023-05-09:
Hallo @0x7878 ,
mir ist gerade aufgefallen, in der Grafik verschwindet das Gerät:
Relais an:
[image]

Relais aus:
[image]

henne49, 2023-05-09:
Works as designed, das ist normal

0x7878, 2023-05-09:
Das wird sicher von Victron so gewollt sein. Meine PV wechselrichter liefern auch nachts 0 Watt. Deswegen war mein erster Vorschlag auch 0 und nicht None. Weil ich es mir aber schon gedacht habe, ist es konfigurierbar
[image]

buntelichter, 2023-05-10:
Hallo @0x7878 ,
für mich kein Problem, ich wollte nur darauf hinweisen.
👍

closed 2023-06-23 by 0x7878