henne49/dbus-opendtu #84: AC over all value missing in Venus OS

Hofyyy, 2023-05-07:
Hello,

I would like to see the "A" Value over time in my Victron VRM Panel.
At the moment only the "W" value is collected.

I think this depends on the missing "A" value over all. See screenshot.

Do I miss a configuration, or is that just not set at the moment?

[image]

0x7878, 2023-05-07:
I'm afraid that this is nothing we could handling on our side. There's no way to set a proper dbus path. See victrons list of all possible paths [Victrons list](https://github.com/victronenergy/dbus_modbustcp/blob/master/CCGX-Modbus-TCP-register-list.xlsx).

Maybe it should be calculated automatically but for some reason it won't work with custom device's like ours.

Hofyyy, 2023-05-07:
Oh no.
Ok I saw it in System: Battery and Voltage and thougt it might be available for PV Inverter too.