henne49/dbus-opendtu #86: Service won’t start anymore

testpaul999, 2023-05-09:
**Follow the Troubleshooting Section**
Change the Logging Parameter under DEFAULT in /data/dbus-opendtu/config.ini to Logging = DEBUG, please revert, once debugging and troubleshooting is complete. Rerun the script and share the current.log and config.ini file.
[Troubleshooting Section](https://github.com/henne49/dbus-opendtu#troubleshooting )

**Describe the Problem**
A clear and concise description of what the issue is.
- Service won’t start anymore
- 
**Device you are using**
Steps to reproduce the behavior:
1. Which Venus OS are you using? - v3.00~37\39
2. Which Victron Device? - RPI 
3. Which DTU? - opendtu - v23.5.3
4. Which Template Device? - no, only pvinverter 0

**Expected behavior**
A clear and concise description of what you expected to happen.
Service won’t start anymore

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
[New Text Document.txt](https://github.com/henne49/dbus-opendtu/files/11428131/New.Text.Document.txt)

dsteinkopf, 2023-05-09:
This seems to be the core problem (from "New Text Document.txt"):
```
root@raspberrypi2:~#  python /data/dbus-opendtu/dbus-opendtu.py
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 16, in <module>
    import tests
  File "/data/dbus-opendtu/tests.py", line 13, in <module>
    from dbus_service import DbusService
  File "/data/dbus-opendtu/dbus_service.py", line 7, in <module>
    from vedbus import VeDbusService
ModuleNotFoundError: No module named 'vedbus'
```

@testpaul999: Is Venus version 'v3.00~37\39' a current beta version? Then, this might be the source of the problem (which we'll have to solve). Did it still work when using Venus 2.x?

testpaul999, 2023-05-09:
Is Venus version 'v3.00~37\39' a current beta version? - yes!
Did it still work when using Venus 2.x? - don’t know, but run fine with Venus-os v3.0.0~32.
I‘ll go back and test, thx for idea.

dsteinkopf, 2023-05-09:
If it is working on the older versions, then it seems to be some change they made in the new version. We'll have to adapt this in our script then...

testpaul999, 2023-05-09:
So have test.
Have Setup a 2.94 plain on RPI.
Install with the suggested way from GitHub.
But same issue.
Then I use a backup und now it run with 2.94.
Next I‘ll upgrade to 3.00-39 and this also run
Maybe some wrong with the files at GitHub?

Grüße aus Berlin

henne49, 2023-05-09:
we need the current.log file again, or can you confirm, that the error is the same?

closed 2023-05-09 by dsteinkopf

testpaul999, 2023-05-09:
👍🏻 thx run fine

PR #87 (2023-05-09): fix import vedbus (must be placed after the sys.path.insert)
fixes #86