henne49/dbus-opendtu #88: AcPosition=2 is not showing in RemoteConsole/Dashboard

ThoWiSi, 2023-05-10:
**Follow the Troubleshooting Section**
Change the Logging Parameter under DEFAULT in /data/dbus-opendtu/config.ini to Logging = DEBUG, please revert, once debugging and troubleshooting is complete. Rerun the script and share the current.log and config.ini file.
[Troubleshooting Section](https://github.com/henne49/dbus-opendtu#troubleshooting )

**Describe the Problem**
When AcPosition=2 is entered in the config.ini (where 3 of my Hoymiles are physically connected to the Multiplus II) then they don‘t show on the Remote Console / Dashboard.
With AcPosition=0 or =1 it works!

**Device you are using**
Steps to reproduce the behavior:
1. Which Venus OS are you using? 2.92
2. Which Victron Device? CerboGX & Multiplus II 48/5000
3. Which DTU? ESP32 v23.5.9
4. Which Template Device? Hoymiles 1200 and 1500

**Expected behavior**
AcPosition=2 should produce a similar result to =1 (as in the Screenshot) just the line from the box „Pv Inverters“ connected slightly lower

**Screenshots**

[image]

**Additional context**
Add any other context about the problem here.

henne49, 2023-05-10:
According to documentation you can use position 2. but It seems it has no effect, also we do not write anything to vrm portal. This is coming from Venus OS. So for now this is nothing can change, other than stating you are right we have noticed the same

ThoWiSi, 2023-05-11:
Thank you for the confirmation!Maybe you could add a note in the documentation that AcPosition=2 is currently not working as expected for other people?It took me a while to figure out that the fault was not in my settings!Regards and thank you very much for your great work, Thorsten