henne49/dbus-opendtu #95: Generation is not subtracted from consumption

Black-Death-HH, 2023-06-18:
Hello, I am using Venus OS 3.00 and have successfully integrated a HM-1500 via opendtu. Everything is detected cleanly, only the generation is not deducted from consumption. The PV inverter sits at AC-Out1 and so it is configured. I currently have NO Multiplus installed, however it worked with Venus OS v2.92. Downgrade is not currently possible as the plant is at a remote location. I can't set anything else in Venus OS to fix the problem. Is there a possibility here? I have already performed a reinstallation of dbus-opendtu.

Black-Death-HH, 2023-06-18:
[image]

0x7878, 2023-06-18:
Ok, I see. You would expect to have ac-lasten L1 to 18w (594-576). First of all I did not Update to v3.00 and I won't since there are no new features I need. Could You try to config your pv-inverter to ac-in? I know it's not what you want since the graphic is wrong but maybe it is correctly deducted in the end. 

Just a sidenote: even though I have only original Victron products installed (in the past), "ac Lasten" was never be accurate to me.

Black-Death-HH, 2023-06-18:
[image]

On ACin it looks like the correct subtraction 😀

magic21nrw, 2023-06-22:
Hello,

I like to link in as it seems that something has changed with me, too. Seems to be the same problem, I guess:

Setup: 
* Grid Meter configured as Grid Meter showing the total consumption of our house.
* Hoymiles Inverter configured like mentioned in your readme and put onto AC Eingang (Parameter 0)

I get the same picture like posted in the post before with the only difference that there is the same amount shown on AC Eingang and AC Lasten. Normally I would expect the AC-Lasten to be higher as the Grid Meter measures the consumption which is lowered by the pvinverter.

My config of the plugin is the following running with the latest Victron OS 3.10 Beta 8.

[DEFAULT]
# Time in minutes how often a status is added to the log-file current.log with log-level INFO
SignOfLifeLog= 1

# Number ob Template Inverter to query
NumberOfTemplates=0

# Which DTU to be used ahoy, opendtu, template
DTU=opendtu

# send YieldDay instead of YieldTotal
useYieldDay=1

#For ESP8266 reduce polling intervall to reduce load
ESP8266PollingIntervall=10000

#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
#To keep current.log small use ERROR
Logging=ERROR

# if ts_last_success is older than this number of seconds, it is not used.
# Set this to < 0 to disable this check.
MagAgeTsLastSuccess=600

# if this is not 0, then no values are actually sent via dbus to vrm/venus.
DryRun=0

#IP of Device to query
Host=10.X.X.X

HTTPTimeout=2.5

# Username/Password leave empty if no authentication is required
Username =
Password =

### Only needed for OpenDTU and ahoy
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2
# 1st inverter
[INVERTER0]
Phase=L3
DeviceInstance=34
AcPosition=0
Servicename=com.victronenergy.pvinverter

# 2nd inverter
[INVERTER1]
Phase=L2
DeviceInstance=35
AcPosition=0

Rest of it unchanged.

The log with debug transmit the values correctly and there seems to be no obvious error. Also the values show up correctly in the victron system.

[image]

This shows the installed devices:

[image]

OpenDTU Vales seem to be transferred correctly but the display on the main screen (see above) seems to be wrong.

[image]

[image]

For the sake of completeness - the grid meter which should be set up correctly (somehow strange as it is not displayed as "grid" in the Venus os...

[image]

magic21nrw, 2023-06-22:
Remark: On Version 2.94 it seems to work properly although the PVinverter is drawn as separate input on the Victron Multiplus (which is not installed in my case).

[image]

0x7878, 2023-06-23:
Thank you for checking this out. I don't know what Victron changed in their system. I already read the changelog but there is nothing about this. :-(

magic21nrw, 2023-07-07:
anyone else who can maybe confirm that is it working again in one of the beta-Versions > Version 3.0 or 3.10 ???

Black-Death-HH, 2023-09-16:
I have tested version 3.10 today. Same Issue.

mapietru, 2023-09-23:
I am also very sad about this. Discovered the behaviour after upgrading from 2.94 to 3.10. What about AC meters like EM112?

snoopymad, 2023-09-26:
Hello, is there an issue request at Victron? Is it worthwhile for several people to open a corresponding case?

henne49, 2023-09-26:
I do not think there is an issue, this is a very cosmetic issue for display and for historic reporting, that being said, It can absolutely make sense to open one. We just did not do it, as it does not affect us to invest the time yet.

mhin76, 2023-10-30:
hi all. I have the same issue (using 4 hoymiles with Venus OS v3.11 and EM540 as grid meter). As soon as the grid meter shows power feed to grid the AC load is equal to inverter load. I wrote that issue on the venus OS github, but they mentioned, that this design is not supported: https://github.com/victronenergy/venus/issues/1169#issuecomment-1783404488

it is not just a cosmetic issue, but also the historic reporting is wrong. I noticed that this solution here is not simulating an inverter in the venus OS under inverter setting, might that be an issue?

0x7878, 2023-10-30:
> I noticed that this solution here is not simulating an inverter in the venus OS under inverter setting, might that be an issue?

What do you mean by that? Our solution was never officially supported. With version 3, Victron dropped the functionality on which this project is based. We don't stimulate a device; we provide the data that Victron needs to know about that device. However, it appears that you have to use version < 3 or purchase Victron products or supported products like Fronius.If I remember correctly, there was someone who got this working as soon as they connected their Multiplus 2 device.

mhin76, 2023-10-30:
> What do you mean by that? Our solution was never officially supported. With version 3, Victron dropped the functionality on which this project is based. We don't stimulate a device; we provide the [...]

thank you very much for your quick answer and that feedback. I have missed that important part! Will downgrade to < 3 version and once I have a MultiPlus test it out. I will also inform others in the venus OS thread about that info! Much appreciated!

mhin76, 2023-10-31:
I downgraded to version v2.94 but still get the same behaviour. As soon as energy is set to grid, AC load is equal to PV inverter amount.
[image]

I tested dowgrading to v2.92 (as some mentioned above that they had the correct calculation at AC load), but seems that my energy meter EM540 is not supported yet. Venus OS did not recognize it. I have not tested yet v2.93 ;-)

0x7878, 2023-10-31:
2.94 is fine. For me, it looks like the AC input may not be set correctly. I did a short research but did not find any clue for a requirement of a specific firmware for the EM540. Could you post your config? (without passwords of course.)

mhin76, 2023-10-31:
> 2.94 is fine. For me, it looks like the AC input may not be set correctly. I did a short research but did not find any clue for a requirement of a specific firmware for the EM540. Could you post your config? (without passwords of course.)

ah, that might be the reason. I try to set the AC input to "Grid", but as soon as I selected it and exist system setup and re-enter, it is back to unknown.
This is now on version 3.12 but it was same behavior on any firmware, also 2.94
[image]

going back to System setup
[image]

mhin76, 2023-10-31:
I amanged to set the AC Input now permanently to Grid by using two commands I found here at the buttom:
https://community.victronenergy.com/questions/76999/ac-inputs-on-venus-os-on-raspberry-pi.html

```
dbus -y com.victronenergy.settings /Settings AddSetting SystemSetup AcInput1 0 i 0 2
dbus -y com.victronenergy.settings /Settings AddSetting SystemSetup AcInput2 0 i 0 2
```
still wrong AC Load figures when power is going to grid. will check how to export my config to share it here.

0x7878, 2023-10-31:
I meant the config of dbus-opendtu. It must be set on the same input as your grid is working on to subtract. Otherwise, the system thinks there is another installation where your PV is connected. Hope you understand

mhin76, 2023-10-31:
@0x7878  that was it!! I had a wrong config!! not sure from what template I copied it but I had AcPosition=1. Now setting it to 0 it shows all correct figures:
[image]

YOU JUST MADE MY DAY!!! Thank you! working on v3.12. Will inform the other thread about this setting, they might have it wrong too.