henne49/dbus-opendtu #100: CRITICAL Error at main Traceback (most recent call last):

nico-hl, 2023-06-23:
Hey there,

today I`ve update Ahoy on 0.6.9 and now it doesn't works with Venus OS

Here is the log:
`KeyError: 'inverter'
2023-06-23 15:06:41,872 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 446, in main
    service = DbusService(
  File "/data/dbus-opendtu/dbus-opendtu.py", line 70, in __init__
    self.numberofinverters  = self._getNumberOfInverters()
  File "/data/dbus-opendtu/dbus-opendtu.py", line 202, in _getNumberOfInverters
    numberofinverters = len(meter_data['inverter'])`

Anyone have an idea ? 

Venus OS is v3.0
Ahoy is on ESP32

nico

nico-hl, 2023-06-23:
Hey :) 
Thanks for your help. But I didn't found that with the key and I uninstalled everything from dbus and reinstalled 
now it works :)

0x7878, 2023-06-23:
Glad to hear that and thanks for the feedback 👍

closed 2023-06-23 by 0x7878