henne49/dbus-opendtu #102: Directory '/data/dbus-opendtu' does not exist

schmidt3152, 2023-06-29:
Directory '/data/dbus-opendtu' does not exist . Bekomme die Daten nicht geladen .

henne49, 2023-06-29:
with so little information, we are unable to help...

schmidt3152, 2023-06-29:
Hello henne I have a problem running my system does not recognize the installation file. when I want to open the ini via nano for editing, no data appears. Best regardsVon meinem iPhone gesendet

henne49, 2023-06-29:
can you please paste the commands and response you see on the terminal?

Also try this lines of commands: 

wget https://github.com/henne49/dbus-opendtu/archive/refs/heads/main.zip
unzip main.zip "dbus-opendtu-main/*" -d /data
mv /data/dbus-opendtu-main /data/dbus-opendtu
chmod a+x /data/dbus-opendtu/install.sh

schmidt3152, 2023-06-29:
Yes 

***@***.***:~# unzip main.zip "dbus-opendtu-main/*" -d /data
Archive:  main.zip
replace dbus-opendtu-main/.github/ISSUE_TEMPLATE/issue-report.md? [y]es, [n]o, [A]ll, [N]one, [r]ename: error: invalid response [m]
replace dbus-opendtu-main/.github/ISSUE_TEMPLATE/issue-report.md? [y]es, [n]o, [A]ll, [N]one, [r]ename: All
error: invalid response [a]
replace dbus-opendtu-main/.github/ISSUE_TEMPLATE/issue-report.md? [y]es, [n]o, [A]ll, [N]one, [r]ename: All
  inflating: dbus-opendtu-main/.github/ISSUE_TEMPLATE/issue-report.md
  inflating: dbus-opendtu-main/.github/workflows/codeql.yml
  inflating: dbus-opendtu-main/.github/workflows/pylint.yml
  inflating: dbus-opendtu-main/.gitignore
  inflating: dbus-opendtu-main/.pylintrc
  inflating: dbus-opendtu-main/.vscode/settings.json
  inflating: dbus-opendtu-main/LICENSE
  inflating: dbus-opendtu-main/README.md
  inflating: dbus-opendtu-main/config.ini
  inflating: dbus-opendtu-main/constants.py
  inflating: dbus-opendtu-main/dbus-opendtu.py
  inflating: dbus-opendtu-main/dbus_service.py
  inflating: dbus-opendtu-main/docs/ahoy_0.5.93_inverter-id-0.json
  inflating: dbus-opendtu-main/docs/ahoy_0.5.93_live.json
  inflating: dbus-opendtu-main/docs/ahoy_0.5.93_record-live.json
  inflating: dbus-opendtu-main/docs/ahoy_0.6.9_live.json
  inflating: dbus-opendtu-main/docs/ahoy_pre_0.5.93.json
  inflating: dbus-opendtu-main/docs/opendtu_status.json
  inflating: dbus-opendtu-main/helpers.py
  inflating: dbus-opendtu-main/img/ahoy-as-inverter.png
  inflating: dbus-opendtu-main/img/ahoy-as-pv-inverter.png
  inflating: dbus-opendtu-main/img/device.png
  inflating: dbus-opendtu-main/img/devicedetails.png
  inflating: dbus-opendtu-main/img/devicelist.png
  inflating: dbus-opendtu-main/img/open-dtu.png
  inflating: dbus-opendtu-main/img/opendtu-security.png
  inflating: dbus-opendtu-main/img/overview.png
  inflating: dbus-opendtu-main/install.sh
  inflating: dbus-opendtu-main/restart.sh
  inflating: dbus-opendtu-main/service/run
  inflating: dbus-opendtu-main/test_helpers.py
  inflating: dbus-opendtu-main/tests.py
  inflating: dbus-opendtu-main/uninstall.sh
***@***.***:~# nano /data/dbus-opendtu/config.ini

henne49, 2023-06-29:
your are not moving the files, execute all commands:

mv /data/dbus-opendtu-main /data/dbus-opendtu
chmod a+x /data/dbus-opendtu/install.sh
nano /data/dbus-opendtu/config.ini

closed 2023-06-29 by henne49