henne49/dbus-opendtu #103: /data/dbus-opendtu/uninstall.sh did NOT prevents it from being restarted

gerd-git, 2023-07-12:
**Describe the Problem**
/data/dbus-opendtu/uninstall.sh did NOT prevents it from being restarted

**Device you are using**
Venus OS v3.00
raspberrypi2
opendtu

**Expected behavior**
"/data/dbus-opendtu/uninstall.sh stops the service and prevents it from being restarted (e.g. after a reboot)."

root@raspberrypi2:~# /data/dbus-opendtu/uninstall.sh
root@raspberrypi2:~# reboot
Broadcast message from root@raspberrypi2 (pts/0) (Wed Jul 12 10:46:29 2023):
The system is going down for reboot NOW!
root@raspberrypi2:~#
login as: root
root@venus.local's password:
Last login: Wed Jul 12 10:47:12 2023
root@raspberrypi2:~# svstat /service/dbus-opendtu
/service/dbus-opendtu: up (pid 2024) 178 seconds
root@raspberrypi2:~#

henne49, 2023-07-13:
I can confirm this. 

The problem is that we write the install.sh script into /data/rc.local
for firmware upgrades. we need to delet the entry from rc.local to fix this.

henne49, 2023-07-13:
PR Request is submitted to fix it

closed 2023-07-14 by henne49