henne49/dbus-opendtu #105: Is there another way to start the service? (not via rc.local?)

ClashC, 2023-07-25:
Is there another way to start the service? Or does the rc.local always have to be available and executable under /data as a prerequisite for starting the dbus-opendtu service? 
I also use SetupHelper and RpiDisplaySetup on a current VictronOS with rpi3b+.

But I would also like to thank you for your great work here. Everything works very well and reliably for me.

---------------
Gibt es einen anderen Weg, den Service zu starten? Oder muss immer als Voraussetzung zum starten des dbus-opendtu-Services die rc.local unter /data bereitstehen und ausführbar sein? 
Ich benutze zusätzlich SetupHelper und RpiDisplaySetup auf einem aktuellem VictronOS mit rpi3b+.

Ich möchte mich aber auch hier für deine tolle Arbeit bedanken. Bei mir funktioniert alles sehr gut und zuverlässig.

henne49, 2023-07-25:
we use the /data folder and rc.local as they are persistent even with upgrades and reboots and allow us to execute the scipts even after updating Venus OS. I would not know any other way, how to how accomplish this.

ClashC, 2023-07-27:
> we use the /data folder and rc.local as they are persistent even with upgrades and reboots and allow us to execute the scipts even after updating Venus OS. I would not know any other way, how to how accomplish this.

These are indeed good arguments and I had not considered the Victron system update. My idea was in the direction of init.d or systemd...
I have to see why with the current release 3.x, the "/data/dbus-opendtu/install.sh" is not called every now and then after a boot... rc.local are executable (chmod 755 /data/rc.local and /data/dbus-opendtu/install.sh too)
Reporting...

dsteinkopf, 2023-07-27:
From time to time, I also have the problem that one of my manually installed services in /data/... were not started after a reboot. As it is not reproducable and the venus (raspi) is not rebooted very often, it is very hard to debug and fix. I am assuming a racing condition or timing problem. So I added a `sleep 5` in `/data/rc.local`, but this did not help reliably.

ramodemah, 2023-10-22:
Ich habe ein Cerbo-GX und bei mir läuft der Dienst (Pythonscript) nicht an. Die Current.log wird auch nicht geschrieben. Habe schon mehrfach alles neu installiert, jedoch ohne Erfolg. Starte ich das Script manuell, bekomme ich einen Syntax-Fehler:

"root@einstein:/data/dbus-opendtu# ./dbus-opendtu.py
  File "./dbus-opendtu.py", line 33
    config.read(f"{(os.path.dirname(os.path.realpath(__file__)))}/config.ini")
                                                                            ^
SyntaxError: invalid syntax
root@einstein:/data/dbus-opendtu#"

Hat jemand eine Idee woran das liegen könnte?

Gruß Omar

henne49, 2023-10-22:
Du musst auch mit python oder python3 das script startenSent from mobile, excuse any typos!

ramodemah, 2023-10-22:
Das liefert das gleiche Ergebnis:
root@einstein:/data/dbus-opendtu# python dbus-opendtu.py
  File "dbus-opendtu.py", line 33
    config.read(f"{(os.path.dirname(os.path.realpath(__file__)))}/config.ini")
                                                                            ^
SyntaxError: invalid syntax
root@einstein:/data/dbus-opendtu#

0x7878, 2023-10-22:
Welche Firmware? Ist die config.ini auch wirklich vorhanden? Ggf. Liegt der Fehler ja auch in der configuration selbst.

ramodemah, 2023-10-22:
Firmware: v2.73
config.ini ist editiert und vorhanden

Gruß Omar

Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/emailmobilapp_ios_smartphone_footerlink.htm>

service? (not via rc.local?) (Issue #105)
Datum: 22.10.2023, 18:53 Uhr
An: henne49/dbus-opendtu ***@***.***
CC: ramodemah ***@***.*** ***@***.***> >, 
Comment ***@***.*** ***@***.***> >

ramodemah, 2023-10-22:
Wie gesagt, der Dienst kommt nicht ans laufen, es wird auch keine current.log geschrieben.

0x7878, 2023-10-22:
Vielleicht ist die verwendete Python Version zu alt. Spricht etwas dagegen auf 2.9* upzugraden? Ich verstehe deinen Frust aber schwer zu sagen was da gerade falsch läuft zumal die Version aus den repo bei uns allen läuft.

0x7878, 2023-10-22:
Der Code denn du gepostet hast, stimmt nicht mit der aktuellen Version überein.

```diff
def main():
    '''main loop'''
    # configure logging
    config = configparser.ConfigParser()
-   config.read(f"{(os.path.dirname(os.path.realpath(file)))}/config.ini")
+   config.read(f"{(os.path.dirname(os.path.realpath(__file__)))}/config.ini")
    logging_level = config["DEFAULT"]["Logging"].upper()
    dtuvariant = config["DEFAULT"]["DTU"]
```
Bitte lade dir die letzte Version noch mal herunter und versuche es damit dann nochmal. 
_Kleiner Hinweis lösche deine alten Downloads, sonst wird dort einfach eine Zahl angehangen und der Setup-Prozess schlägt fehl._

ramodemah, 2023-10-22:
Ich habe den Download-Link aus der Anleitung genommen, ist der falsch?
https://github.com/henne49/dbus-opendtu/archive/refs/heads/main.zip

Wo finde ich die aktuelle Version, sorry ich bin erstmalig bei github unterwegs......

ramodemah, 2023-10-22:
Habe alles gelöscht und die aktuelle herunter geladen, dass Ergebnis ist jedoch ähnlich:

root@einstein:/data/dbus-opendtu# python dbus-opendtu.py
  File "dbus-opendtu.py", line 33
    config.read(f"{(os.path.dirname(os.path.realpath(__file__)))}/config.ini")
                                                                            ^
SyntaxError: invalid syntax
root@einstein:/data/dbus-opendtu#

0x7878, 2023-10-22:
```
root@einstein:~# python -V
Python 3.8.13
```

Ein Upgrade sollte kein Thema sein. Also ich meine damit, die Firmware deines cerbos/Raspberry zu updaten. Ich fahre mit folgender Version sehr gut und wie oben zu sehen ist, wird auch eine ebenfalls neuere Python Version verwendet. Es könnte sein, dass es genau daran liegt.

[image]

ramodemah, 2023-10-23:
Yes!!! Das war es, auf die 2.93 upgedated und schon läufts :-)))

[image]

[image]

Vielen,vielen Dank für die Unterstützung und jetzt werde ich wohl erst einmal nicht weiter updaten :-)

Gruß Omar

closed 2023-11-17 by henne49