henne49/dbus-opendtu #109: Template Wmin to kwh

caraddiction, 2023-08-27:
Hi,

at the Template for Shelly Plus 1PM the CUST_Total_Mult Value is 1
In my opinion it should be 0.000017 to have the exact amount of kwh? 

Regards

caraddiction, 2023-08-27:
Jetzt habe ich selber gesehen unter config.ini Template 1 sind die 0.000017 unter Template 2 für den Shelly Plus 1PM nicht. 
passt es wenn ich hier die 0.000017 einfach ändere?

So schaut Template 2 Original aus:

[TEMPLATE2]
## Shelly Gen 2 - Only works with Digest Auth = True and Username and Password!
## Documentation https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/EM#status
##
## Username/Password leave empty if no authentication is required
Username = 
Password =
Host=172.16.1.1
DigestAuth = True
CUST_SN = 12345678
CUST_API_PATH= rpc/Switch.GetStatus?id=0
CUST_POLLING = 2000
CUST_Total= aenergy/total
CUST_Total_Mult = 1
CUST_Power= apower
CUST_Power_Mult = 1
CUST_Voltage=  voltage
CUST_Current=  current
Phase=L3
DeviceInstance=46
AcPosition=1
Name= ShellyPlus1PM
Servicename=com.victronenergy.pvinverter

henne49, 2023-08-28:
yes that should work if you want to use kwh, as this is a template, people might want to use wh, therefore it was still 1. 

https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/Switch#status

caraddiction, 2023-08-28:
CUST_Total= aenergy/total

Do I have to multiply this value with 1000 to geht kwh?

Regards

henne49, 2023-08-28:
you would have to multiply with 0.001 to get kWH

closed 2023-08-31 by caraddiction