henne49/dbus-opendtu #110: AcPosition=0 is not showing in RemoteConsole/Dashboard

svenshub, 2023-09-01:
**Follow the Troubleshooting Section**
Change the Logging Parameter under DEFAULT in /data/dbus-opendtu/config.ini to Logging = DEBUG, please revert, once debugging and troubleshooting is complete. Rerun the script and share the current.log and config.ini file.
[Troubleshooting Section](https://github.com/henne49/dbus-opendtu#troubleshooting )

**Describe the Problem**
I have 2 Hoymiles PV Inverter physically connected tho the OpenDTU.
When AcPosition=0 (ac_input) is entered in the config.ini, then they don‘t show on the Remote Console / Dashboard.
With AcPosition=1 it works!
But, if i use the the AcPosition=1 (ac_output) in the config.ini, then is the AC_Loads calculation incorrect.
What can i do?

**Device you are using**
Steps to reproduce the behavior:
1. Which Venus OS are you using? -> FW: v3.10-32 @raspberryPI 3B
2. Which Victron Device? -> MP2 3000 / 48
3. Which DTU? -> OpenDTU / FW: v23.8.22
4. Which Template Device? -> Two Hoymiles HM600, (more to come)

**Expected behavior**
AC_IN coppled PV Inverter with correct show on the Remote Console / Dashboard and correct AC_Loads calculation.

**Screenshots**
AC_Position = 0
[image]

AC_Position = 1
[image]

**Additional context**
Add any other context about the problem here.
[config.ini.txt](https://github.com/henne49/dbus-opendtu/files/12496837/config.ini.txt)
[current.log](https://github.com/henne49/dbus-opendtu/files/12496838/current.log)

henne49, 2023-09-06:
Do you have another PV or only hoymiles? Can you try to downgrade the multiplus firmware?

henne49, 2023-09-06:
Can you see the Hoymiles in the DeviceList, even with AcPosition=0?
[image]

svenshub, 2023-09-06:
hello henn49,
thank you for your feedback.

In this 1st install phase, i have only two Hoymiles interverter installed yet. More to come.

The grid meter is SDM630 type and is emulated with the bridge from:
https://github.com/stormmurdoc/victron_sdm630_bridge
The Source data comes from:
https://github.com/volkszaehler/mbmd
and this setup running normaly.

Grid meter:
[image]

1st Hoymiles interver:
[image]

2nd: Hoymiles inverter:
[image]

Device List in AcPosition=0:
[image]

The raw data are in venusOS included and in victronVRM reported:
[image]

**The problem is, i can´t see the PV inverter with the config.ini AcPosition=0 in the remote console!**
[image]

I tested multiple settings with the Gui Mods settings:
(Flow overwiev=simple, =DC Coupled, =AC Coupled, Victron Stock as well)
[image]

and together in combination with AC input/output loads and Show Loads On Input: 
[image]

But in AcPosition=0 is **NO** PV Inverter displayed in the remote Console:
[image]

######

If i switch the two invter in the config.ini to AcPosition=1, the inverter are now displyed. Looks like good:
[image]

But in the 3 phase details view, you can see, the AC Loads calulation is wrong.
[image]

The negatives AC Input phases (L1 and L3) are not subtracted in the AC Loads!
The PV Inverter Power (L1 and L3) are 1 to 1 in the AC Loads displayed.

Where is the bug?
Thank you for your help.

regards
sven

svenshub, 2023-09-06:
additional comment:

I tested with the following venus OS firmware on a Raspberry Pi 3B:
-v3.01
-v3.10-32
-v3.10-34 (currently installed)

The same results.

regards
sven

henne49, 2023-09-06:
Sorry, it seems that it is more a display issue, than a bug, it works for me also with latest release candidate, I do not even know where to change the GUI display mode.

henne49, 2023-09-06:
can you also check the VRM Portal, what is displayed there?

Cassiopea1, 2023-09-06:
Hi,

Same here. I've just installed the HM-1500, but it does not show up in the Overview page or VRM as PV inverter.
Venus OS large v3.01

[image]
[image]
[image]

My config.ini :

`[DEFAULT]
SignOfLifeLog= 1

NumberOfTemplates=0

DTU=ahoy

useYieldDay=0

ESP8266PollingIntervall=10000

Logging=ERROR

MaxAgeTsLastSuccess=600

DryRun=0

Host=192.168.1.147

HTTPTimeout=2.5

Username =
Password =

[INVERTER0]
Phase=L1
DeviceInstance=34
AcPosition=0
Servicename=com.victronenergy.pvinverter
`

svenshub, 2023-09-07:
with config.ini setup AcPosition=1 looks like as follow:

remote console overview:
[image]
remote console device List:
[image]

VRM Portal:
[image]
update is life:
[image]

but both views with **incorrect AC Loads** (see L1 and L3 Loads).

------

with config.ini setup AcPosition=0 looks like as follow:
remote console overview:
[image]
remote console device List:
[image]

VRM Portal:
[image]
update is also life:
[image]

you see, the data inputs are correct and reported to the VRM Portal.
But i have no PV Inverter in the remote console and also the **incorrect AC Loads** (see L1 and L3 Load = 0 Watts).

I use the GuiMods v8.02
[image]
mey be there is the problem?!?

But in the Victron stock overview, also no PV Inverter displayed and the AC Loads are incorrect.

Witch module is calculating the AC Loads?

hd-schardt, 2023-09-07:
Hello,
I can confirm the same behaviour here with me.
VenusOS 3.01 large
Greetings Heinz-Dieter

0x7878, 2023-09-07:
I think Victron changed something in their major version (3.0) and above. I'm running 3 inverters in Total on AC position 0. 

I'm sorry for your situation but I don't think that there is anything we can do about it. 

[image]
[image]

hd-schardt, 2023-09-07:
I also think it is due the changeover to Version 3.x and possibly also to changes in the VRM.
Thanks for your help!

henne49, 2023-09-07:
Let us check, if we need to set the inverter value negativ for newer Venus Versions.

henne49, 2023-09-08:
that does not work as well. Then Critical Loads is 0. 

So for now, nothing we can do. 
[image]

svenshub, 2023-09-11:
Thank you all for your inputs!

I downgraded the firmware to last v2.x (v2.94) version, and everything is fine!
The two inverter ar now running in AcPosition=0 (where they are physicly located as well) and the AC Load calculation is correct now!

Thanks a lot.

regards
sven

[image]

[image]

[image]

closed 2023-09-12 by henne49

DerJungeHerr, 2023-10-05:
@henne49 @svenshub is there any news from Victron about this behavior?

0x7878, 2023-10-06:
@DerJungeHerr At the moment it's still the same situation. No one has tried a newer version because of various problems. If there is a new version you could try and tell us if something has changed. The documentation from Victron does not change and we have done everything as described.

DerJungeHerr, 2023-10-06:
> @DerJungeHerr At the moment it's still the same situation. No one has tried a newer version because of various problems. If there is a new version you could try and tell us if something has changed. The documentation from Victron does not change and we have done everything as described.

I installed the latest beta  3.1xxxx something .... still not working. I will downgrade my system to the last working version 2.94 .. I hope Victron is working on the issue ...

0x7878, 2023-10-06:
> I installed the latest beta  3.1xxxx something .... still not working. I will downgrade my system to the last working version 2.94 .. I hope Victron is working on the issue ...

Thanks for your time and effort. I hope so, too.

Cassiopea1, 2023-10-06:
For me the problem corrected by itself after I actually installed the multiplus. It just appeared...

[image]

DerJungeHerr, 2023-10-06:
> For me the problem corrected by itself after I actually installed the multiplus. It just appeared...
> 
> [image]

Version of VenusOS ? Can you give more details about your System?

@henne49 @0x7878 @svenshub I will start an issue soon I guess I will inform you here, maybe we can push it a little bit in the Victron Venus repo ;)

Cassiopea1, 2023-10-07:
> Version of VenusOS ? Can you give more details about your System?
>
> @henne49 @0x7878 @svenshub I will start an issue soon I guess I will inform you here, maybe we can push it a little bit in the [...]

Venus OS: v3.10 on RPi4
Ahoy DTU: v0.7.36

When I integrated the HM-1500 into the Venus OS, the multiplus was not connected to the Venus OS (was not delievered at that time) and the PV Inverter was not visible on the display. Later, when I connected the multiplus, the PV Inverter appeared on the display.

addo88, 2024-05-13:
Same Problem with my HM-600 and HM-800 via ahoyDTU,  subscribed and hopefully 

will temporary downgrade to ver. 2.94 :(

 @henne49 thanks a lot for your amazing work !!!