henne49/dbus-opendtu #113: No device in Remote Console visible

rela7, 2023-09-13:
1. Venus OS: v3.01
2. Victron Device: Raspberry 3
3. Which DTU: AhoyDTU, release 0.7.36
4. Which Template Device? ???, no

I've used the recent scripts from your website for installation.
The Ahoy firmware is up to date.

When I started the process I've got the following (Logging=DEBUG):
2023-09-13 09:47:32,20 root DEBUG is_data_up2date: inverter #0: age_seconds=14421705, max_age_ts=600
2023-09-13 09:47:32,25 root DEBUG is_data_up2date: inverter #0: age_seconds=10, max_age_ts=600
2023-09-13 09:47:32,27 root DEBUG tests have passed
2023-09-13 09:47:32,29 root INFO Start
2023-09-13 09:47:32,42 root INFO Registering dtu devices
2023-09-13 09:47:32,73 root DEBUG calling http://192.168.0.143/api/live with timeout=2.5
2023-09-13 09:47:32,108 urllib3.connectionpool DEBUG Starting new HTTP connection (1): 192.168.0.143:80
2023-09-13 09:47:32,233 urllib3.connectionpool DEBUG http://192.168.0.143:80 "GET /api/live HTTP/1.1" 200 545
2023-09-13 09:47:32,242 root DEBUG function fetch_url finished in 169 ms
2023-09-13 09:47:32,245 root DEBUG calling http://192.168.0.143/api/inverter/id/1 with timeout=2.5
2023-09-13 09:47:32,260 urllib3.connectionpool DEBUG Starting new HTTP connection (1): 192.168.0.143:80
2023-09-13 09:47:32,285 urllib3.connectionpool DEBUG http://192.168.0.143:80 "GET /api/inverter/id/1 HTTP/1.1" 200 405
2023-09-13 09:47:32,294 root DEBUG function fetch_url finished in 49 ms
2023-09-13 09:47:32,296 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 109, in main
    service = DbusService(
  File "/data/dbus-opendtu/dbus_service.py", line 84, in __init__
    self.numberofinverters = self.get_number_of_inverters()
  File "/data/dbus-opendtu/dbus_service.py", line 307, in get_number_of_inverters
    meter_data = self._get_data()
  File "/data/dbus-opendtu/dbus_service.py", line 492, in _get_data
    self._refresh_data()
  File "/data/dbus-opendtu/dbus_service.py", line 387, in _refresh_data
    self.check_and_enrich_ahoy_data(meter_data)
  File "/data/dbus-opendtu/dbus_service.py", line 418, in check_and_enrich_ahoy_data
    meter_data.append({})
AttributeError: 'dict' object has no attribute 'append'

The content of the config.ini
[DEFAULT]
# Time in minutes how often a status is added to the log-file current.log with log-level INFO
SignOfLifeLog= 1

# Number ob Template Inverter to query
NumberOfTemplates=0

# Which DTU to be used ahoy, opendtu, template
DTU=ahoy

# send YieldDay instead of YieldTotal
useYieldDay=0

#For ESP8266 reduce polling intervall to reduce load
ESP8266PollingIntervall=20000

#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
#To keep current.log small use ERROR
Logging=ERROR

# if ts_last_success is older than this number of seconds, it is not used.
# Set this to < 0 to disable this check.
MaxAgeTsLastSuccess=600

# if this is not 0, then no values are actually sent via dbus to vrm/venus.
DryRun=0

#IP of Device to query
Host=192.168.0.143

HTTPTimeout=2.5

# Username/Password leave empty if no authentication is required
Username =
Password =

### Only needed for OpenDTU and ahoy
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2
# 1st inverter
[INVERTER0]
Phase=L1
DeviceInstance=34
AcPosition=0
#Servicename=com.victronenergy.charger
Servicename=com.victronenergy.pvinverter

# 2nd inverter
[INVERTER1]
Phase=L2
DeviceInstance=35
AcPosition=0

# 3rd inverter
[INVERTER2]
Phase=L3
DeviceInstance=36
AcPosition=1

# 4th inverter
[INVERTER3]
Phase=L1
DeviceInstance=37
AcPosition=1

# 5th inverter
[INVERTER4]
Phase=L2
DeviceInstance=38
AcPosition=1

# 6th inverter
[INVERTER5]
Phase=L3
DeviceInstance=39
AcPosition=1

# 7th inverter
[INVERTER6]
Phase=L1
DeviceInstance=40
AcPosition=1

# 8th inverter
[INVERTER7]
Phase=L2
DeviceInstance=41
AcPosition=1

# 9th inverter
[INVERTER8]
Phase=L3
DeviceInstance=42
AcPosition=1

# 10th inverter
[INVERTER9]
Phase=L1
DeviceInstance=43
AcPosition=1

################## TEMPLATES #####################
####Rearrange and customite Templates as necessary
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2

[TEMPLATE0]
## Tasmota Example
##
## Username/Password leave empty if no authentication is required
Username =
Password =
DigestAuth = False
Host=172.16.1.1
CUST_SN = 12345678
CUST_API_PATH= cm?cmnd=STATUS+8
CUST_POLLING = 2000
CUST_Total= StatusSNS/ENERGY/Total
CUST_Total_Mult = 1
CUST_Power= StatusSNS/ENERGY/Power
CUST_Power_Mult = 1
CUST_Voltage=  StatusSNS/ENERGY/Voltage
CUST_Current=  StatusSNS/ENERGY/Current
Phase=L1
DeviceInstance=47
AcPosition=1
Name= Tasmota
Servicename=com.victronenergy.grid

[TEMPLATE1]
## Shelly Gen 1 authenticated and unathenticated example
## Documentation: https://shelly-api-docs.shelly.cloud/gen1/#shelly1-1pm-status
##
## Username/Password leave empty if no authentication is required
Username =
Password =
DigestAuth = False
Host=172.16.1.1
CUST_SN = 12345678
CUST_API_PATH= status
CUST_POLLING = 2000
CUST_Total= meters/0/total
###Shelly1PM Multiplier Watt/min 0.000017
CUST_Total_Mult = 0.000017
CUST_Power= meters/0/power
CUST_Power_Mult = 1
CUST_Voltage= none
CUST_Current= none
Phase=L2
DeviceInstance=45
AcPosition=1
Name= Shelly 1PM
Servicename=com.victronenergy.genset

[TEMPLATE2]
## Shelly Gen 2 - Only works with Digest Auth = True and Username and Password!
## Documentation https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/EM#status
##
## Username/Password leave empty if no authentication is required
Username =
Password =
Host=172.16.1.1
DigestAuth = True
CUST_SN = 12345678
CUST_API_PATH= rpc/Switch.GetStatus?id=0
CUST_POLLING = 2000
CUST_Total= aenergy/total
CUST_Total_Mult = 1
CUST_Power= apower
CUST_Power_Mult = 1
CUST_Voltage=  voltage
CUST_Current=  current
Phase=L3
DeviceInstance=46
AcPosition=1
Name= ShellyPlus1PM
Servicename=com.victronenergy.pvinverter

Can anyone tell me what is wrong?

dsteinkopf, 2023-09-13:
(I "allowed" myself to add tripple backticks around your quotes)

dsteinkopf, 2023-09-13:
Your problem really seems to be a bug in the code which happens when you have a "gap" in the sequence of inverter numbers that are returned by Ahoy DTU. Can you check that? So the numbering should be 0,1,2,... and not 0,2,3...
As a quick workaround you could try to change the numbers.

Anyway we should fix the bug.

dsteinkopf, 2023-09-13:
From your logs you seem to have no inverter ID "0". This is probably causing the bug to appear. Try to change your inverter "1" to be a "0".

dsteinkopf, 2023-09-13:
Could you provide us (for testing) the result of your `http://192.168.0.143/api/live`?

rela7, 2023-09-13:
Thank you. The missing inverter ID "0" had caused the problem. After deleting the inverter in the AHOY settings and reinstalling it all works well.

dsteinkopf, 2023-09-13:
I am happy to hear that. fine!

> Could you provide us (for testing) the result of your `http://192.168.0.143/api/live`?

it would be great if you could get an example output of this call anyway

rela7, 2023-09-14:
Here is the output of http://192.168.0.143/api/live:
`{"generic":{"wifi_rssi":-92,"ts_uptime":2777,"ts_now":1694698541,"version":"0.7.36","build":"ba218ed","menu_prot":false,"menu_mask":61,"menu_protEn":false,"esp_type":"ESP8266"},"refresh":30,"ch0_fld_units":["V","A","W","Hz","","°C","kWh","Wh","W","%","var","W"],"ch0_fld_names":["U_AC","I_AC","P_AC","F_AC","PF_AC","Temp","YieldTotal","YieldDay","P_DC","Efficiency","Q_AC","MaxPower"],"fld_units":["V","A","W","Wh","kWh","%","W"],"fld_names":["U_DC","I_DC","P_DC","YieldDay","YieldTotal","Irradiation","MaxPower"],"iv":[true,false,false,false]}`

rela7, 2023-09-14:
Output of `http://192.168.0.143/api/live`:
> {"generic":{"wifi_rssi":-92,"ts_uptime":2777,"ts_now":1694698541,"version":"0.7.36","build":"ba218ed","menu_prot":false,"menu_mask":61,"menu_protEn":false,"esp_type":"ESP8266"},"refresh":30,"ch0_fld_u [...]

dsteinkopf, 2023-09-14:
This is the corrected one, I assume... :-(

PR #116 (2023-09-17): fix bug in handling of gaps in Ahoy inverter number sequence
fixes #113
closed 2023-09-17 by 0x7878