henne49/dbus-opendtu #117: Request: Multiple openDTU in one VenusOS

TheGrumpyBadger, 2023-09-28:
I would be nice to have the ability to add multiple openDTU at once.

Usecases:
- Hoymiles devices to fare to connect to one openDTU
- to many Hoymiles devices for one openDTU

henne49, 2023-09-28:
I guess that would be possible, but just iterating over multiple ahoy or opendtus

UdoD5, 2023-10-13:
> I guess that would be possible, but just iterating over multiple ahoy or opendtus

Hi, I would appriciate this too. There is an additional reason. Hoyemiles HMS* Inverter need another RF Module instead of the NRF24L01+ for the HM* series it uses CMT2300A for the new one. I have the "old" RF module running for my two HM Inverter and a device with CMT2300A for the one HMS Inverter. I don't necessarely want to combine them because I can place the new one closer to the actual inverter. 

Is this already possible or how do I itarate across two instances?

0x7878, 2023-10-13:
I keep this in mind when it comes to refactoring the base.  For now, you can simply copy opendtu to opendtu2 and start it. So you will have 2 instances of it. This is how it runs in my installation.

UdoD5, 2023-10-13:
> I keep this in mind when it comes to refactoring the base. For now, you can simply copy opendtu to opendtu2 and start it. So you will have 2 instances of it. This is how it runs in my installation.

so you mean the folder /data/dbus-opendtu to /data/dbus-opendtu2 right?  ... and make sure the DeviceInstance ID's do not in conflict  after that start the service for the second folder .. right?

0x7878, 2023-10-13:
Yes, that's right. It works very well. Give it a try. I've got ahoyDTU and openDTU running this way. Different folders, of course, and different instance IDs.

OlafWalther1969, 2024-10-07:
Hello,
Would someone be kind enough to post the installation script for a second instance.
Unfortunately, I'm not sure where I need to rename the folders.

Thank you very much!

Hallo,
würde Jemand so nett sein, dass Installationsskript für eine zweite Instanz zu posten.
Ich bin mir leider nicht sicher, an welchen Stellen ich die Ordner umbennen muss.

Dankeschön!

DerWintergriller, 2025-02-06:
Hallo.

Hat schon jemand eine Lösung gefunden?
Ich habe es versucht. Funktioniert aber noch nicht.
Bin wie folgt vorgegangen:
Den Ordner /data/dbus-opendtu in /data/dbus-opendtudach kopiert
Befehle:  
cp /data/dbus-opendtu -r  /data/dbus-opendtudach
chmod a+x /data/dbus-opendtudach/install.sh

Config angepasst: nano /data/dbus-opendtudach/config.ini Mit Anpassen: (DeviceInstance=44-53; IP;Phase;AcPosition=0)

Ausführen /data/dbus-opendtudach/install.sh
Da bekomme ich den Fehler:
ln: /service/dbus-opendtudach/service: File exists

Was ist falsch?

Danke. :-)

DasQwarx, 2025-07-20:
Ich versuche gerade eine zweite Opendtu als template einzupflegen und scheitere an den APis knoten. Mit Ahoy DTU gings (in bin jetzt von ahoy auf open gewechselt) kann mir bitte jemand da bei den nötigen Template daten helfen?

`[TEMPLATE0]
## Tasmota Example
##
## Username/Password leave empty if no authentication is required
Username =admin
Password =meinpasswort
DigestAuth =FALSE
Host =192.168.1.116
CUST_SN =116183138979
CUST_API_PATH =api/livedata/status?inv=116183XXXXXX
CUST_POLLING =10000
CUST_Total =inverters/0/INV/0/YieldDay
CUST_Total_Mult =0.001
CUST_Power =inverters/0/AC/0/Power/v
CUST_Power_Mult =1
CUST_Voltage =inverters/0/AC/0/Voltage/v
CUST_Current =inverters/0/AC/0/Current/v
CUST_DCVoltage =inverters/0/DC/0/Voltage/v
Phase =L2
DeviceInstance = 44
AcPosition =0
Name =HM15Terras
Servicename=com.victronenergy.pvinverter
`