henne49/dbus-opendtu #119: DTU missing under dbus-spy Err 111

1411Alex, 2023-10-28:
Hi together,

I used the current script to install everything. More than once... but it doesn't work.

1. Which Venus OS are you using : 2,93, 2.94
2. Which Victron Device: Raspi4 and 3
3. Which DTU_ DTULite with an HM-800 inverter
4. Which Template Device: What it is?

I try differnt OS version and also diffent Raspi, it is not working.
What I do wrong? ;-(

Ping from the Pi to the IP is working.

Here is e screenshot from my debug file:

2023-10-28 15:30:30,821 root DEBUG is_data_up2date: inverter #0: age_seconds=18330283, max_age_ts=600
2023-10-28 15:30:30,823 root DEBUG is_data_up2date: inverter #0: age_seconds=10, max_age_ts=600
2023-10-28 15:30:30,824 root DEBUG tests have passed
2023-10-28 15:30:30,825 root INFO Start
2023-10-28 15:30:30,831 root INFO Registering dtu devices
2023-10-28 15:30:30,843 root DEBUG MaxAgeTsLastSuccess: 'MaxAgeTsLastSuccess'
2023-10-28 15:30:30,845 root DEBUG MaxAgeTsLastSuccess not set, using default
2023-10-28 15:30:30,846 root DEBUG calling http://192.168.178.106/api/livedata/status with timeout=2.5
2023-10-28 15:30:30,860 urllib3.connectionpool DEBUG Starting new HTTP connection (1): 192.168.178.106:80
2023-10-28 15:30:31,500 root DEBUG calling http://192.168.178.106/api/livedata/status with timeout=2.5
2023-10-28 15:30:31,507 urllib3.connectionpool DEBUG Starting new HTTP connection (1): 192.168.178.106:80
2023-10-28 15:30:32,20 root DEBUG calling http://192.168.178.106/api/livedata/status with timeout=2.5
2023-10-28 15:30:32,27 urllib3.connectionpool DEBUG Starting new HTTP connection (1): 192.168.178.106:80
2023-10-28 15:30:32,40 root CRITICAL Error at main
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/urllib3/connection.py", line 159, in _new_conn
    conn = connection.create_connection(
  File "/usr/lib/python3.8/site-packages/urllib3/util/connection.py", line 84, in create_connection
    raise err
  File "/usr/lib/python3.8/site-packages/urllib3/util/connection.py", line 74, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/urllib3/connectionpool.py", line 665, in urlopen
    httplib_response = self._make_request(
  File "/usr/lib/python3.8/site-packages/urllib3/connectionpool.py", line 387, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.8/http/client.py", line 1256, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1302, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1251, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1011, in _send_output
    self.send(msg)
  File "/usr/lib/python3.8/http/client.py", line 951, in send
    self.connect()
  File "/usr/lib/python3.8/site-packages/urllib3/connection.py", line 187, in connect
    conn = self._new_conn()
  File "/usr/lib/python3.8/site-packages/urllib3/connection.py", line 171, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object at 0xb5cadb50>: Failed to establish>

And this is the config:

[DEFAULT]
# Time in minutes how often a status is added to the log-file current.log with log-level INFO
SignOfLifeLog= 1

# Number ob Template Inverter to query
NumberOfTemplates=0

# Which DTU to be used ahoy, opendtu, template
DTU=opendtu

# send YieldDay instead of YieldTotal
useYieldDay=0

#For ESP8266 reduce polling intervall to reduce load
ESP8266PollingIntervall=10000

#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
#To keep current.log small use ERROR
Logging=DEBUG

# if ts_last_success is older than this number of seconds, it is not used.
# Set this to < 0 to disable this check.
MagAgeTsLastSuccess=600

# if this is not 0, then no values are actually sent via dbus to vrm/venus.
DryRun=0

#IP of Device to query
Host=192.168.178.106

HTTPTimeout=2.5

# Username/Password leave empty if no authentication is required
Username =
Password =

### Only needed for OpenDTU and ahoy
# AcPosition 0=AC input 1; 1=AC output; 2=AC output 2
# 1st inverter
[INVERTER0]
Phase=L2
DeviceInstance=34
AcPosition=1
#Servicename=com.victronenergy.charger

BR
Alex

henne49, 2023-10-28:
I guess you have setup a password and you do not allow access to opendtu?

1411Alex, 2023-10-28:
I've been looking for that too. I just haven't found a place at the DTU where I could use it...

henne49, 2023-10-28:
Check the documentation https://github.com/henne49/dbus-opendtu#security-settings-in-opendtu

1411Alex, 2023-10-28:
Maybe my problem is that I use the Hoymiles DTU Lite Stick?

0x7878, 2023-10-28:
I think so too :sad . I don't have one on my own. If there is an API then you could use a template instead of ahoyDTU or openDTU. Both variants work only with their hardware and software. Alternatively, just tinker or buy an ahoyDTU or openDTU device.

closed 2023-11-17 by henne49