henne49/dbus-opendtu #121: Show "Total Yield"

andreas-bulling, 2023-11-09:
It'd be nice if "Total Yield" would be shown in Venus OS like it is for MPPTs.

henne49, 2023-11-17:
where should this be shown? as we do show total Yield

[image]

andreas-bulling, 2023-11-17:
True - sorry, I missed that.

andreas-bulling, 2023-11-17:
Ah, but there is something else: I am using a used inverter that accumulated kWh already. I can fix the display in the OpenDTU webinterface through a negative offset but in the remote console the total is shown (i.e. the offset is not taken into account)

henne49, 2023-11-17:
Where can you configure a negative offset? If that is not working please report the issue at opendtu, as they need to provide the right data via api

andreas-bulling, 2023-11-17:
Inverter Settings -> String

henne49, 2023-11-18:
yep and that works as it should.

closed 2023-11-18 by henne49

andreas-bulling, 2023-11-19:
No, it doesn't. See

https://github.com/tbnobody/OpenDTU/issues/1512#issuecomment-1817970649

henne49, 2023-11-19:
Use useYieldDay=1 in config.ini

tbnobody/OpenDTU#1512 (comment)<https://github.com/tbnobody/OpenDTU/issues/1512#issuecomment-1817970649>