henne49/dbus-opendtu #127: Question: OpenDTU inverter selection

moonilightshaker, 2023-11-27:
Question: 
Is it possible to integrate only some of the OpenDTU inverters in Victron?
For example: I have several inverters in OpenDTU. But only 1 is relevant.

If I add only one to the configuration I get the following error:
KeyError: 'INVERTER1'

Best Regards
M.

0x7878, 2023-11-27:
Could you provide an anonymous output of the json from openDTU and your config? This should be possible. Maybe something is missing on our side or maybe something is not configured properly. Let's check.

0x7878, 2023-11-28:
> @0x7878 Attached are the files.
> [config.txt](https://github.com/henne49/dbus-opendtu/files/13491930/config.txt)
> [status.json](https://github.com/henne49/dbus-opendtu/files/13491931/status.json)
> 
> 

Thank you. I'm going to check that.

henne49, 2024-01-24:
@0x7878 any update or can we close?

henne49, 2024-03-03:
https://github.com/henne49/dbus-opendtu/pull/148 should help, as you can order the hoymiles to read only the first ones.

moonilightshaker, 2024-03-04:
> #148 should help, as you can order the hoymiles to read only the first ones.

Thank you for the change. Tested and working.

But I have one problem. How can I sort the inverters in OpenDTU? They are always sorted automatically by serial number.

henne49, 2024-03-04:
[image]
you can order by drag and drop and then save

andreas-doerfler, 2024-03-04:
Sorry but this index order does not reflect in the JSON coming from status page, one needs to consider the "order" in JSON itself:

[image]
[image]

2flea, 2024-03-04:
Sorry but this index order does not reflect in the JSON coming from status page, one needs to consider the "order" in JSON itself:

[image]
[image]

moonilightshaker, 2024-03-04:
> Sorry but this order does not reflect in the JSON coming from status page(index != order), one needs to consider the "order" in JSON itself:
> 
> ![image](https://private-user-images.githubusercontent.com/119484258/309673580-d821e2c1-f893-489f-891c-3de7ec1e0945.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk1Njc5MTEsIm5iZiI6MTcwOTU2NzYxMSwicGF0aCI6Ii8xMTk0ODQyNTgvMzA5NjczNTgwLWQ4MjFlMmMxLWY4OTMtNDg5Zi04OTFjLTNkZTdlYzFlMDk0NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwNFQxNTUzMzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iZjA2YmQxMTZjNmRkNjI4YmZjNGVlNWI0YWQ4NTAyNGMzNjYwYjlmMDkxMTk5YTYwMWNmMTBkZjY5MzQ2YjE3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.1uPjujvCOCdTI0rFeYO9bIL1WCPd3HGl7OVSZQhsX50) ![image](https://private-user-images.githubusercontent.com/119484258/309673877-6c45a18f-fb2a-420a-a8de-dfbd6aa49e8a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk1Njc5MTEsIm5iZiI6MTcwOTU2NzYxMSwicGF0aCI6Ii8xMTk0ODQyNTgvMzA5NjczODc3LTZjNDVhMThmLWZiMmEtNDIwYS1hOGRlLWRmYmQ2YWE0OWU4YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwNFQxNTUzMzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YjM3YTIwM2QzZmY1YTlkNzY3ODFkOWE5ZTQzZjM0MTM0OGM4Yjc5ZjE4NTlmZmU1MzcyZjg3ZGI4MTBiNjFkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Po614tffqlMBrfVCZ79Xnv8tPQSvpnLMCEMkmxDthog)

You're right. The order cannot be changed with this method.

henne49, 2024-03-06:
Good catch, then we need to modify the code. Somewthing we need to look into

KarlMeier1933, 2024-05-26:
Gibt es hierzu Neuigkeiten? Ich würde auch gerne nur auf EINEN Wechselrichter zugreifen, aber die Reihenfolge bleibt bei der Abfrage immer gleich, egal wie ich die Wechselrichter in der OpenDTU sortiere.

henne49, 2024-05-27:
nein, Stand heute nicht möglich und auch ein Nischen Use Case für die meisten Nutzer. Du könntest ja auch einfach 2 DTU betreiben, und den den du abfragen möchtest an einer DTU haben. Aktuell ist es ein weitere Code Workaround, den wir so nicht wollen und gut supporten können. 

Wir müssen zuerst einen kompletten re-work des Codes machen, dann können wir solchen Feature in Zukunft einfacher umsetzen. Daher kann ich auch nicht sagen wann das kommen wird, aber dass wir es gerne einbauen, wenn der Code es zulässt.

Das zu den Hintergründen.

KarlMeier1933, 2024-05-27:
Danke für die schnelle Antwort und die ausführliche Erklärung.
Die Lösung mit der zweiten DTU hatte ich auch schon in Betracht gezogen. Dann werde ich das erstmal so lösen. 👍🏻