henne49/dbus-opendtu #128: Question: Integration of APSystems EZ1-M

twachti, 2023-12-02:
I am trying to integrate an APSystems EZ1-M at the template section and i have to set a different port (8050). ist i possible to add the port an the end of the host address? (host=x.x.x.x:8050)

**Device you are using**
Steps to reproduce the behavior:
1. Which Venus OS are you using? 3.12
2. Which Victron Device? Raspberry Pi 3B+
3. Which DTU? none
4. Which Template Device? APSystems EZ1-M

The data output of  EZ1-M looks like this:

http://192.168.x.x:8050/getOutputData

{
    "data": {
        "p1": 16,
        "e1": 0.04427,
        "te1": 73.22394,
        "p2": 32,
        "e2": 0.0842,
        "te2": 95.60328
    },
    "message": "SUCCESS",
    "deviceId": "E07000000000"
}

henne49, 2023-12-02:
I guess you tested and it didn‘t work?

http://192.168.x.x:8050/getOutputData

{

twachti, 2023-12-02:
I will try to connect a Shelly Plus 1pm instead of EZ1-M to learn and debug und give the EZ1 another try if the Shelly is working...

henne49, 2023-12-02:
you have to set NumberOfTemplates=1 and please next time upload the config.ini as file github is mixing the comment with markup formating

henne49, 2023-12-03:
Well, nice, that it works now

closed 2023-12-03 by henne49