henne49/dbus-opendtu #129: keine Anzeige im VRM-Dashboard, in Remote-Konsole nur Anzeige von "VA" (keine Watt-Angabe)

ahhkoch, 2023-12-17:
**Follow the Troubleshooting Section**
Change the Logging Parameter under DEFAULT in /data/dbus-opendtu/config.ini to Logging = DEBUG, please revert, once debugging and troubleshooting is complete. Rerun the script and share the current.log and config.ini file.
[Troubleshooting Section](https://github.com/henne49/dbus-opendtu#troubleshooting )

**Describe the Problem**
In CerboGX-Remote-Console only value in "VA" is shown, not "Watt"

**Device you are using**
Hoymiles-600 with OpenDTU on ESP32, CerboGX

Steps to reproduce the behavior:
1. Which Venus OS are you using? => latest
3. Which Victron Device? => MP2, MPPT, CerboGX, SmartShunt
4. Which DTU => OPENDTU
5. Which Template Device? => None? Not required for Opendtu?

**Expected behavior**
See the OpenDTU in VRM as PV-Inverter with "Watts"
See the OpenDTU "Watt"-Value in Remote-Console

[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
My OpenDTU has the name "Stromer".

henne49, 2023-12-17:
Kannst du bitte die genaue Version des Venus OS angeben und die Konfig Datei anhängen, bitte nicht einfach rein pasten. 

Ist der der Schuppen auch eine OpenDTU?

ahhkoch, 2023-12-18:
Schuppen ist ein Victron-MPPT. 
Einen Fronius WR gibt es auch noch.

VenusOS ist Version 3.13

[config.ini.txt](https://github.com/henne49/dbus-opendtu/files/13701847/config.ini.txt)

henne49, 2023-12-18:
bitte mal die Line Servicename=com.victronenergy.inverter löschen, dann setzen wir den richtigen Default Service für PV Inverter. Ich habe auch auf die 3.13 geupdated uns schaue morgen mal, wenn die Sonne scheint, ob ich dann da VA oder W habe. Ich glaube aber es liegt am Servicename, den du falsch gesetzt hast

ahhkoch, 2023-12-19:
Hi,
jetzt sehe ich in der Remotekonsole  auch korrekt die W-Angabe. Danke dafür! :) Nur im Dashboard erscheint da nix vom dem Opendtu-PV-Inverter. Das wäre natürlich perfekt...

Grüße
[image]

henne49, 2023-12-20:
er sollte im Fronius versteckt sein ACPosition steht auf 0 bei dir. 

 Position shown in Remote Console (0=AC input 1; 1=AC output; 2=AC input 2 please do not use)

henne49, 2023-12-20:
Ins Dashboard geht nicht separat, du nur über den AC Out was machen, wenn da noch nichts ist. Aber dann ist es eben auch falsch, wenn es an AC In hängt. 

Danke für das Lob, das muss ich aber an die echten Kollegen wie @0x7878 und @dsteinkopf weitergeben.

closed 2023-12-21 by henne49