henne49/dbus-opendtu #131: Upgrade to latest version causes error message

andreas-bulling, 2024-01-05:
Since there are no upgrade instructions available I installed the latest version (1.0.0) over the already running previous one.

An error message is repeating frequently in the log:

```
2024-01-05 18:20:17,815 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-opendtu/dbus-opendtu.py", line 109, in main
    service = DbusService(
  File "/data/dbus-opendtu/dbus_service.py", line 97, in __init__
    self._dbusservice = VeDbusService(f"{servicename}.http_{self.deviceinstance}", dbus_conn)
  File "/opt/victronenergy/dbus-systemcalc-py/ext/velib_python/vedbus.py", line 78, in __init__
    self._dbusname = dbus.service.BusName(servicename, self._dbusconn, do_not_queue=True)
  File "/usr/lib/python3.8/site-packages/dbus/service.py", line 144, in __new__
    raise NameExistsException(name)
dbus.exceptions.NameExistsException: Bus name already exists: com.victronenergy.pvinverter.http_34
```

I didn't change the configuration and expected for the new version to just be a drop-in replacement, using the old configuration...

henne49, 2024-01-05:
This looks like the service is already or still running somewhere, did you start it manually? have you rebooted?

henne49, 2024-01-05:
move the old directory to a backup name reinstall new and copy config.ini back.

andreas-bulling, 2024-01-05:
I have rebotted - error remains. I had actually renamed the old directory and did a fresh install, copying the config back. Exactly as you suggested...

henne49, 2024-01-05:
how old was your config? and your install? can you upload your config.ini? please upload do not paste.

andreas-bulling, 2024-01-05:
now it suddenly seems to work. I emptied the log file, rebooted again - and I don't see any error messages anymore. Thanks a lot for your time/messages - sorry, it seems this was a false alarm

closed 2024-01-05 by andreas-bulling