henne49/dbus-opendtu #135: No function after reinstallation

e-i-k-e, 2024-01-24:
I once completely removed obendtu and downloaded and reinstalled it. 
Now I can't find any functionality.

Running VenusOS 3.14 on a Raspberry 3b.

[image]
[current.zip](https://github.com/henne49/dbus-opendtu/files/14044904/current.zip)

henne49, 2024-01-25:
you set number of templates to 1, but only configure the second template, therefore it is not working, put your template configuration under [TEMPLATE0], then it should work

e-i-k-e, 2024-01-26:
Danke!
Ja, dass war bereits der Fehler - wenn man mal schnell was ändert. ;) 
[config.zip](https://github.com/henne49/dbus-opendtu/files/14069404/config.zip)

Ich habe aber ein größeres Problem, und ich vermute das es durch diesen Treiber bzw. eher durch eine falsche Konfiguration verursacht wird.

Ich habe sehr sehr hohe seit einigen Wochen sehr Hohe Verbrauchs sowie Solardaten, die nicht stimmen und anscheinen so ausufern, sobald ich den Treiber aktiv habe. 

Ich füge einige Bilder an, die zeigen, was ich meine.  
[image]
[image]
[image]
[image]

Im vergleich zu vorgestern, bei nicht aktivem Treiber.
[image]

henne49, 2024-01-27:
Ist das Problem jeden Tag so oder nur seitdem es nun wieder läuft. Weil der Treiber ja einmal lief hat das System die alten Werte und jetzt kamen prlotzlich wieder Solar Daten (Größenordnung von 117kwh) hinzu und das sollte ein einmal off sein?
Stimmen den die Live Verbrauchsdaten?

henne49, 2024-01-27:
Und schau mal bitte im Template gibt es Faktoren, die du nutzen musst, gibt dein Template den Wert evtl in Wh und Victron nimmt KwH

Außerdem hast du beim Tasmota Servicename=com.victronenergy.inverter obwohl ein Wechselrichter ist, sollte also Servicename=com.victronenergy.pvinverter sein

e-i-k-e, 2024-01-27:
Ich glaube, ich habe es

Das Template für den Shelly2, anscheinend wird hier Wh übermittelt.

[image]

[image]

[image]

Das mit dem "Tasmota Servicename=com.victronenergy.inverter " sollte korrekt sein, da der Wechselrichter an der Batterie hängt, quasi so wie der MP2.

henne49, 2024-01-28:
ja den CUST_TOTAL_MULT musst du anpassen auf 0.001, dann sollte es kwh sein

bezüglich VRM kann ich dir nicht sagen, da müsstest du da mal nachfragen

e-i-k-e, 2024-01-28:
Vielleicht könnt ihr das in der config direkt für das Template des Shelly2 ändern.

closed 2024-01-28 by e-i-k-e

henne49, 2024-01-29:
eigentlich ist das Template in KWH lauffähig, so wurde es von mir in einem Shelly PM gen 2 auch getestet. welchen Shelly 2 meinst du denn?