henne49/dbus-opendtu #137: Feature Request: Support for 3Phase Inverters (HMT Series)

stxShadow, 2024-01-28:
As far as i see it is not possible to define an inverter to be multi phase? I own serveral HMT Inverters and i have to define a phase for them, what is not right. Power is distributed over all 3 phases so measurement over all is wrong in VenusOS.

Would be nice to have that enhancement.

Thanks ! 

Jens

0x7878, 2024-01-28:
Please read our readme to configure it. I also use HMT Inverters which works fine.

[readme](https://github.com/henne49/dbus-opendtu?tab=readme-ov-file#inverter-options)

henne49, 2024-01-28:
That is supported since 7 month

https://github.com/henne49/dbus-opendtu#inverter-options

3P as option.
Please use latest release and follow the upgrade instructions in the README

closed 2024-01-28 by 0x7878