henne49/dbus-opendtu #139: OpenDTU v24.2.12 breaking API changes

mapietru, 2024-02-13:
Just tried the latest version of OpenDTU, but as expected dbus-opdendtu stops working. I had to go back to v24.1.26...

"REAKING CHANGE: Web API Endpoint /api/livedata/status
To reduce the heap usage it is necessary to send the inverters one by one instead of a huge response. A simple call to `/api/livedata/status` returns just some very general information. If detailed inverter information are required the inverter serial number has to appended `?inv=<serial number>`.
The websocket also returns only one inverter at a time. It as to be assembled at client side."

0x7878, 2024-02-13:
Thank you for sharing. We will have to implement the newer firmware by time. You don't have the output of the JSON, don't you?

tehre85, 2024-02-13:
just tested with openDTU [v24.2.12]and run in the same.
json of api/livedata/status
{"inverters":[{"serial":"xxxxxxxxx","name":"xxxxxxx","order":0,"data_age":0,"poll_enabled":true,"reachable":true,"producing":true,"limit_relative":100,"limit_absolute":xxxxxx}],"total":{"Power":{"v":382.6000061,"u":"W","d":1},"YieldDay":{"v":378,"u":"Wh","d":0},"YieldTotal":{"v":1578.530029,"u":"kWh","d":3}},"hints":{"time_sync":false,"radio_problem":false,"default_password":false}} 

The new version provides much less compared to the v24.1.18.

Just swapped to 24.1.18 as this was my last stable version.

henne49, 2024-02-13:
First get inverster and serial numbers:
/api/livedata/status
https://www.opendtu.solar/firmware/web_api/#get-common-live-data
second:
/api/livedata/status?inv=inverter-serialnumber

We have to query every SN now separately:
https://www.opendtu.solar/firmware/web_api/

tehre85, 2024-02-15:
@henne49 i just anonymised the serial number and name, as well as the limit_absolute.

Ayo91, 2024-02-15:
same issue here... with the last version we have to request the data by inverter :O

henne49, 2024-02-16:
[inverter_details.json](https://github.com/henne49/dbus-opendtu/files/14310070/inverter_details.json)
[status.json](https://github.com/henne49/dbus-opendtu/files/14310071/status.json)

2bbionic, 2024-02-17:
ran into the samme gap :D
anyway, I collected and anoned the JSON files for you - just in case you should need them.
I use three H350; tied together in OpenDTU
[status.json](https://github.com/henne49/dbus-opendtu/files/14318757/status.json)
[inv-status.json](https://github.com/henne49/dbus-opendtu/files/14318758/inv-status.json)

henne49, 2024-02-19:
I have created a PR which should fix the API change and work with old and new API, please test the PR, if you can.

2bbionic, 2024-02-19:
It seems that at least the power of my individual panels is shown now - but not the summary yet:
[image]
The wattage itself fits and the log ("current.log") remains empty

henne49, 2024-02-19:
VRM has a delay sometimes. The remote console should show the correct data.
[image]

tehre85, 2024-02-20:
@henne49 many thanks!!
Just upgraded the DTU to latest [v24.2.16] and the dbus-opendtu as well, working as accepted.

2bbionic, 2024-02-20:
> VRM has a delay sometimes. The remote console should show the correct data. <img alt="grafik" width="488" src="https://private-user-images.githubusercontent.com/4662326/306005659-87d16ddb-53d6-4092-aaa0-aa9a97558810.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg0NDc1ODIsIm5iZiI6MTcwODQ0NzI4MiwicGF0aCI6Ii80NjYyMzI2LzMwNjAwNTY1OS04N2QxNmRkYi01M2Q2LTQwOTItYWFhMC1hYTlhOTc1NTg4MTAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjBUMTY0MTIyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTQ4ZjA5YWQ5YWE1NDcxZGUzNjE4MmI5NTFiYjQ3ZWIyMGE5NTM0OTRiZDEzYzE5NjMyYWE2NzU4ZjI0YWY1ZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.3PeM6kIwamLHle-AsRybpgealKaheP9JpUSGctK-jFg">

My Cerbo has obviouly issues with transfering its data to the portal - -don't know why.
Anyway - data are received correctly from the OpenDTU and injected to the GUI.
Thanks!

mapietru, 2024-02-21:
Just upgraded opendtu to v24.2.16 and can confirm venus os and vrm portal are working fine with dbus-opdendtu-1.1.0. Thanks for the quick fix!